coolsnowwolf/lede #10052: R22.8.2使用一段时间之后，端口无法DHCP，只能设置静态IP地址

yanxin152133, 2022-09-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：端口无法进行DHCP ，只能设置静态IP地址

### (2) 路由器型号和固件版本

A：
[image]

### (3) 详细日志

A：
以`eth1`为例：
[image]

系统日志：
Sat Sep  3 18:34:16 2022 daemon.notice netifd: Network device 'eth1' link is down
Sat Sep  3 18:34:16 2022 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sat Sep  3 18:34:16 2022 kern.info kernel: [  328.693292] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Down
Sat Sep  3 18:34:20 2022 daemon.notice netifd: Network device 'eth1' link is up
Sat Sep  3 18:34:20 2022 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Sep  3 18:34:20 2022 kern.info kernel: [  333.025910] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Sat Sep  3 18:34:26 2022 user.notice shadowsocksr: ssrplus redir udp error.restart!
Sat Sep  3 18:34:32 2022 daemon.crit dnsmasq[11961]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:34:32 2022 daemon.crit dnsmasq[11961]: FAILED to start up
Sat Sep  3 18:34:37 2022 daemon.crit dnsmasq[12139]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:34:37 2022 daemon.crit dnsmasq[12139]: FAILED to start up
Sat Sep  3 18:34:42 2022 daemon.crit dnsmasq[12264]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:34:42 2022 daemon.crit dnsmasq[12264]: FAILED to start up
Sat Sep  3 18:34:51 2022 daemon.crit dnsmasq[12696]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:34:51 2022 daemon.crit dnsmasq[12696]: FAILED to start up
Sat Sep  3 18:34:56 2022 daemon.crit dnsmasq[12785]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:34:56 2022 daemon.crit dnsmasq[12785]: FAILED to start up
Sat Sep  3 18:35:00 2022 cron.err crond[8589]: USER root pid 12820 cmd /usr/share/unblockneteasemusic/log_check.sh
Sat Sep  3 18:35:01 2022 daemon.crit dnsmasq[12827]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:35:01 2022 daemon.crit dnsmasq[12827]: FAILED to start up
Sat Sep  3 18:35:06 2022 daemon.crit dnsmasq[12868]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:35:06 2022 daemon.crit dnsmasq[12868]: FAILED to start up
Sat Sep  3 18:35:11 2022 daemon.crit dnsmasq[12905]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:35:11 2022 daemon.crit dnsmasq[12905]: FAILED to start up
Sat Sep  3 18:35:16 2022 daemon.crit dnsmasq[12942]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:35:16 2022 daemon.crit dnsmasq[12942]: FAILED to start up
Sat Sep  3 18:35:16 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Sat Sep  3 18:39:51 2022 user.notice shadowsocksr: ssrplus redir udp error.restart!
Sat Sep  3 18:39:57 2022 daemon.crit dnsmasq[15366]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:39:57 2022 daemon.crit dnsmasq[15366]: FAILED to start up
Sat Sep  3 18:40:00 2022 cron.err crond[8589]: USER root pid 15471 cmd /usr/share/unblockneteasemusic/log_check.sh
Sat Sep  3 18:40:02 2022 daemon.crit dnsmasq[15529]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:02 2022 daemon.crit dnsmasq[15529]: FAILED to start up
Sat Sep  3 18:40:07 2022 daemon.crit dnsmasq[15654]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:07 2022 daemon.crit dnsmasq[15654]: FAILED to start up
Sat Sep  3 18:40:15 2022 daemon.crit dnsmasq[16085]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:15 2022 daemon.crit dnsmasq[16085]: FAILED to start up
Sat Sep  3 18:40:20 2022 daemon.crit dnsmasq[16173]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:20 2022 daemon.crit dnsmasq[16173]: FAILED to start up
Sat Sep  3 18:40:25 2022 daemon.crit dnsmasq[16236]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:25 2022 daemon.crit dnsmasq[16236]: FAILED to start up
Sat Sep  3 18:40:30 2022 daemon.crit dnsmasq[16249]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:30 2022 daemon.crit dnsmasq[16249]: FAILED to start up
Sat Sep  3 18:40:35 2022 daemon.crit dnsmasq[16293]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sat Sep  3 18:40:35 2022 daemon.crit dnsmasq[16293]: FAILED to start up
[... 114 lines cut ...]
[    6.045922] usbcore: registered new interface driver ums-sddr09
[    6.046262] usbcore: registered new interface driver ums-sddr55
[    6.046593] usbcore: registered new interface driver ums-usbat
[    6.051502] xt_time: kernel timezone is -0000
[    6.051942] usbcore: registered new interface driver ax88179_178a
[    6.064020] PPP generic driver version 2.4.2
[    6.064366] PPP MPPE Compression module registered
[    6.064710] NET: Registered PF_PPPOX protocol family
[    6.065517] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    6.065720] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    6.068844] kmodloader: done loading kernel modules from /etc/modules.d/*
[    7.124722] random: jshn: uninitialized urandom read (4 bytes read)
[    7.126473] random: ubusd: uninitialized urandom read (4 bytes read)
[    7.126698] random: ubus: uninitialized urandom read (4 bytes read)
[    9.225049] random: crng init done
[    9.225161] random: 68 urandom warning(s) missed due to ratelimiting
[   10.812249] 8021q: adding VLAN 0 to HW filter on device eth1
[   12.300246] 8021q: adding VLAN 0 to HW filter on device eth2
[   13.609923] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   13.700098] 8021q: adding VLAN 0 to HW filter on device eth0
[   13.700357] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  328.693292] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Down
[  333.025910] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[  949.577285] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Down
[ 1013.101907] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX

rongcenter, 2022-09-03:
估计还是adbyby的问题

yanxin152133, 2022-09-03:
> 估计还是adbyby的问题

好像是我多点了几次重新连接lan口，然后那个dnsmasq出问题了。我恢复出厂设置好了

closed 2022-09-03 by yanxin152133

uosX, 2022-09-20:
> 估计还是adbyby的问题

Wed Sep 21 00:12:22 2022 daemon.crit dnsmasq[19674]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:12:22 2022 daemon.crit dnsmasq[19674]: FAILED to start up
Wed Sep 21 00:12:22 2022 cron.err crond[19715]: crond (busybox 1.35.0) started, log level 8

x86的软路由也会这样子；
补充了以下2条订阅规则
https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
https://easylist-downloads.adblockplus.org/easylistchina.txt

关闭Turbo ACC网络加速的，DNS缓存选项后，问题依旧；
Wed Sep 21 00:15:25 2022 daemon.crit dnsmasq[23302]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:25 2022 daemon.crit dnsmasq[23302]: FAILED to start up
Wed Sep 21 00:15:26 2022 kern.warn kernel: [168501.324993] test 65534
Wed Sep 21 00:15:27 2022 kern.warn kernel: [168502.377737] test 65534
Wed Sep 21 00:15:27 2022 kern.warn kernel: [168503.433491] test 65534
Wed Sep 21 00:15:28 2022 kern.warn kernel: [168503.753960] test 65534
Wed Sep 21 00:15:29 2022 kern.warn kernel: [168504.498719] test 65534
Wed Sep 21 00:15:29 2022 kern.warn kernel: [168505.550004] test 65534
Wed Sep 21 00:15:30 2022 kern.warn kernel: [168505.836201] test 512
Wed Sep 21 00:15:30 2022 daemon.crit dnsmasq[23325]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:30 2022 daemon.crit dnsmasq[23325]: FAILED to start up
Wed Sep 21 00:15:31 2022 kern.warn kernel: [168506.604748] test 65534
Wed Sep 21 00:15:32 2022 kern.warn kernel: [168507.658410] test 65534
Wed Sep 21 00:15:33 2022 kern.warn kernel: [168508.713028] test 65534
Wed Sep 21 00:15:34 2022 kern.warn kernel: [168509.768541] test 65534
Wed Sep 21 00:15:35 2022 daemon.crit dnsmasq[23346]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:35 2022 daemon.crit dnsmasq[23346]: FAILED to start up
Wed Sep 21 00:15:35 2022 kern.warn kernel: [168510.819120] test 65534
Wed Sep 21 00:15:36 2022 kern.warn kernel: [168511.868564] test 65534
Wed Sep 21 00:15:37 2022 kern.warn kernel: [168512.922035] test 65534
Wed Sep 21 00:15:37 2022 kern.warn kernel: [168513.818559] test 65534
Wed Sep 21 00:15:38 2022 kern.warn kernel: [168513.977830] test 65534
Wed Sep 21 00:15:39 2022 kern.warn kernel: [168515.029287] test 65534
Wed Sep 21 00:15:40 2022 daemon.crit dnsmasq[23378]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:40 2022 daemon.crit dnsmasq[23378]: FAILED to start up
Wed Sep 21 00:15:40 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Wed Sep 21 00:15:40 2022 kern.warn kernel: [168516.084328] test 65534
Wed Sep 21 00:15:41 2022 kern.warn kernel: [168517.137872] test 65534
Wed Sep 21 00:15:41 2022 kern.warn kernel: [168517.241848] test 512
Wed Sep 21 00:15:43 2022 kern.warn kernel: [168518.189266] test 65534

yanxin152133, 2022-10-11:
我直接把这个插件给卸载了，反正也没啥用处