coolsnowwolf/lede #10055: 尝试编译5.10内核的R2S的固件报错

bi4tms, 2022-09-04:
日志如下：
touch /home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/r2s.510/lede/target/linux/rockchip/image'
cp -fpR /home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/COPYING /home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/COPYING.linux
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/arch/arm64/boot/dts -I/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/arch/arm64/boot/dts/include -I/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/include/ -undef -D__DTS__  -o /home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-mrkaio-m68s.dtb.tmp /home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/arch/arm64/boot/dts/rockchip/rk3568-mrkaio-m68s.dts
aarch64-openwrt-linux-musl-cpp: error: /home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.138/arch/arm64/boot/dts/rockchip/rk3568-mrkaio-m68s.dts: No such file or directory
aarch64-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
aarch64-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.
Makefile:78: recipe for target '/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-mrkaio-m68s.dtb' failed
make[5]: *** [/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-mrkaio-m68s.dtb] Error 1
make[5]: Leaving directory '/home/r2s.510/lede/target/linux/rockchip/image'
Makefile:24: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/r2s.510/lede/target/linux/rockchip'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/r2s.510/lede/target/linux'
time: target/linux/install#14.18#64.78#83.93
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/r2s.510/lede'
target/Makefile:19: recipe for target '/home/r2s.510/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install' failed
make[1]: *** [/home/r2s.510/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/r2s.510/lede'
/home/r2s.510/lede/include/toplevel.mk:228: recipe for target 'world' failed

zhangguanzhang, 2022-09-04:
https://github.com/coolsnowwolf/lede/issues/9922

```
sed -ri '/^\s*TARGET_DEVICES\s.+?(fastrhino_r66s|firefly_station-p2|ezpro_mrkaio-m68s)/d' target/linux/rockchip/image/armv8.mk
```

bi4tms, 2022-09-04:
> #9922
> 
> ```
> sed -ri '/^\s*TARGET_DEVICES\s.+?(fastrhino_r66s|firefly_station-p2|ezpro_mrkaio-m68s)/d'  \
>   target/linux/rockchip/image/armv8.mk
> ```

看更新日志以为修好这个问题了

closed 2022-09-04 by bi4tms