coolsnowwolf/lede #10056: 疑似VMware文件生成失败，已经经过反复编译测试无法完成

Zaphkito, 2022-09-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：VMware所生成的对应vmdk文件生成报错，原镜像文件生成正常，Hyper-v镜像生成正常，最新5.15内核源码，我看有这问题我就来提一句，虽然我用不到VMware的文件

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：make[5]: [Makefile:164: /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk] Error 1 (ignored)
cp -fpR /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/image/grub2/boot.img /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot/boot/grub/
cp -fpR /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot/boot/grub/core.img
install -d -m0755 /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot/efi/boot
cp -fpR /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot/efi/boot/
PADDING="1" SIGNATURE="cceacd73" GUID="cceacd73-a2f6-46ca-e37d-25fe7bebd600" /home/zaphkito/Source/OpenWrt/scripts/gen_image_generic.sh /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk 16 /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot 480 /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs 256
+ '[' 6 -ne 5 ']'
+ '[' 6 -ne 6 ']'
+ OUTPUT=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk
+ KERNELSIZE=16
+ KERNELDIR=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot
+ ROOTFSSIZE=480
+ ROOTFSIMAGE=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs
+ ALIGN=256
+ rm -f /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk
+ head=16
+ sect=63
++ ptgen -o /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk -h 16 -s 63 -g -p 16m -p 480m -l 256 -S 0xcceacd73 -G cceacd73-a2f6-46ca-e37d-25fe7bebd600
part 0 16384
part 0 491520
+ set 262144 16777216 17039360 503316480
+ KERNELOFFSET=512
+ KERNELSIZE=16777216
+ ROOTFSOFFSET=33280
+ ROOTFSSIZE=983040
+ '[' -n 1 ']'
+ dd if=/dev/zero of=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk bs=512 seek=33280 conv=notrunc count=983040
983040+0 records in
983040+0 records out
503316480 bytes (503 MB, 480 MiB) copied, 6.27075 s, 80.3 MB/s
+ dd if=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs of=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk bs=512 seek=33280 conv=notrunc
256542+1 records in
256542+1 records out
131349659 bytes (131 MB, 125 MiB) copied, 1.13285 s, 116 MB/s
+ '[' -n cceacd73-a2f6-46ca-e37d-25fe7bebd600 ']'
+ '[' -n 1 ']'
+ dd if=/dev/zero of=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk bs=512 seek=1016320 conv=notrunc count=63
63+0 records in
63+0 records out
32256 bytes (32 kB, 32 KiB) copied, 0.000537984 s, 60.0 MB/s
+ mkfs.fat -n kernel -C /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.kernel -S 512 16384
mkfs.fat 4.1 (2017-01-24)
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
+ mcopy -s -i /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.kernel /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot/boot /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.boot/efi ::/
+ dd if=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.kernel of=/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk bs=512 seek=512 conv=notrunc
32768+0 records in
32768+0 records out
16777216 bytes (17 MB, 16 MiB) copied, 0.163721 s, 102 MB/s
+ rm -f /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.kernel
rm -fR /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2
install -d -m0755 /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2
cp -fpR /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/image/grub2/boot.img /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/
cp -fpR /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/core.img
echo '(hd0) /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk' > /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/device.map
/home/zaphkito/Source/OpenWrt/staging_dir/host/bin/grub-bios-setup -m "/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2/device.map" -d "/home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.grub2" -r "hd0,gpt1" /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk
/home/zaphkito/Source/OpenWrt/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..
if command -v qemu-img; then qemu-img convert -f raw -O vmdk /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.new; mv /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk.new /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk; else echo "WARNING: Install qemu-img to create VDI/VMDK images" >&2; exit 1; fi
WARNING: Install qemu-img to create VDI/VMDK images
make[5]: *** [Makefile:164: /home/zaphkito/Source/OpenWrt/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.vmdk] Error 1
make[5]: Leaving directory '/home/zaphkito/Source/OpenWrt/target/linux/x86/image'
make[4]: *** [Makefile:27: install] Error 2
make[4]: Leaving directory '/home/zaphkito/Source/OpenWrt/target/linux/x86'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/zaphkito/Source/OpenWrt/target/linux'
time: target/linux/install#384.30#20.41#214.25
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/zaphkito/Source/OpenWrt'
make[1]: *** [target/Makefile:19: /home/zaphkito/Source/OpenWrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/zaphkito/Source/OpenWrt'
make: *** [/home/zaphkito/Source/OpenWrt/include/toplevel.mk:230: world] Error 2

coolsnowwolf, 2022-09-04:
WARNING: Install qemu-img to create VDI/VMDK images

Zaphkito, 2022-09-04:
> WARNING: Install qemu-img to create VDI/VMDK images

首先声明我没动过qemu-img，我主要勾选了Luci内的application，所以是默认的config有问题，没有qemu-img却勾选了VMware的镜像生成，麻烦修改下默认勾选内容

coolsnowwolf, 2022-09-04:
[image]

zhangguanzhang, 2022-09-04:
```
grep -i vmdk .config
```

lxl6125, 2022-09-09:
> 

意思是你缺依赖

closed 2022-09-19 by aiamadeus