coolsnowwolf/lede #10060: 7621无线还是有问题，

yuzhe1129, 2022-09-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
https://github.com/coolsnowwolf/lede/commit/39c57aca18c0befcfb2321b27ec0bf7c492f8c89
有线修复了，无线还是有问题，昨晚我编译了B70固件，有线正常了，无线还是存在问题

### (2) 路由器型号和固件版本

7621

### (3) 详细日志

A：

MeIsReallyBa, 2022-09-06:
b70 r3g这些用dsa驱动的7621真惨，有线只能跑1g带宽，无线开了nat加速也抽风。

MeIsReallyBa, 2022-09-06:
只是用dsa的7621有问题，而k2p 新三改了交换机驱动的7621一直正常。

看来b70这种用dsa的7621只能跑1g带宽+无线不开nat加速了。

MeIsReallyBa, 2022-09-06:
https://github.com/coolsnowwolf/lede/blob/6bdfad193bf2d37e6f22d250e5b86a67e028cb2e/target/linux/ramips/dts/mt7621.dtsi#L567

抽风的自己把wlan0 wlan1删掉，反正开源mt76速度本来也跑不起来。
无线offload只给闭源无线+交换机驱动的7621留着就是。

nanchuci, 2022-09-06:
听楼上大佬的没错！

yuzhe1129, 2022-09-06:
> https://github.com/coolsnowwolf/lede/blob/6bdfad193bf2d37e6f22d250e5b86a67e028cb2e/target/linux/ramips/dts/mt7621.dtsi#L567
> 
> 抽风的自己把wlan0 wlan1删掉，反正开源mt76速度本来也跑不起来。 无线offload只给闭源无线+交换机驱动的7621留着就是。

我能说我不会吗，

MeIsReallyBa, 2022-09-06:
> > https://github.com/coolsnowwolf/lede/blob/6bdfad193bf2d37e6f22d250e5b86a67e028cb2e/target/linux/ramips/dts/mt7621.dtsi#L567
> > 
> > 抽风的自己把wlan0 wlan1删掉，反正开源mt76速度本来也跑不起来。 无线offload只给闭源无线+交换机驱动的7621留着就是。
> 
> 我能说我不会吗，

位置都发出来了，直接删除就行了啊。

britneyjeanspears, 2022-09-06:
下午更新后7621 wifi的hnat被完全取消。本来正常的k2p无线速度寄了。

closed 2022-09-06 by yuzhe1129