coolsnowwolf/lede #10064: arm-trusted-firmware-rockchip-vendor 无法 Download 完成

online2311, 2022-09-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：arm-trusted-firmware-rockchip-vendor 无法 Download 完成

### (2) 路由器型号和固件版本

A：r68s

### (3) 详细日志

make[2]: Entering directory '/Data/r68s/package/boot/arm-trusted-firmware-rockchip-vendor'
Makefile:80: WARNING: skipping arm-trusted-firmware-rk3568 -- package has no install section
mkdir -p /Data/r68s/dl
SHELL= flock /Data/r68s/tmp/.arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz.flock -c '         /Data/r68s/scripts/download.pl "/Data/r68s/dl" "arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz" "17723ac8f6ec446c759444ee29ba4fe544cebb3785e26d8e10c91c54b9df3f1a" "" || (  /Data/r68s/scripts/dl_github_archive.py --dl-dir="/Data/r68s/dl" --url="https://github.com/rockchip-linux/rkbin.git" --version="b0c100f1a260d807df450019774993c761beb79d" --subdir="arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1" --source="arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz" --hash="17723ac8f6ec446c759444ee29ba4fe544cebb3785e26d8e10c91c54b9df3f1a" || (     echo "Checking out files from the git repository..."; mkdir -p /Data/r68s/tmp/dl && cd /Data/r68s/tmp/dl && rm -rf arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1 && [ \! -d arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1 ] && git clone  https://github.com/rockchip-linux/rkbin.git arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1 && (cd arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1 && git checkout b0c100f1a260d807df450019774993c761beb79d && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1 && git log -1 --format='\''@%ct'\''` && rm -rf arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1/.git &&   tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1 |    xz -zc -7e > /Data/r68s/tmp/dl/arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz && mv /Data/r68s/tmp/dl/arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz /Data/r68s/dl/ && rm -rf arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1; );  )    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.

closed 2022-09-07 by coolsnowwolf

Jayant-Tang, 2022-12-10:
我刚刚遇到了和你一样的问题，也是make download失败。看日志发现是 arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz 这个文件下载不到了，我根据日志进了这个网站：https://sources.cdn.openwrt.org/，这里确实下载不到这个文件。
还好我9月底编译过一次，所以我把我当时那个仓库里的tar.xz文件提取出来放到新的仓库里lede/dl/的位置，就好了。
需要的话可以留个邮箱，我发给你。

rc0903, 2023-11-16:
> 我刚刚遇到了和你一样的问题，也是make download失败。看日志发现是 arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz 这个文件下载不到了，我根据日志进了这个网站：https://sources.cdn.openwrt.org ，这里确实下载不到这个文件。 还好我9月底编译过一次，所以我把我当时那个仓库里的tar.xz文件提取出来放到新的仓库里lede/dl/的位置，就好了。 需要的话可以留个邮箱，我发给你。
可以给我发一个吗？rc0903@live.com

LX2035, 2024-01-07:
> 我刚刚遇到了和你一样的问题，也是make download失败。看日志发现是 arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz 这个文件下载不到了，我根据日志进了这个网站：https://sources.cdn.openwrt.org ，这里确实下载不到这个文件。 还好我9月底编译过一次，所以我把我当时那个仓库里的tar.xz文件提取出来放到新的仓库里lede/dl/的位置，就好了。 需要的话可以留个邮箱，我发给你。

可以发我一个吗，邮箱2071921351@qq.com

ashley3232, 2024-04-09:
> 我刚刚遇到了和你一样的问题，也是make download失败。看日志发现是 arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz 这个文件下载不到了，我根据日志进了这个网站：https://sources.cdn.openwrt.org ，这里确实下载不到这个文件。 还好我9月底编译过一次，所以我把我当时那个仓库里的tar.xz文件提取出来放到新的仓库里lede/dl/的位置，就好了。 需要的话可以留个邮箱，我发给你。

帮我也发一个把642868363@qq.com 谢谢哈！

you-yu324, 2024-09-12:
可以给我吗？邮箱63316828@qq.com

you-yu324, 2024-09-12:
> > 我刚刚遇到了和你一样的问题，也是make download失败。看日志发现是 arm-trusted-firmware-rockchip-vendor-2022-08-01-b0c100f1.tar.xz 这个文件下载不到了，我根据日志进了这个网站：https://sources.cdn.openwrt.org ，这里确实下载不到这个文件。 还好我9月底编译过一次，所以我把我当时那个仓库里的tar.xz文件提取出来放到新的仓库里lede/dl/的位置，就好了。 需要的话可以留个邮箱，我发给你。
> 
> 可以发一个给我吗？邮箱63316828@qq.com 谢谢！