coolsnowwolf/lede #10067: 编译测试版内核X86时报错

kunger97, 2022-09-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我希望编译5.19内核的x86固件，再menu config中的 global build setting 中勾选了use testing kernel version后编译报错

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

make[5]: Entering directory '/home/kunger/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19.7'
  CC [M]  /home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module/xt_RTPENGINE.o
/home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module/xt_RTPENGINE.c: In function 'proc_status':
/home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module/xt_RTPENGINE.c:1114:34: error: implicit declaration of function 'PDE_DATA'; did you mean 'NODE_DATA'? [-Werror=implicit-function-declaration]
  id = (uint32_t) (unsigned long) PDE_DATA(inode);
                                  ^~~~~~~~
                                  NODE_DATA
/home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module/xt_RTPENGINE.c: In function 'proc_control_read_write.isra.40':
/home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module/xt_RTPENGINE.c:3396:1: warning: the frame size of 1080 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:249: /home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module/xt_RTPENGINE.o] Error 1
make[5]: *** [Makefile:1847: /home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module] Error 2
make[5]: Leaving directory '/home/kunger/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19.7'
make[4]: *** [Makefile:34: modules] Error 2
make[4]: Leaving directory '/home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/kernel-module'
make[3]: *** [Makefile:287: /home/kunger/lede/build_dir/target-x86_64_musl/rtpengine-no-transcode/rtpengine-mr9.5.3.2/.built] Error 2
make[3]: Leaving directory '/home/kunger/lede/feeds/telephony/net/rtpengine'

guoshh1978, 2022-09-08:
我都是直接修改lede/.config和lede/target/linux/x86/Makefile来编译
.config

编译过程中遇到了cannot find -liconv的报错，然后修改了lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/prog/sensors/Module.mk里面的第40行：
 40 LIBICONV := $(shell if /sbin/ldconfig -p | grep -q '/libiconv\.so$$' ; then echo; else echo; fi)
 41 #LIBICONV := $(shell if /sbin/ldconfig -p | grep -q '/libiconv\.so$$' ; then echo \-liconv; else echo; fi)
然后就能够成功编译了

我重新编译安装了本地的libiconv这个报错了还是存在，搞不明白是怎么回事，反正能编译成功，就懒得再折腾了。

Jejz168, 2022-09-26:
.config中CONFIG_TESTING_KERNEL=y

closed 2024-01-27 by kunger97