coolsnowwolf/lede #10071: 编译5.15内核x86固件报错

bi4tms, 2022-09-09:
拉取最新源码后编译报错，日志如下：
Applying /home/x86/lede/target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch using plaintext:
patching file include/linux/if_bridge.h
patching file include/linux/skbuff.h
patching file include/linux/timer.h
patching file include/net/netfilter/nf_conntrack_ecache.h
patching file net/Kconfig
patching file net/bridge/br_if.c
patching file net/core/dev.c
Hunk #3 FAILED at 5275.
1 out of 4 hunks FAILED -- saving rejects to file net/core/dev.c.rej
patching file net/netfilter/nf_conntrack_ecache.c
Patch failed!  Please fix /home/x86/lede/target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch!
Makefile:27: recipe for target '/home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.64/.prepared_23e45094cc468f6a23dc110bf8ff4a74' failed
make[4]: *** [/home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.64/.prepared_23e45094cc468f6a23dc110bf8ff4a74] Error 1
make[4]: Leaving directory '/home/x86/lede/target/linux/x86'
Makefile:11: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/x86/lede/target/linux'
time: target/linux/compile#18.78#89.46#165.05
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 1
make[2]: Leaving directory '/home/x86/lede'
target/Makefile:18: recipe for target '/home/x86/lede/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
make[1]: *** [/home/x86/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/x86/lede'
/home/x86/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

Vohrt, 2022-09-09:
same issue here

z1790, 2022-09-09:
Applying /home/haowan/lede/target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch using plaintext: 
patching file include/linux/if_bridge.h
patching file include/linux/skbuff.h
patching file include/linux/timer.h
patching file include/net/netfilter/nf_conntrack_ecache.h
patching file net/Kconfig
patching file net/bridge/br_if.c
patching file net/core/dev.c
Hunk #3 FAILED at 5275.
1 out of 4 hunks FAILED -- saving rejects to file net/core/dev.c.rej
patching file net/netfilter/nf_conntrack_ecache.c
Patch failed!  Please fix /home/haowan/lede/target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch!
make[3]: *** [Makefile:123: /home/haowan/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/linux-5.15.64/.prepared] Error 1
make[3]: Leaving directory '/home/haowan/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#10.66#9.22#23.78
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/haowan/lede'
make[1]: *** [toolchain/Makefile:93: /home/haowan/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/haowan/lede'
make: *** [/home/haowan/lede/include/toplevel.mk:230: world] Error 2

一样 报错

SkyTeamWater, 2022-09-10:
Patch failed!  Please fix /home/ubuntu/lede/target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch!

breakings, 2022-09-10:
等大雕更新，目前解决方案是去 hack-5.19 复制同名文件替换

cike-567, 2022-09-10:
我也出现了同样的报错

adamgonglj, 2022-09-10:
R4S也是同样的错误

lovehackintosh, 2022-09-10:
#10074

commit 64b3d5a0 (2022-09-10): kernel: fixes build error and refresh 5.15 patches (#10074)

Manually rebased:
  target/linux/generic/hack-5.15/953-net-patch-linux-kernel-to-support-shortcut-fe.patch

All other patches automatically rebased.

Build system: x86_64
Fixes: #10071
closed 2022-09-10 by aiamadeus