coolsnowwolf/lede #10075: rtw88模块加载失败

Suyuan406, 2022-09-10:
## 1.关于你要提交的问题
rtw88模块加载失败，提示不能加载rtw88/rtw8821c_fw.bin
Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
添加了kmod-rtw88，插入rtl8821cu的无线网卡无法启动，提示不能加载bin文件。

### (1) 具体问题

A：加载rtw88（rtl8821cu无线网卡）失败，提示不能加载rtw88/rtw8821c_fw.bin

### (2) 路由器型号和固件版本

A：R68S最新版本

### (3) 详细日志

Sat Sep 10 16:29:14 2022 kern.warn kernel: [    9.579783] rtw_8821cu 1-1:1.0: Direct firmware load for rtw88/rtw8821c_fw.bin failed with error -2
Sat Sep 10 16:29:14 2022 kern.warn kernel: [    9.580591] rtw_8821cu 1-1:1.0: Falling back to sysfs fallback for: rtw88/rtw8821c_fw.bin
Sat Sep 10 16:29:14 2022 kern.err kernel: [   70.623239] rtw_8821cu 1-1:1.0: failed to request firmware
Sat Sep 10 16:29:14 2022 kern.err kernel: [   70.623772] rtw_8821cu 1-1:1.0: failed to load firmware
Sat Sep 10 16:29:14 2022 kern.err kernel: [   70.624237] rtw_8821cu 1-1:1.0: failed to setup chip efuse info
Sat Sep 10 16:29:14 2022 kern.err kernel: [   70.624758] rtw_8821cu 1-1:1.0: failed to setup chip information
Sat Sep 10 16:29:14 2022 kern.warn kernel: [   70.625447] rtw_8821cu: probe of 1-1:1.0 failed with error -22
Sat Sep 10 16:29:14 2022 kern.info kernel: [   70.626102] usbcore: registered new interface driver rtw_8821cu
Sat Sep 10 16:29:14 2022 user.info kernel: [   70.634964] kmodloader: done loading kernel modules from /etc/modules.d/*

Suyuan406, 2022-09-10:
Sat Aug  5 17:01:35 2017 daemon.err procd: Could not find firmware /lib/firmware/rtw88/rtw8821c_fw.bin: No such file or directory
Sat Aug  5 17:01:35 2017 daemon.err procd: Failed to write to /sys//devices/platform/fcc00000.usb/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/firmware/rtw88!rtw8821c_fw.bin/loading: Bad file descriptor

Suyuan406, 2022-09-10:
[log.txt](https://github.com/coolsnowwolf/lede/files/9539880/log.txt)
完整日志

aiamadeus, 2022-09-10:
你要选上 rtl8821ce-firmware

Suyuan406, 2022-09-11:
感谢，现在已经有无线选项了，但是我不能搜索到无线信号
内核日志提示这个 
[    9.707450] rtw_8821cu 1-1:1.0: Firmware version 24.11.0, H2C version 12
[    9.775562] usbcore: registered new interface driver rtw_8821cu
[    9.782126] kmodloader: done loading kernel modules from /etc/modules.d/*
[   34.349498] EXT4-fs (mmcblk0p1): warning: mounting unchecked fs, running e2fsck is recommended
[   34.351370] EXT4-fs (mmcblk0p1): mounted filesystem without journal. Opts: . Quota mode: none.
[   40.430903] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-1:00] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   40.433340] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   40.444486] dwmac4: Master AXI performs any burst length
[   40.445007] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   40.445695] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   40.446827] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii-id link mode
[   40.480305] br-lan: port 1(eth0) entered blocking state
[   40.480801] br-lan: port 1(eth0) entered disabled state
[   40.481480] device eth0 entered promiscuous mode
[   40.492534] eth2: 0xffff800009750000, 92:1b:81:0f:68:90, IRQ 50
[   40.539687] br-lan: port 2(eth2) entered blocking state
[   40.540172] br-lan: port 2(eth2) entered disabled state
[   40.540788] device eth2 entered promiscuous mode
[   40.543669] eth3: 0xffff8000094a0000, 92:1b:81:0f:68:90, IRQ 51
[   40.590499] br-lan: port 3(eth3) entered blocking state
[   40.591034] br-lan: port 3(eth3) entered disabled state
[   40.591648] device eth3 entered promiscuous mode
[   40.686946] rk_gmac-dwmac fe2a0000.ethernet eth1: PHY [stmmac-0:00] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   40.688617] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[   40.698874] dwmac4: Master AXI performs any burst length
[   40.699379] rk_gmac-dwmac fe2a0000.ethernet eth1: No Safety Features support found
[   40.700061] rk_gmac-dwmac fe2a0000.ethernet eth1: IEEE 1588-2008 Advanced Timestamp supported
[   40.701177] rk_gmac-dwmac fe2a0000.ethernet eth1: configuring for phy/rgmii-id link mode
[   42.651225] rtw_8821cu 1-1:1.0: start vif e0:e1:a9:1c:36:8b on port 0
[   42.695731] br-lan: port 4(wlan0) entered blocking state
[   42.696238] br-lan: port 4(wlan0) entered disabled state
[   42.696891] device wlan0 entered promiscuous mode
[   44.080423] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   44.081160] br-lan: port 4(wlan0) entered blocking state
[   44.081643] br-lan: port 4(wlan0) entered forwarding state
[   44.082500] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   45.603789] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   45.604629] br-lan: port 1(eth0) entered blocking state
[   45.605106] br-lan: port 1(eth0) entered forwarding state
[   49.625955] rtw_8821cu 1-1:1.0: error beacon valid
[   49.626438] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   55.694388] rtw_8821cu 1-1:1.0: error beacon valid
[   55.694880] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   61.761507] rtw_8821cu 1-1:1.0: error beacon valid
[   61.761988] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   66.739593] overlayfs: filesystem on '/opt/docker/check-overlayfs-support4142148303/upper' not supported as upperdir
[   66.755563] overlayfs: filesystem on '/opt/docker/check-overlayfs-support3609877041/upper' not supported as upperdir
[   67.831509] rtw_8821cu 1-1:1.0: error beacon valid
[   67.831976] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   73.928348] rtw_8821cu 1-1:1.0: error beacon valid
[   73.928829] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   80.021958] rtw_8821cu 1-1:1.0: error beacon valid
[   80.022441] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   86.115842] rtw_8821cu 1-1:1.0: error beacon valid
[   86.116325] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   92.208023] rtw_8821cu 1-1:1.0: error beacon valid
[   92.208503] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[   98.305031] rtw_8821cu 1-1:1.0: error beacon valid
[   98.305513] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  104.399470] rtw_8821cu 1-1:1.0: error beacon valid
[  104.399951] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  110.489022] rtw_8821cu 1-1:1.0: error beacon valid
[  110.489501] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  116.584849] rtw_8821cu 1-1:1.0: error beacon valid
[  116.585333] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  122.677355] rtw_8821cu 1-1:1.0: error beacon valid
[  122.677834] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  128.772218] rtw_8821cu 1-1:1.0: error beacon valid
[  128.772696] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  134.864354] rtw_8821cu 1-1:1.0: error beacon valid
[  134.864836] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  140.932472] rtw_8821cu 1-1:1.0: error beacon valid
[  140.932951] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  147.027103] rtw_8821cu 1-1:1.0: error beacon valid
[  147.027585] rtw_8821cu 1-1:1.0: failed to download drv rsvd page
[  151.252324] device wlan0 left promiscuous mode
[  151.252857] br-lan: port 4(wlan0) entered disabled state
[  151.320502] rtw_8821cu 1-1:1.0: stop vif e0:e1:a9:1c:36:8b on port 0
[  151.719126] rtw_8821cu 1-1:1.0: error beacon valid
[  151.719603] rtw_8821cu 1-1:1.0: failed to download rsvd page
[  151.720256] rtw_8821cu 1-1:1.0: failed to download firmware
[  187.528202] rtw_8821cu 1-1:1.0: error beacon valid
[  187.528669] rtw_8821cu 1-1:1.0: failed to download rsvd page
[  187.529308] rtw_8821cu 1-1:1.0: failed to download firmware

zhangsd20, 2023-03-02:
您好,

我编译的固件为什么加载后插rtl8821cu出现下面打印.也无法驱动起来?

[   25.952736] usb 1-1.2: new high-speed USB device number 3 using ehci-platform
[   26.415237] rtw_8821cu 1-1.2:1.2: Firmware version 24.11.0, H2C version 12
[   26.657188] rtw_8821cu 1-1.2:1.2: rfe 38 isn't supported
[   26.677542] rtw_8821cu 1-1.2:1.2: failed to setup chip efuse info
[   26.683930] rtw_8821cu 1-1.2:1.2: failed to setup chip information
[   26.902732] usb 1-1.3: new full-speed USB device number 4 using ehci-platform
[   27.372732] usb 1-1.4: new full-speed USB device number 5 using ehci-platform

coolsnowwolf, 2026-09-05:
核验当前 master：原 issue 报告的 `rtw88/rtw8821c_fw.bin` 缺失已修复。`package/kernel/rtw88-usb/Makefile` 的 `rtl8821cu` 已依赖 `rtl8821ce-firmware`，该固件包安装 `rtw88/rtw8821c_fw.bin`。issue 后续日志已能加载 Firmware version 24.11.0；剩余 `rfe 38 isn't supported` 属于特定 USB 网卡硬件 RFE 兼容性，不能通过补装固件解决。故按已修复原始问题关闭。

closed 2026-09-05 by coolsnowwolf