coolsnowwolf/lede #10076: 刚刚本地编译出错

qqsir-dev, 2022-09-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency samba4-libs for samba4-server
 * pkg_hash_fetch_best_installation_candidate: Packages for samba4-server found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 *      samba4-libs
 * opkg_install_cmd: Cannot install package autosamba.
 * check_conflicts_for: The following packages conflict with samba4-server:
 * check_conflicts_for:         samba36-server *
 * opkg_install_cmd: Cannot install package samba4-server.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/user/openwrt'
make[1]: *** [package/Makefile:111: /home/user/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:230: world] Error 2

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

aiamadeus, 2022-09-10:
rm .config 重新选

qqsir-dev, 2022-09-10:
好的谢谢

closed 2022-09-10 by qqsir-dev