coolsnowwolf/lede #10087: 编译报错

achinese888, 2022-09-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错 Your BIOS Boot Partition is under 1 MiB, please increase its size..

### (2) 路由器型号和固件版本

A：x86 

### (3) 详细日志

A：+ [ -n 1 ]
+ dd if=/dev/zero of=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=66048 conv=notrunc count=12582912
345754+1 records in
345754+1 records out
177026206 bytes (177 MB, 169 MiB) copied, 0.789784 s, 224 MB/s
dd if=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new bs=6442450944 conv=sync
dd: memory exhausted by input buffer of size 6442450944 bytes (6.0 GiB)
make[5]: *** [Makefile:164: /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz] Error 1
make[5]: *** Waiting for unfinished jobs....
12582912+0 records in
12582912+0 records out
6442450944 bytes (6.4 GB, 6.0 GiB) copied, 42.6912 s, 151 MB/s
+ dd if=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs of=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=66048 conv=notrunc
345754+1 records in
345754+1 records out
177026206 bytes (177 MB, 169 MiB) copied, 8.84496 s, 20.0 MB/s
+ [ -n 0d4b6a54-929d-7448-0f76-2a948245c200 ]
+ [ -n 1 ]
+ dd if=/dev/zero of=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=12648960 conv=notrunc count=63
63+0 records in
63+0 records out
32256 bytes (32 kB, 32 KiB) copied, 0.00018143 s, 178 MB/s
+ mkfs.fat -n kernel -C /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel -S 512 32768
mkfs.fat 4.1 (2017-01-24)
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
+ mcopy -s -i /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/efi ::/
+ dd if=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel of=/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=512 conv=notrunc
65536+0 records in
65536+0 records out
33554432 bytes (34 MB, 32 MiB) copied, 0.0907887 s, 370 MB/s
+ rm -f /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel
rm -fR /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2
install -d -m0755 /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2
cp -fpR /home/zym/lede/staging_dir/target-x86_64_musl/image/grub2/boot.img /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/
cp -fpR /home/zym/lede/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/core.img
echo '(hd0) /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz' > /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/device.map
/home/zym/lede/staging_dir/host/bin/grub-bios-setup -m "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/device.map" -d "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2" -r "hd0,gpt1" /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz
/home/zym/lede/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..
gzip -f -9n -c /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz  > /home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.new
[ ! -s "/home/zym/lede/key-build" -o ! -s "/home/zym/lede/key-build.ucert" -o ! -s "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" ] || { cp "/home/zym/lede/key-build.ucert" "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.ucert" ; usign -S -m "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" -s "/home/zym/lede/key-build" -x "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.sig" ; ucert -A -c "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.ucert" -x "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.sig" ; fwtool -S "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.ucert" "/home/zym/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz" ; }
make[5]: Leaving directory '/home/zym/lede/target/linux/x86/image'
make[4]: *** [Makefile:27: install] Error 2
make[4]: Leaving directory '/home/zym/lede/target/linux/x86'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/zym/lede/target/linux'
time: target/linux/install#489.07#77.20#349.03
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/zym/lede'
make[1]: *** [target/Makefile:19: /home/zym/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/zym/lede'
make: *** [/home/zym/lede/include/toplevel.mk:230：world] 错误 2

1054009064, 2022-09-11:
```
dd: memory exhausted by input buffer of size 6442450944 bytes (6.0 GiB)
```

coolsnowwolf, 2022-09-11:
镜像太大，内存不足了

achinese888, 2022-09-12:
ok，谢谢

closed 2022-09-12 by achinese888