coolsnowwolf/lede #10091: 最新分支编译 samba 报错

BenjaminX, 2022-09-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
基于最新分支编译 ARM 版本，samba 没有编译出 ARM 打包固件报错。

### (1) 具体问题

A：看起来是编译升级后对应架构版本没有输出或者路径错误了。

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志

A：Configuring luci-i18n-samba-en.
[4626](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4627)
 * pkg_hash_check_unresolved: cannot find dependency samba4-libs for samba4-server
[4627](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4628)
Configuring luci-i18n-mwan3-zh-cn.
[4628](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4629)
 * pkg_hash_fetch_best_installation_candidate: Packages for samba4-server found, but incompatible with the architectures configured
[4629](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4630)
Configuring ddns-scripts_nsupdate.
[4630](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4631)
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
[4631](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4632)
Configuring default-settings.
[4632](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4633)
 * 	samba4-libs
[4633](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4634)
Configuring ddns-scripts_route53-v1.
[4634](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4635)
 * opkg_install_cmd: Cannot install package autosamba.
[4635](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4636)
 * check_conflicts_for: The following packages conflict with samba4-server:
[4636](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4637)
Configuring luci-i18n-samba4-zh-cn.
[4637](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4638)
 * check_conflicts_for: 	samba36-server * 
[4638](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4639)
Configuring htop.
[4639](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4640)
 * opkg_install_cmd: Cannot install package samba4-server.
[4640](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4641)
Configuring kmod-crypto-gcm.
[4641](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4642)
make[2]: *** [package/Makefile:70: package/install] Error 255
[4642](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4643)
Configuring luci-i18n-base-zh-cn.
[4643](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4644)
Configuring kmod-crypto-cmac.
[4644](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4645)
Configuring kmod-usb-ohci-pci.
[4645](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4646)
Configuring fdisk.
[4646](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4647)
Configuring kmod-usb-uhci.
[4647](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4648)
Configuring gzip.
[4648](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4649)
Configuring luci-i18n-upnp-zh-cn.
[4649](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4650)
Configuring luci-i18n-base-en.
[4650](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4651)
Configuring kmod-button-hotplug.
[4651](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4652)
Configuring luci-i18n-bonding-en.
[4652](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4653)
Configuring kmod-fs-squashfs.
[4653](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4654)
Configuring ddns-scripts_freedns_42_pl.
[4654](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4655)
Configuring luci-i18n-firewall-zh-cn.
[4655](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4656)
Configuring ppp-mod-pppoe.
[4656](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4657)
Configuring luci-i18n-nlbwmon-zh-cn.
[4657](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4658)
make[2]: Leaving directory '/workdir/openwrt'
[4658](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4659)
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
[4659](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4660)
make[1]: Leaving directory '/workdir/openwrt'
[4660](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4661)
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
[4661](https://github.com/BenjaminX/OpenWrtR4S/runs/8291551853?check_suite_focus=true#step:11:4662)
Error: Process completed with exit code 2.

1054009064, 2022-09-12:
重新生成 .config

BenjaminX, 2022-09-12:
> 重新生成 .config

不是 config 文件问题，github actions 编译的每次 config 都是新的

1054009064, 2022-09-12:
我看你 .config 固定了autosmaba 啊，现在源码改动 defconfig 就必定自动选 samba4，然后就一堆冲突了，你的编译 log 还不清楚吗？

1054009064, 2022-09-12:
https://github.com/coolsnowwolf/lede/commit/20550724175da39ef2909b504226d4495c756456

BenjaminX, 2022-09-12:
> 我看你 .config 固定了autosmaba 啊，现在源码改动 defconfig 就必定自动选 samba4，然后就一堆冲突了，你的编译 log 还不清楚吗？

好吧，那就是说 autosamba 可以删掉了？

Plutonium141, 2022-09-12:
看着是samba4-server缺一个samba4-libs的依赖 还有samba4-server与samba36-server冲突
只要把samba4-libs添加到package目录里
然后取消samba36-server的相关选项大概就行了

Nothend, 2022-09-12:
> > 重新生成 .config
> 
> 不是 config 文件问题，github actions 编译的每次 config 都是新的

你的config怎么每次都是最新的呢？我的云编译也报这个错误，但是我的config 文件是固定的，看样子要重新用本地机重新生成一个，然后拷贝到云编译里

Ginua8292, 2022-09-12:
> > 我看你 .config 固定了autosmaba 啊，现在源码改动 defconfig 就必定自动选 samba4，然后就一堆冲突了，你的编译 log 还不清楚吗？
> 
> 好吧，那就是说 autosamba 可以删掉了？

现在autosamba固定了samba4，我现在没选autosamba了！samba4我刷固件后共享不成功没有目录！所以我抛弃了autosamba

Nothend, 2022-09-12:
[image]

我看网上说要这样，但是我没成功

Ginua8292, 2022-09-12:
> [image]
> 
> 我看网上说要这样，但是我没成功

是共享不成还是编译失败？

liwenjie119, 2022-09-12:
取消勾选luci-app-samba，autosamba保留，编译应该能通过

Nothend, 2022-09-12:
> 取消勾选luci-app-samba，autosamba保留，编译应该能通过

好的我来试试

ZGNCZMY, 2022-09-12:
> > 取消勾选luci-app-samba，autosamba保留，编译应该能通过
> 
> 云编译中我两个都取消了，还是不行，明天在本地试试

还要把Samba36取消了

nuo12138, 2022-09-13:
> > > 我看你 .config 固定了autosmaba 啊，现在源码改动 defconfig 就必定自动选 samba4，然后就一堆冲突了，你的编译 log 还不清楚吗？
> > 
> > 
> > 好吧，那就是说 autosamba 可以删掉了？
> 
> 现在autosamba固定了samba4，我现在没选autosamba了！samba4我刷固件后共享不成功没有目录！所以我抛弃了autosamba

samba4第一次启用需要重启一次路由器才能生效

nuo12138, 2022-09-13:
> > 取消勾选luci-app-samba，autosamba保留，编译应该能通过
> 
> 云编译中我两个都取消了，还是不行，明天在本地试试

> > > > 我看你 .config 固定了autosmaba 啊，现在源码改动 defconfig 就必定自动选 samba4，然后就一堆冲突了，你的编译 log 还不清楚吗？
> > > 
> > > 
> > > 好吧，那就是说 autosamba 可以删掉了？
> > 
> > 
> > 现在autosamba固定了samba4，我现在没选autosamba了！samba4我刷固件后共享不成功没有目录！所以我抛弃了autosamba
> 
> 那我把samba都去掉可以吧？不用了。。

> > 取消勾选luci-app-samba，autosamba保留，编译应该能通过
> 
> 云编译中我两个都取消了，还是不行，明天在本地试试

如果是ARM路由的话建议重新拉库编译，因为GCC也改了，我的R4S固件也是重新拉库才能编译成功

BenjaminX, 2022-09-13:
autosamba 和 luci-app-samba 设置为 n 是可以编译成功的

Meltdown0403, 2022-09-13:
看了各位的发言，最新分支编译无论怎样设置都是不能成功的。
尝试了如下两个平台：
（1）X86、64
（2）Rockchip R2S
操作系统Ubuntu Server 22.04.1LTS
git 拉取最新分支代码
如下3种配置都不能正常编译，全部报错[.../toplevel.mk:230: world] Error 2
（1）禁用autosamba+取消luci-app-samba和luci-app-samba4+Network中取消samba36和samba4的所以相关依赖（所有）
（2）禁用autosamba+取消luci-app-samba+勾选luci-app-samba4+Network中取消samba36，勾选samba4的相关依赖（默认）
（3）禁用autosamba+勾选luci-app-samba+取消luci-app-samba4+Network中勾选samba36，取消samba4的相关依赖（所有）

参考（3）的场合：
make[3]: Leaving directory '/home/zhaojun/lede/package/network/services/samba36'
time: package/network/services/samba36/compile#319.86#73.06#659.34
make[2]: Leaving directory '/home/zhaojun/lede'
make[1]: *** [package/Makefile:110: /home/zhaojun/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhaojun/lede'
make: *** [/home/zhaojun/lede/include/toplevel.mk:230: world] Error 2

等待代码修正中。。。

Nothend, 2022-09-13:
> 看了各位的发言，最新分支编译无论怎样设置都是不能成功的。 尝试了如下两个平台： （1）X86、64 （2）Rockchip R2S 操作系统Ubuntu Server 22.04.1LTS git 拉取最新分支代码 如下3种配置都不能正常编译，全部报错[.../toplevel.mk:230: world] Error 2 [...]

感谢，我安装你上面的3中配置试了，又报了docker的错

`make[3]: *** [Makefile:186: /workdir/openwrt/build_dir/target-x86_64_musl/dockerd-20.10.18/.prepared_d5d126c49d359cac770342276dac244a_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/dockerd'
time: package/feeds/packages/dockerd/compile#0.61#0.46#0.86
    ERROR: package/feeds/packages/dockerd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/dockerd/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2`

IDeLoveYou, 2022-09-13:
插个眼，修复了踢我一下

Nothend, 2022-09-13:
代码修复了，云编译和本地都编译成功了

Meltdown0403, 2022-09-13:
> 代码修复了，云编译和本地都编译成功了
感谢lean大佬，拉取最新分支，编译成功了。
r5054-cebb316d0

Ginua8292, 2022-09-13:
> 代码修复了，云编译和本地都编译成功了

有不行了！autosamba同时绑定了autosamba和samba4-server而且这次取消不了！

uniessey, 2022-09-14:
去年3月份开始，我就把autosamba拷贝到到openwrt的主线上了，并且当时就改成samba4了，没发现任何问题啊。

Meltdown0403, 2022-09-14:
> > 代码修复了，云编译和本地都编译成功了
> 
> 有不行了！autosamba同时绑定了autosamba和samba4-server而且这次取消不了！

我刚更新后编译成功了。
r5064-4a2ff4b1c

IDeLoveYou, 2022-09-14:
> 

我还是编译失败

closed 2022-09-14 by aiamadeus