coolsnowwolf/lede #10095: ESXi下虚拟机直通8125网卡不定时断网

pisheng, 2022-09-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ESXi下虚拟机直通8125网卡不定时断网，断网后通过esxi界面能够进入操作openwrt，但是ssh和web界面均无法进入，重启虚拟机后恢复，但是短则几个小时，长则一天左右就会在此断网。
取消8125网卡直通，采用虚拟网卡无此问题，因此初步判断是8125的问题。

### (2) 路由器型号和固件版本

A：esxi 7 虚拟openwrt 最新版本自编译（5.15.63），直通8125网卡

### (3) 详细日志

A：日志最后虽然现实8125 up，但是实际上仍然是断网状态。

kern.warn kernel: [ 9702.035597] ------------[ cut here ]------------
kern.info kernel: [ 9702.036194] NETDEV WATCHDOG: eth0 (r8125): transmit queue 1 timed out
kern.warn kernel: [ 9702.036794] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x29b/0x2a0
kern.warn kernel: [ 9702.037382] Modules linked in: shortcut_fe_cm xt_FULLCONENAT pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT wireguard pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack libchacha20poly1305 ipt_REJECT curve25519_x86_64 chacha_x86_64 ax88179_178a zstd xt_time xt_tcpudp xt_socket xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8125 poly1305_x86_64 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6
kern.warn kernel: [ 9702.037425]  nf_socket_ipv4 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mlx5_core mlx4_en mlx4_core mdev macvlan lzo libcurve25519_generic libchacha kvm iptable_raw iptable_mangle iptable_filter ip_tables igc iavf i915 i40e forcedeth e1000e crc_ccitt br_netfilter bnx2x bnx2 alx 8139too 8139cp configs evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfs bonding ixgbe igbvf e1000 amd_xgbe mdio_devres dummy ip6_udp_tunnel udp_tunnel mdio udp_diag tcp_diag raw_diag inet_diag veth tun ntfs3 ntfs lockd sunrpc grace dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 ena shortcut_fe_ipv6
kern.warn kernel: [ 9702.043462]  shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_ssse3 sha1_ssse3 sha1_generic jitterentropy_rng drbg hmac authenc arc4 crypto_acompress nls_iso8859_1 nls_cp437 vfat fat uas sdhci_pltfm igb exfat btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug ptp pps_core mii libphy cbc encrypted_keys trusted tpm oid_registry asn1_encoder asn1_decoder
kern.warn kernel: [ 9702.055023] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.63 #0
kern.warn kernel: [ 9702.056081] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.18452719.B64.2108091906 08/09/2021
kern.warn kernel: [ 9702.057985] RIP: 0010:dev_watchdog+0x29b/0x2a0
kern.warn kernel: [ 9702.058891] Code: ff ff ff 4c 8b 7d d0 c6 05 35 13 b9 00 01 4c 89 ff e8 a9 30 fb ff 89 d9 4c 89 fe 48 c7 c7 88 af 06 82 48 89 c2 e8 bc c4 13 00 <0f> 0b eb b3 90 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49
kern.warn kernel: [ 9702.060957] RSP: 0018:ffffc90000003e88 EFLAGS: 00010292
kern.warn kernel: [ 9702.061917] RAX: 0000000000000039 RBX: 0000000000000001 RCX: 0000000000000027
kern.warn kernel: [ 9702.062919] RDX: 0000000000000027 RSI: ffffc90000003ce0 RDI: ffff88803ec1c588
kern.warn kernel: [ 9702.063932] RBP: ffffc90000003ec0 R08: 0000000000000000 R09: c0000000ffffefff
kern.warn kernel: [ 9702.064930] R10: 0000000000000001 R11: ffffc90000003c88 R12: ffff888004958440
kern.warn kernel: [ 9702.065922] R13: ffff888006e80478 R14: 0000000000000004 R15: ffff888006e80000
kern.warn kernel: [ 9702.066914] FS:  0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
kern.warn kernel: [ 9702.067946] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kern.warn kernel: [ 9702.068911] CR2: 00007fe2b2f5c000 CR3: 00000000072e6000 CR4: 0000000000350ef0
kern.warn kernel: [ 9702.069978] Call Trace:
kern.warn kernel: [ 9702.070881]  <IRQ>
kern.warn kernel: [ 9702.071781]  ? pfifo_fast_enqueue+0x150/0x150
kern.warn kernel: [ 9702.072715]  call_timer_fn.isra.39+0x13/0x80
kern.warn kernel: [ 9702.073633]  run_timer_softirq+0x2a3/0x2c0
kern.warn kernel: [ 9702.074561]  ? clockevents_program_event+0x8c/0xf0
kern.warn kernel: [ 9702.075521]  ? tick_program_event+0x3f/0x70
kern.warn kernel: [ 9702.076440]  __do_softirq+0xc6/0x209
kern.warn kernel: [ 9702.077345]  irq_exit_rcu+0x52/0x70
kern.warn kernel: [ 9702.078241]  sysvec_apic_timer_interrupt+0x7c/0xa0
kern.warn kernel: [ 9702.079174]  </IRQ>
kern.warn kernel: [ 9702.080038]  <TASK>
kern.warn kernel: [ 9702.080886]  asm_sysvec_apic_timer_interrupt+0x1a/0x20
kern.warn kernel: [ 9702.081793] RIP: 0010:native_safe_halt+0x17/0x20
kern.warn kernel: [ 9702.082665] Code: 02 20 48 8b 00 a8 08 75 bf e9 73 ff ff ff cc cc cc cc cc 55 8b 05 b1 45 bb 00 48 89 e5 85 c0 7e 07 0f 00 2d db f2 50 00 fb f4 <5d> c3 cc cc cc cc 0f 1f 00 55 8b 05 91 45 bb 00 48 89 e5 85 c0 7e
kern.warn kernel: [ 9702.084603] RSP: 0018:ffffffff82203db8 EFLAGS: 00000242
kern.warn kernel: [ 9702.085469] RAX: 0000000000000001 RBX: ffff8880013e6864 RCX: ffff88803ec00000
kern.warn kernel: [ 9702.086358] RDX: 0000000000000001 RSI: ffffffff822d01a0 RDI: ffff8880013e6864
kern.warn kernel: [ 9702.087233] RBP: ffffffff82203db8 R08: ffff8880013e6800 R09: 0000000000000000
kern.warn kernel: [ 9702.088102] R10: 000000000000011c R11: 00000000fa83b2da R12: 0000000000000001
kern.warn kernel: [ 9702.088941] R13: 0000000000000001 R14: ffffffff822d01a0 R15: ffffffff822d0208
kern.warn kernel: [ 9702.089783]  acpi_idle_do_entry+0x50/0x60
kern.warn kernel: [ 9702.090538]  acpi_idle_enter+0xbd/0xd0
kern.warn kernel: [ 9702.091271]  cpuidle_enter_state+0x7b/0x2e0
kern.warn kernel: [ 9702.092012]  cpuidle_enter+0x2f/0x50
kern.warn kernel: [ 9702.092701]  do_idle+0xfb/0x1e0
kern.warn kernel: [ 9702.093362]  cpu_startup_entry+0x18/0x20
kern.warn kernel: [ 9702.094019]  rest_init+0xb5/0xc0
kern.warn kernel: [ 9702.094645]  arch_call_rest_init+0x9/0x14
kern.warn kernel: [ 9702.095273]  start_kernel+0x6fd/0x713
kern.warn kernel: [ 9702.095885]  x86_64_start_reservations+0x29/0x2f
kern.warn kernel: [ 9702.096470]  x86_64_start_kernel+0x8b/0x92
kern.warn kernel: [ 9702.097036]  secondary_startup_64_no_verify+0xb0/0xbb
kern.warn kernel: [ 9702.097609]  </TASK>
kern.warn kernel: [ 9702.098106] ---[ end trace 420d63bf0ff87e13 ]---
daemon.notice netifd: Network device 'eth0' link is down
daemon.notice netifd: Interface 'lan' has link connectivity loss
kern.info kernel: [ 9702.142465] r8125: eth0: link down
daemon.notice netifd: Network device 'eth0' link is up
daemon.notice netifd: Interface 'lan' has link connectivity
kern.info kernel: [ 9705.547704] r8125: eth0: link up

iKira, 2022-09-13:
你用的ESXi是6.7版本吗，给8125装的是哪个驱动？我之前用的是8125B，ESXi上8125B的驱动非常不稳定，一有长时间大流量就会断开，基本可以确定是驱动问题。

LemoFire, 2022-09-14:
> 你用的ESXi是6.7版本吗，给8125装的是哪个驱动？我之前用的是8125B，ESXi上8125B的驱动非常不稳定，一有长时间大流量就会断开，基本可以确定是驱动问题。

我觉得他说的是直通

iKira, 2022-09-15:
> > 你用的ESXi是6.7版本吗，给8125装的是哪个驱动？我之前用的是8125B，ESXi上8125B的驱动非常不稳定，一有长时间大流量就会断开，基本可以确定是驱动问题。
> 
> 我觉得他说的是直通

确实，他标题说的是直通。

pisheng, 2022-09-20:
> 你用的ESXi是6.7版本吗，给8125装的是哪个驱动？我之前用的是8125B，ESXi上8125B的驱动非常不稳定，一有长时间大流量就会断开，基本可以确定是驱动问题。

7.0的exsi，只通网卡，用的lede集成的8125B驱动。

cncqcyc, 2022-09-21:
既然是直通，应该与ESXi没有关系，我是PVE虚拟机，之前用8125B，现在用8125E和8156BUSB网卡，都没有出现过断网现象，所以应该跟驱动程序没有关系，看你日志，你是编译了turboACC里的shortcut-fe-cm模块吧，很久以前这个模块好像会引起断流，我编译的时候turboACC里的模块一个都没有选，反正重来没有遇见断网的现象，你把相关加速的组件去掉试试

pisheng, 2022-09-23:
> 既然是直通，应该与ESXi没有关系，我是PVE虚拟机，之前用8125B，现在用8125E和8156BUSB网卡，都没有出现过断网现象，所以应该跟驱动程序没有关系，看你日志，你是编译了turboACC里的shortcut-fe-cm模块吧，很久以前这个模块好像会引起断流，我编译的时候turboACC里的模块一个都没有选，反正重来没有遇见断网的现象，你把相关加速的组件去掉试试

感谢，再去研究下，看下怎么修改相关参数，我是用github action编译的。

pisheng, 2022-09-23:
> 既然是直通，应该与ESXi没有关系，我是PVE虚拟机，之前用8125B，现在用8125E和8156BUSB网卡，都没有出现过断网现象，所以应该跟驱动程序没有关系，看你日志，你是编译了turboACC里的shortcut-fe-cm模块吧，很久以前这个模块好像会引起断流，我编译的时候turboACC里的模块一个都没有选，反正重来没有遇见断网的现象，你把相关加速的组件去掉试试

CONFIG_PACKAGE_luci-app-turboacc=n
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_DNSFORWARDER=n
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_DNSPROXY=n
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_OFFLOADING=n

你好，是关掉这几个吗？

cncqcyc, 2022-09-23:
> CONFIG_PACKAGE_luci-app-turboacc=n CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_DNSFORWARDER=n CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_DNSPROXY=n CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_OFFLOADING=n [...]

我是把整个luci都关掉了，这里面的几个模块在这里取消，实际上有些默认选中编译了的，所以还要去模块所在的地方取消编译，我只单独把kmod-tcp-bbr编译了

closed 2026-09-05 by coolsnowwolf