coolsnowwolf/lede #10097: 编译5.15内核x86固件报错

bi4tms, 2022-09-12:
拉取最新源码，编译报错，具体日志如下：
Packaged contents of /home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-nf-nathelper-extra into /home/x86/lede/bin/targets/x86/64/packages/kmod-nf-nathelper-extra_5.15.64-1_x86_64.ipk
removed '/home/x86/lede/bin/targets/x86/64/packages/kmod-ip6tables_5.15.64-1_x86_64.ipk'
rstrip.sh: /home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ip6tables/lib/modules/5.15.64/ip6table_filter.ko: relocatable
rstrip.sh: /home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ip6tables/lib/modules/5.15.64/ip6table_mangle.ko: relocatable
rstrip.sh: /home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ip6tables/lib/modules/5.15.64/ip6t_REJECT.ko: relocatable
Packaged contents of /home/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-ip6tables into /home/x86/lede/bin/targets/x86/64/packages/kmod-ip6tables_5.15.64-1_x86_64.ipk
removed '/home/x86/lede/bin/targets/x86/64/packages/kmod-ipt-tproxy_5.15.64-1_x86_64.ipk'
Package kmod-ipt-tproxy is missing dependencies for the following libraries:
nf_tproxy_ipv4.ko
nf_tproxy_ipv6.ko
modules/netfilter.mk:681: recipe for target '/home/x86/lede/bin/targets/x86/64/packages/kmod-ipt-tproxy_5.15.64-1_x86_64.ipk' failed
make[3]: *** [/home/x86/lede/bin/targets/x86/64/packages/kmod-ipt-tproxy_5.15.64-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/x86/lede/package/kernel/linux'
time: package/kernel/linux/compile#8.71#48.71#53.27
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[2]: *** [package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/x86/lede'
package/Makefile:110: recipe for target '/home/x86/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/x86/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86/lede'
/home/x86/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

joketans, 2022-09-12:
同样问题，同反馈。

SHELL= flock /home/lede/tmp/.root-copy.flock -c 'cp -fpR /home/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-ipt-tproxy is missing dependencies for the following libraries:
nf_tproxy_ipv4.ko
nf_tproxy_ipv6.ko
make[3]: *** [modules/netfilter.mk:681: /home/lede/bin/targets/x86/64/packages/kmod-ipt-tproxy_5.15.64-1_x86_64.ipk] Error 1

closed 2022-09-12 by coolsnowwolf

koliner, 2022-09-12:
Package kmod-ipt-tproxy is missing dependencies for the following libraries:
nf_tproxy_ipv4.ko
nf_tproxy_ipv6.ko
make[3]: *** [modules/netfilter.mk:681: /home/lede/lede/bin/targets/ramips/mt7621/packages/kmod-ipt-tproxy_5.4.203-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/lede/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.64#0.26#3.88
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/lede/lede'
make[1]: *** [package/Makefile:110: /home/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lede/lede'
make: *** [/home/lede/lede/include/toplevel.mk:230: world] Error 2

brumegoo, 2022-09-13:
the same problem.