coolsnowwolf/lede #10100: 最新的固件编译失败

madcat0513, 2022-09-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
编译时发生冲突，删除tmp文件无用，编译时选kmod-ip6tables模块就会出错
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：r7800

### (3) 详细日志

A：Collected errors:
 * check_data_file_clashes: Package kmod-ip6tables wants to install file /home/madcat/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/lib/modules/5.4.203/nf_log_ipv6.ko
        But that file is already provided by package  * kmod-nf-log6
 * opkg_install_cmd: Cannot install package kmod-ip6tables.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/madcat/lean'
make[1]: *** [package/Makefile:111: /home/madcat/lean/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/madcat/lean'
make: *** [/home/madcat/lean/include/toplevel.mk:230：world] 错误 2

Curious-r, 2022-09-13:
同样有这个问题，机型CM520

1054009064, 2022-09-13:
删了 .config 重新来吧

madcat0513, 2022-09-13:
> 删了 .config 重新来吧

忘记说了，删除了.config后只要选kmod-ip6tables还是报错

yulinsoft, 2022-09-13:
7621的hc5962编译时也有这个问题，我回退到9月8日的版本，编译通过。

closed 2022-09-14 by coolsnowwolf