coolsnowwolf/lede #10103: ipq807x 5.15内核编译出错

mitsukileung, 2022-09-13:
ipq807x

貌似是/target/linux/ipq807x/patches-5.15/0109-PCI-qcom-fix-IPQ8074-Gen2-support.patch 这个patch出错？

[image]

zhangguanzhang, 2022-09-13:
https://github.com/coolsnowwolf/lede/commit/f66cf9d38c3d013902543f7ab09bee1bedc8f7a4
再试试

mitsukileung, 2022-09-13:
> [f66cf9d](https://github.com/coolsnowwolf/lede/commit/f66cf9d38c3d013902543f7ab09bee1bedc8f7a4) 再试试

我是更新了这个PR更编译的，还是不行

zhangguanzhang, 2022-09-13:
是那个补丁有问题

mitsukileung, 2022-09-13:
应该是 0109-PCI-qcom-fix-IPQ8074-Gen2-support.patch 这个补丁的问题
不懂怎么改，等大佬修复吧

mitsukileung, 2022-09-13:
Applying /workdir/openwrt/target/linux/ipq807x/patches-5.15/0140-clk-qcom-ipq8074-fix-NSS-core-PLL-s.patch using plaintext: 
[1752](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1753)
patching file drivers/clk/qcom/gcc-ipq8074.c
[1753](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1754)
Hunk #1 succeeded at 4417 with fuzz 1 (offset 46 lines).
[1754](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1755)
Hunk #2 FAILED at 4814.
[1755](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1756)
1 out of 2 hunks FAILED -- saving rejects to file drivers/clk/qcom/gcc-ipq8074.c.rej
[1756](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1757)
Patch failed!  Please fix /workdir/openwrt/target/linux/ipq807x/patches-5.15/0140-clk-qcom-ipq8074-fix-NSS-core-PLL-s.patch!
[1757](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1758)
make[4]: *** [Makefile:36: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.67/.prepared_8a08e17af136dd17494d851bba0ee4d0] Error 1
[1758](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1759)
make[3]: *** [Makefile:11: compile] Error 2
[1759](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1760)
make[4]: Leaving directory '/workdir/openwrt/target/linux/ipq807x'
[1760](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1761)
make[3]: Leaving directory '/workdir/openwrt/target/linux'
[1761](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1762)
time: target/linux/compile#12.41#6.06#14.97
[1762](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1763)
    ERROR: target/linux failed to build.
[1763](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1764)
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
[1764](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1765)
make[2]: Leaving directory '/workdir/openwrt'
[1765](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1766)
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
[1766](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1767)
make[1]: Leaving directory '/workdir/openwrt'
[1767](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3044673482/jobs/4905326917#step:13:1768)
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

这个补丁也是出错

closed 2022-09-13 by coolsnowwolf