coolsnowwolf/lede #10110: 使用新内核5.15.67编译的360v6固件，启动卡在uboot

zzcand111, 2022-09-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
x
## 2. 详细叙述
使用新内核5.15.67编译的360v6固件，启动卡在uboot 
### (1) 具体问题

A：路由不能正常启动

### (2) 路由器型号和固件版本

A：360v6 

### (3) 详细日志

A：****************************
*    FIRMWARE UPGRADING    *
* DO NOT POWER OFF DEVICE! *
****************************

Executing: nand erase 0x1180000 0x5940000; nand write 0x44000000 0x1180000 0x2380000

NAND erase: device 0 offset 0x1180000, size 0x5940000
Erasing at 0x6aa0000 -- 100% complete.
OK

NAND write: device 0 offset 0x1180000, size 0x2380000
 37224448 bytes written: OK
HTTP ugrade is done! Rebooting...

resetting ...

Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.0.3-00077-IPQ60xxLZB-2
S - IMAGE_VARIANT_STRING=IPQ6018LA
S - OEM_IMAGE_VERSION_STRING=crm-ubuntu64
S - Boot Interface: NAND
S - Secure Boot: Off
S - Boot Config @ 0x000a602c = 0x000002e5
S - JTAG ID @ 0x000a607c = 0x001390e1
S - OEM ID @ 0x000a6080 = 0x00000000
S - Serial Number @ 0x000a4128 = 0x1b321e53
S - OEM Config Row 0 @ 0x000a4188 = 0x0000000000000000
S - OEM Config Row 1 @ 0x000a4190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x000a4130 = 0x0000800018200021
S - Feature Config Row 1 @ 0x000a4138 = 0x02c3e83783000009
S - PBL Patch Ver: 1
S - I-cache: On
S - D-cache: On
B -      3413 - PBL, Start
B -       592 - bootable_media_detect_entry, Start
B -      4339 - bootable_media_detect_success, Start
B -      5207 - elf_loader_entry, Start
B -      5380 - auth_hash_seg_entry, Start
B -      7847 - auth_hash_seg_exit, Start
B -      8342 - elf_segs_hash_verify_entry, Start
B -    110491 - elf_segs_hash_verify_exit, Start
B -    114917 - auth_xbl_sec_hash_seg_entry, Start
B -    115061 - auth_xbl_sec_hash_seg_exit, Start
B -    121607 - xbl_sec_segs_hash_verify_entry, Start
B -    121608 - xbl_sec_segs_hash_verify_exit, Start
B -    122537 - PBL, End
B -    103303 - SBL1, Start
B -    243390 - GCC [RstStat:0x0, RstDbg:0x600000] WDog Stat : 0x4
B -    245830 - clock_init, Start
D -      2745 - clock_init, Delta
B -    254309 - boot_flash_init, Start
D -     30164 - boot_flash_init, Delta
B -    287767 - sbl1_ddr_set_default_params, Start
D -       244 - sbl1_ddr_set_default_params, Delta
B -    294386 - boot_config_data_table_init, Start
D -      4819 - boot_config_data_table_init, Delta - (575 Bytes)
B -    304298 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:0
B -    309056 - Image Load, Start
D -      6619 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    318389 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    326258 - sbl1_ddr_set_params, Start
B -    331260 - CPR configuration: 0x366
B -    334432 - Pre_DDR_clock_init, Start
D -       183 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    371093 - Image Load, Start
D -       427 - APDP Image Loaded, Delta - (0 Bytes)
B -    384208 - Image Load, Start
D -       427 - QTI_MISC Image Loaded, Delta - (0 Bytes)
B -    386618 - Image Load, Start
D -       823 - Auth Metadata
D -       610 - Segments hash check
D -     22600 - QSEE Dev Config Image Loaded, Delta - (36354 Bytes)
B -    411170 - Image Load, Start
D -      6558 - Auth Metadata
D -     10645 - Segments hash check
D -    343491 - QSEE Image Loaded, Delta - (1470632 Bytes)
B -    755088 - Image Load, Start
D -       671 - Auth Metadata
D -      1007 - Segments hash check
D -     35624 - RPM Image Loaded, Delta - (102664 Bytes)
B -    792390 - Image Load, Start
D -       702 - Auth Metadata
D -      3203 - Segments hash check
D -    124501 - APPSBL Image Loaded, Delta - (562732 Bytes)
B -    932537 - SBL1, End
D -    829539 - SBL1, Delta
S - Flash Throughput, 4000 KB/s  (2173629 Bytes,  446480 us)
S - Core 0 Frequency, 800 MHz
S - DDR Frequency, 466 MHz

U-Boot 2016.01 (Dec 26 2021 - 11:26:00 +0800)

DRAM:  smem ram ptable found: ver: 2 len: 4
512 MiB
NAND:  ONFI device found
ID = 158061c8
Vendor = c8
Device = 61
SF: Unsupported flash IDs: manuf 00, jedec 0000, ext_jedec 0000
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
128 MiB
MMC:   <NULL>: 0
PCI0 is not defined in the device tree
In:    serial@78B1000
Out:   serial@78B1000
Err:   serial@78B1000
machid: 8030200
eth0 MAC Address from ART is not valid
eth1 MAC Address from ART is not valid
eth2 MAC Address from ART is not valid
eth3 MAC Address from ART is not valid
eth4 MAC Address from ART is not valid
eth5 MAC Address from ART is not valid
Net:   MAC0 addr:0:3:7f:ba:db:ad
PHY ID1: 0x4d
PHY ID2: 0xd0b1
EDMA ver 1 hw init
Num rings - TxDesc:1 (0-0) TxCmpl:1 (0-0)
RxDesc:1 (15-15) RxFill:1 (7-7)
ipq6018_edma_alloc_rings: successfull
ipq6018_edma_setup_ring_resources: successfull
ipq6018_edma_configure_rings: successfull
ipq6018_edma_hw_init: successfull
eth0
Hit any key to stop autoboot:  0 
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: volume 2 ("rootfs_data") re-sized from 9 to 408 LEBs
ubi0: attached mtd1 (name "mtd=0", size 89 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 714, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1663122550
ubi0: available PEBs: 0, total reserved PEBs: 714, PEBs reserved for bad PEB handling: 20
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (4825088)
Config not availabale
IPQ6018# 
IPQ6018# 
IPQ6018#

sky2016cn, 2022-09-14:
新固件大小默认400MB，是不是你的空间不够？

zzcand111, 2022-09-14:
> 新固件大小默认400MB，是不是你的空间不够？

[image]
4.4.60固件正常启动

closed 2022-09-14 by aiamadeus

LXY1226, 2022-09-18:
bad kernel configuration?

hcym, 2022-11-30:
我到，内核版本
还以为是分区，明天再试试

zzcand111, 2022-11-30:
> 我到，内核版本 还以为是分区，明天再试试

目前，5.15内核玩不起来，等大佬修复吧