coolsnowwolf/lede #10111: 本地编译出错

qqsir-dev, 2022-09-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency lsblk for luci-app-diskman
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-diskman found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-diskman.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-diskman-zh-cn:
 *      lsblk
 * opkg_install_cmd: Cannot install package luci-i18n-diskman-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/user/openwrt'
make[1]: *** [package/Makefile:111: /home/user/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:230: world] Error 2
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

zzcand111, 2022-09-14:
缺少依赖，配置勾选上lsblk，再试试

qqsir-dev, 2022-09-14:
> 缺少依赖，配置勾选上lsblk，再试试

果然是这样，可是之前是默认勾上的，这次不知道为什么没有勾。不管怎样，谢谢！

closed 2022-09-14 by aiamadeus