coolsnowwolf/lede #10112: 云编译并没有选择samba36-server，提示和samba4冲突了

qqsir-dev, 2022-09-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
* pkg_hash_check_unresolved: cannot find dependency samba4-libs for samba4-server
Configuring wsdd2.
 * pkg_hash_fetch_best_installation_candidate: Packages for samba4-server found, but incompatible with the architectures configured
Configuring kmod-usb-net-kaweth.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
Configuring lscpu.
 * 	samba4-libs
Configuring luci-i18n-qos-zh-cn.
 * opkg_install_cmd: Cannot install package luci-app-samba4.
Configuring kmod-mppe.
Configuring ppp.
 * check_conflicts_for: The following packages conflict with samba4-server:
Configuring whereis.
 * check_conflicts_for: 	samba36-server * 
A：

DHDAXCW, 2022-09-14:
看到你配置选了luci-app-samba4，需要手动自己加**`# CONFIG_luci-app-samba is net`**或者去编译菜单关掉即可。要么取消samba4，二选一~

qqsir-dev, 2022-09-15:
> 这里默认 [luci-app-samba](https://github.com/coolsnowwolf/lede/blob/master/target/linux/rockchip/Makefile#L21)开着的， 看到你配置选了luci-app-samba4，形成配置它俩是开着的。需要手动自己加`# CONFIG_luci-app-samba is net`或者去编译菜单关掉即可。要么取消samba4，二选一~

收到，谢谢。

closed 2022-09-16 by qqsir-dev

occultyang, 2022-11-19:
> 这里默认 [luci-app-samba](https://github.com/coolsnowwolf/lede/blob/master/target/linux/rockchip/Makefile#L21)开着的， 看到你配置选了luci-app-samba4，形成配置它俩是开着的。需要手动自己加`# CONFIG_luci-app-samba is net`或者去编译菜单关掉即可。要么取消samba4，二选一~
对不起，我是小白，我也遇到此问题，请问是那个编译菜单可以关闭？