coolsnowwolf/lede #10113: ubuntu20.04 ，编译libxlts（host）不通过，提示configure: error: Could not find libxml2 anywhere,

FUjr, 2022-09-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* 有类似issue，但均未解决
* #9744
* #9686
## 2. 详细叙述
ubuntu20.04 已安装readme要求的所有驱动，编译libxlts不通过，提示configure: error: Could not find libxml2 anywhere,
### (1) 具体问题

A：ubuntu20.04 已安装readme要求的所有驱动，编译libxlts（host）不通过，提示configure: error: Could not find libxml2 anywhere,

编译配置：
先选上了所有包，逐个删除不需要的包，libxslt（host）取消勾选了仍要编译
<blockcode>
-*- Select all target specific packages by default 
-*- Select all kernel module packages by default
[*] Select all userspace packages by default
</blockcode>
<details>
<summary>日志</summary>
<pre><blockcode>
fjr@DESKTOP-C3PE1OB:~/lede$ make  package/feeds/packages/libxslt/host/compile V=s

time: target/linux/prereq#0.23#0.40#0.63
Checking 'rsync'... ok.
make[1]: Entering directory '/home/fjr/lede'
make[2]: Entering directory '/home/fjr/lede/feeds/packages/libs/libxml2'
mkdir -p /home/fjr/lede/dl
SHELL= flock /home/fjr/lede/tmp/.libxml2-2.9.14.tar.xz.flock -c '       /home/fjr/lede/scripts/download.pl "/home/fjr/lede/dl" "libxml2-2.9.14.tar.xz" "60d74a257d1ccec0475e749cba2f21559e48139efba6ff28224357c7c798dfee" "" "@GNOME/libxml2/2.9"    '
make[2]: Leaving directory '/home/fjr/lede/feeds/packages/libs/libxml2'
time: package/feeds/packages/libxml2/host-compile#0.30#0.06#0.29
make[2]: Entering directory '/home/fjr/lede/feeds/packages/libs/libxslt'
mkdir -p /home/fjr/lede/dl
SHELL= flock /home/fjr/lede/tmp/.libxslt-1.1.34.tar.gz.flock -c '       /home/fjr/lede/scripts/download.pl "/home/fjr/lede/dl" "libxslt-1.1.34.tar.gz" "98b1bd46d6792925ad2dfe9a87452ea2adebf69dcb9919ffd55bf926a7f93f7f" "" "http://xmlsoft.org/sources/" "ftp://fr.rpmfind.net/pub/libxml/"    '
(cd /home/fjr/lede/build_dir/hostpkg/libxslt-1.1.34; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/fjr/lede/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/fjr/lede/staging_dir/host/bin/autom4te AUTOCONF=/home/fjr/lede/staging_dir/host/bin/autoconf AUTOMAKE=/home/fjr/lede/staging_dir/host/bin/automake ACLOCAL=/home/fjr/lede/staging_dir/host/bin/aclocal AUTOHEADER=/home/fjr/lede/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/fjr/lede/staging_dir/host/bin/libtoolize LIBTOOL=/home/fjr/lede/staging_dir/host/bin/libtool M4=/home/fjr/lede/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/home/fjr/lede/staging_dir/host/bin/libtoolize --install' /home/fjr/lede/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/fjr/lede/staging_dir/host/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/fjr/lede/staging_dir/host/bin/aclocal -I m4 -I . --force
autoreconf: configure.ac: tracing
autoreconf: running: /home/fjr/lede/staging_dir/host/bin/libtoolize --install --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./config.guess'
OpenWrt-libtoolize: linking file `./config.sub'
OpenWrt-libtoolize: linking file `./install-sh'
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.ac and
OpenWrt-libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
OpenWrt-libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: running: /home/fjr/lede/staging_dir/host/bin/autoconf --include=m4 --include=. --prepend-include=/home/fjr/lede/staging_dir/host/share/aclocal --force
autoreconf: running: /home/fjr/lede/staging_dir/host/bin/autoheader --include=m4 --include=. --prepend-include=/home/fjr/lede/staging_dir/host/share/aclocal --force
autoreconf: running: /home/fjr/lede/staging_dir/host/bin/automake --add-missing --force-missing
autoreconf: Leaving directory `.'
(cd /home/fjr/lede/build_dir/hostpkg/libxslt-1.1.34/; if [ -x configure ]; then cp -fpR /home/fjr/lede/scripts/config.{guess,sub} /home/fjr/lede/build_dir/hostpkg/libxslt-1.1.34// && CC="gcc" CFLAGS="-O2 -I/home/fjr/lede/staging_dir/host/include -I/home/fjr/lede/staging_dir/hostpkg/include -I/home/fjr/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include" CXX="g++" CPPFLAGS="-I/home/fjr/lede/staging_dir/host/include -I/home/fjr/lede/staging_dir/hostpkg/include -I/home/fjr/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include" LDFLAGS="-L/home/fjr/lede/staging_dir/host/lib -L/home/fjr/lede/staging_dir/hostpkg/lib -L/home/fjr/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/fjr/lede/staging_dir/hostpkg --exec-prefix=/home/fjr/lede/staging_dir/hostpkg --sysconfdir=/home/fjr/lede/staging_dir/hostpkg/etc --localstatedir=/home/fjr/lede/staging_dir/hostpkg/var --sbindir=/home/fjr/lede/staging_dir/hostpkg/bin --with-libxml-prefix=/usr/include/libxml2 --without-python --without-crypto --without-debug --without-mem-debug --without-debugger --without-profiler --without-plugins ; fi )
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for x86_64-pc-linux-gnu-gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /home/fjr/lede/staging_dir/host/bin/grep
[... 151 lines cut ...]
checking for vfprintf... yes
checking for vsprintf... yes
checking for vsnprintf... yes
checking for sscanf... yes
checking for clock_gettime... yes
checking for perl... perl
Disabling crypto support
Disabling debug support
Disabling debugger
Disabling profiler
checking for x86_64-pc-linux-gnu-xml2-config... no
checking for xml2-config... no
configure: error: Could not find libxml2 anywhere, check ftp://xmlsoft.org/.
make[2]: *** [Makefile:167: /home/fjr/lede/build_dir/hostpkg/libxslt-1.1.34/.configured] Error 1
make[2]: Leaving directory '/home/fjr/lede/feeds/packages/libs/libxslt'
time: package/feeds/packages/libxslt/host-compile#11.07#2.26#16.13
    ERROR: package/feeds/packages/libxslt [host] failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/libxslt/host/compile] Error 1
make[1]: Leaving directory '/home/fjr/lede'
make: *** [/home/fjr/lede/include/toplevel.mk:230: package/feeds/packages/libxslt/host/compile] Error 2
</bockcode></pre>
</details>

### (2) 路由器型号和固件版本
hiwifi c526a 极路由x

hzjnet, 2022-09-14:
这2个月的源码都存在这个情况，但是不是每次编译都存在，触发的机会很少，make distclean后就可能正常但是也可能不行，重新下载源码基本就解决了

FUjr, 2022-09-15:
> 

我是几乎100%触发，两台电脑都不行。

LinuxSzp, 2022-09-20:
feeds/packages/libs//libxml2在构建host库只构建了静态库，feeds/packages/libs//libxslt host构建时未指出libxml2的库路径和头文件路径
可以做以下更改通过编译
1.修改feeds/packages/libs/libxml2/Makefile，HOST_CONFIGURE_ARGS中，--disable-shared更改为--enable-shared
2.修改feeds/packages/libs/libxslt/Makefile，HOST_CONFIGURE_ARGS中添加
--with-libxml-libs-prefix=$(STAGING_DIR_HOSTPKG)/lib \
--with-libxml-include-prefix=$(STAGING_DIR_HOSTPKG)/include/libxml2/ \
3.build_dir/hostpkg/libxslt-1.1.34在构建xsltproc时，未链接libxml2库，需要修改build_dir/hostpkg/libxslt-1.1.34/xsltproc/Makefile中LIBXML_LIBS，修改为LIBXML_LIBS = -lxml2 -L/home/embzp/Project/lede-embzp/staging_dir/hostpkg/lib（-L绝对路径根据实际修改）

hzjnet, 2022-09-20:
> feeds/packages/libs//libxml2在构建host库只构建了静态库，feeds/packages/libs//libxslt host构建时未指出libxml2的库路径和头文件路径
> 可以做以下更改通过编译 [...]

谢谢，非常感谢！   如果能做成补丁或者提交修订给这个源就更好了

closed 2023-12-19 by aiamadeus

sxml, 2024-02-17:
卡在libxslt 编译失败
不懂 把上面的代码 用gpt 生成 在云编译 libxlts（host）不再报错 也不知道对不对
就加载自定义配置时 sed: can't read build_dir/hostpkg/libxslt-1.1.34/xsltproc/Makefile: No such file or directory 没差吧

# 修改libxml2的Makefile，启用shared
sed -i '/HOST_CONFIGURE_ARGS/ s/--disable-shared/--enable-shared/' feeds/packages/libs/libxml2/Makefile
# 修改libxslt的Makefile，指定libxml的库路径和头文件路径
sed -i '/HOST_CONFIGURE_ARGS/ a\--with-libxml-libs-prefix=$(STAGING_DIR_HOSTPKG)/lib' feeds/packages/libs/libxslt/Makefile
sed -i '/HOST_CONFIGURE_ARGS/ a\--with-libxml-include-prefix=$(STAGING_DIR_HOSTPKG)/include/libxml2/' feeds/packages/libs/libxslt/Makefile
# 修改xsltproc的Makefile，链接libxml2库
sed -i 's|LIBXML_LIBS =|LIBXML_LIBS = -lxml2 -L$(STAGING_DIR_HOSTPKG)/lib|' build_dir/hostpkg/libxslt-1.1.34/xsltproc/Makefile

Firsgith, 2024-03-08:
> 卡在libxslt 编译失败
> 不懂 把上面的代码 用gpt 生成 在云编译 libxlts（host）不再报错 也不知道对不对
> 就加载自定义配置时 sed: can't read build_dir/hostpkg/libxslt-1.1.34/xsltproc/Makefile: No such file or directory 没差吧
>
>
> # [...]

请问下你这个解决了吗？

rq1025330, 2024-03-22:
> 请问下你这个解决了吗？

解决了吗？

Firsgith, 2024-03-22:
没有，还是报错，准备尝试以前更早的commit

Firsgith, 2024-03-22:
> 解决了吗？

没有，准备尝试更早的commit

rq1025330, 2024-03-22:
> > > > 卡在libxslt编译失败
> > > > 不懂把上面的代码用gpt生成在云编译libxlts（host）不再报错也不知道对不对
> > > > 就加载配置时 sed: can't read build_dir/hostpkg/libxslt-1.1.34 /xsltproc/Makefile: 没有这样的文件或目录没差吧
> > > > # 修改libxml2的Makefile，启用共享
> > > > sed -i '/HOST_CONFIGURE_ARGS/ s/--disable-shared/--enable-shared/' feeds/packages/libs/libxml2/Makefile
> > > > # 修改libxslt的Makefile，指定libxml的库路径和头文件路径
> > > > sed -i '/HOST_CONFIGURE_ARGS/ a--with-libxml-libs-prefix=$(STAGING_DIR_HOSTPKG)/lib' feeds/packages/libs/libxslt/Makefile 
> > > > sed -i '/HOST_CONFIGURE_ARGS/ a--with-libxml-include-前缀=$(STAGING_DIR_HOSTPKG)/include/libxml2/' feeds/packages/libs/libxslt/Makefile
> > > > # 修改xsltproc的Makefile，链接libxml2库
> > > > sed -i 's|LIBXML_LIBS =|LIBXML_LIBS = -lxml2 -L$(STAGING_DIR_HOSTPKG)/lib|' build_dir/hostpkg/libxslt-1.1.34/xsltproc/Makefile
> > > 
> > > 
> > > 请问下你这个解决了吗？
> > 
> > 
> > 解决了吗？
> 
> 没有，准备尝试更早的提交

云编译也不行  同样错误 ？