coolsnowwolf/lede #10120: 9.13日开始，lede源makefile时出现严重的问题，请检查tab键

anjue39, 2022-09-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：9.13日开始出现，至今仍有问题。

### (2) 路由器型号和固件版本

A：经测试，x86没这个问题。斐讯k3，有问题。均用最新lede版本

### (3) 详细日志

A：
[image]

[image]

[image]

coolsnowwolf, 2022-09-15:
重新生成 .config ， 内核版本改变了

anjue39, 2022-09-15:
试过了，还是不行。luci-app-screen没了？

发自我的iPhone

anjue39, 2022-09-16:
> 重新生成 .config ， 内核版本改变了

Makefile missing separator. Stop.的问题，请确认makefile文件是用了tab键还是用了空格，代码有问题

closed 2022-09-17 by coolsnowwolf