coolsnowwolf/lede #10121: ERROR: tools/llvm-bpf failed to build.

Aes64X, 2022-09-15:
ERROR: tools/llvm-bpf failed to build.

似乎。。。炸了？？？
llvm-project-14.0.6 单线程太慢。。。。

已经有的提示：
build_dir/host/llvm-project-14.0.6.src/llvm/lib/CodeGen/MLRegallocEvictAdvisor.cpp:499:1: warning: '{anonymous}::MLEvictAdvisor::MLEvictAdvisor(llvm::MachineFunction&, const llvm::RAGreedy&, llvm::MLModelRunner*, const llvm::MachineBlockFrequencyInfo&, const llvm::MachineLoopInfo&)' defined but not used [-Wunused-function]
 MLEvictAdvisor::MLEvictAdvisor(MachineFunction &MF, const RAGreedy &RA,
 ^~~~~~~~~~~~~~

有进一步提示会再更新。

Aes64X, 2022-09-15:
更新
得了。。。。 一个 llvm-project-14.0.6 单线程跑了40分钟没跑完。。。

flyhong666, 2022-09-15:
这个llvm 是干啥用的，编译好鬼慢。。。能默认取消吗。

WYC-2020, 2022-09-15:
应该和pdf相关的，没啥卵用

Aes64X, 2022-09-15:
退回了。。。等排爆小组吧，个人能力搞不定。

closed 2022-09-15 by coolsnowwolf