coolsnowwolf/lede #10123: 下载 dl 库出错

adwzlad, 2022-09-15:
执行make download -j8或者make download -j1提示
make[2] -C package/network/services/bridger download
    ERROR: package/network/services/bridger failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

执行make download -j1 V＝s或者make download -j1 V=sc提示
make[2]: Entering directory '/home/o/lede/package/network/services/bridger'
bash: line 1: /home/o/lede/staging_dir/host/llvm-bpf/bin/clang: No such file or directory
bash: line 1: [: : integer expression expected
/home/o/lede/include/bpf.mk:70: *** ERROR: LLVM/clang version too old. Minimum required: 12, found: .  Stop.
make[2]: Leaving directory '/home/o/lede/package/network/services/bridger'
time: package/network/services/bridger/download#0.05#0.00#0.06
    ERROR: package/network/services/bridger failed to build.
make[1]: *** [package/Makefile:116: package/network/services/bridger/download] Error 1

moonerick, 2022-09-15:
同样的错误……好像是新增就的一个bridger有问题

xslingcn, 2022-09-15:
同样的错误+1

zhoulijun89, 2022-09-15:
同样的错误+1

ricksuzade-maker, 2022-09-15:
神奇的是，居然不影响编译通过。。。

mitsukileung, 2022-09-15:
> 神奇的是，居然不影响编译通过。。。

本地编译也编不过呀，会自动退出

Vohrt, 2022-09-15:
same issue

closed 2022-09-15 by coolsnowwolf

WYC-2020, 2022-09-22:
> > 神奇的是，居然不影响编译通过。。。
> 
> 本地编译也编不过呀，会自动退出

我是直接卡死，cpu直接爆炸