coolsnowwolf/lede #10126: ERROR: package/network/utils/umbim failed to build.

zhangguanzhang, 2022-09-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

`r2s` 开启

```
CONFIG_PACKAGE_umbim=y
```

### (3) 详细日志

```
make[3]: Entering directory '/home/guanzhang/lede/package/network/utils/umbim'
mkdir -p /home/guanzhang/lede/dl
SHELL= flock /home/guanzhang/lede/tmp/.umbim-2019-09-11-184b707d.tar.xz.flock -c '  	/home/guanzhang/lede/scripts/download.pl "/home/guanzhang/lede/dl" "umbim-2019-09-11-184b707d.tar.xz" "482ff69144f81fafed99035840f5a24e772472f2df2f3ac0219d6de791ac5835" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /home/guanzhang/lede/tmp/dl && cd /home/guanzhang/lede/tmp/dl && rm -rf umbim-2019-09-11-184b707d && [ \! -d umbim-2019-09-11-184b707d ] && git clone  https://git.openwrt.org/project/umbim.git umbim-2019-09-11-184b707d && (cd umbim-2019-09-11-184b707d && git checkout 184b707ddaa0acee84d02e0ffe599cb8b67782bd && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd umbim-2019-09-11-184b707d && git log -1 --format='\''@%ct'\''` && rm -rf umbim-2019-09-11-184b707d/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c umbim-2019-09-11-184b707d | 	xz -zc -7e > /home/guanzhang/lede/tmp/dl/umbim-2019-09-11-184b707d.tar.xz && mv /home/guanzhang/lede/tmp/dl/umbim-2019-09-11-184b707d.tar.xz /home/guanzhang/lede/dl/ && rm -rf umbim-2019-09-11-184b707d;  )    '
rm -f /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d/.built
touch /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d/.built_check
MAKEFLAGS="" /home/guanzhang/lede/staging_dir/host/bin/ninja  -j1 -C /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d 
ninja: Entering directory `/home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d'
[1/5] Building C object CMakeFiles/umbim.dir/cli.c.o
FAILED: CMakeFiles/umbim.dir/cli.c.o 
/home/guanzhang/lede/staging_dir/host/bin/ccache /home/guanzhang/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc   -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d=umbim-2019-09-11-184b707d -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/guanzhang/lede/staging_dir/target-aarch64_generic_musl/usr/include -ffunction-sections -fdata-sections -DNDEBUG   -Os -ggdb -Wextra -Wall -Werror --std=gnu99 -Wmissing-declarations -Wno-unused-parameter -MD -MT CMakeFiles/umbim.dir/cli.c.o -MF CMakeFiles/umbim.dir/cli.c.o.d -o CMakeFiles/umbim.dir/cli.c.o -c /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d/cli.c
cli.c: In function 'mbim_device_caps_response':
cli.c:55:36: error: taking address of packed member of 'struct mbim_basic_connect_device_caps_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
   55 |         deviceid = mbim_get_string(&caps->deviceid, buffer);
      |                                    ^~~~~~~~~~~~~~~
cli.c:56:40: error: taking address of packed member of 'struct mbim_basic_connect_device_caps_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
   56 |         firmwareinfo = mbim_get_string(&caps->firmwareinfo, buffer);
      |                                        ^~~~~~~~~~~~~~~~~~~
cli.c:57:40: error: taking address of packed member of 'struct mbim_basic_connect_device_caps_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
   57 |         hardwareinfo = mbim_get_string(&caps->hardwareinfo, buffer);
      |                                        ^~~~~~~~~~~~~~~~~~~
cli.c: In function 'mbim_home_provider_response':
cli.c:111:39: error: taking address of packed member of 'struct mbimprovider' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  111 |         provider_id = mbim_get_string(&provider->providerid, buffer);
      |                                       ^~~~~~~~~~~~~~~~~~~~~
cli.c:112:41: error: taking address of packed member of 'struct mbimprovider' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  112 |         provider_name = mbim_get_string(&provider->providername, buffer);
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~
cli.c: In function 'mbim_registration_response':
cli.c:135:39: error: taking address of packed member of 'struct mbim_basic_connect_register_state_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  135 |         provider_id = mbim_get_string(&state->providerid, buffer);
      |                                       ^~~~~~~~~~~~~~~~~~
cli.c:136:41: error: taking address of packed member of 'struct mbim_basic_connect_register_state_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  136 |         provider_name = mbim_get_string(&state->providername, buffer);
      |                                         ^~~~~~~~~~~~~~~~~~~~
cli.c:137:39: error: taking address of packed member of 'struct mbim_basic_connect_register_state_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  137 |         roamingtext = mbim_get_string(&state->roamingtext, buffer);
      |                                       ^~~~~~~~~~~~~~~~~~~
cli.c: In function 'mbim_subscriber_response':
cli.c:171:40: error: taking address of packed member of 'struct mbim_basic_connect_subscriber_ready_status_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  171 |         subscriberid = mbim_get_string(&state->subscriberid, buffer);
      |                                        ^~~~~~~~~~~~~~~~~~~~
cli.c:172:36: error: taking address of packed member of 'struct mbim_basic_connect_subscriber_ready_status_r' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  172 |         simiccid = mbim_get_string(&state->simiccid, buffer);
      |                                    ^~~~~~~~~~~~~~~~
cli.c: In function 'mbim_connect_request':
cli.c:407:36: error: taking address of packed member of 'struct mbim_basic_connect_connect_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  407 |                 mbim_encode_string(&c->accessstring, apn);
      |                                    ^~~~~~~~~~~~~~~~
cli.c:418:44: error: taking address of packed member of 'struct mbim_basic_connect_connect_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  418 |                         mbim_encode_string(&c->username, _argv[2]);
      |                                            ^~~~~~~~~~~~
cli.c:419:44: error: taking address of packed member of 'struct mbim_basic_connect_connect_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  419 |                         mbim_encode_string(&c->password, _argv[3]);
      |                                            ^~~~~~~~~~~~
cli.c: In function 'mbim_pin_unlock_request':
cli.c:475:28: error: taking address of packed member of 'struct mbim_basic_connect_pin_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  475 |         mbim_encode_string(&p->pin, _argv[0]);
      |                            ^~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:50: /home/guanzhang/lede/build_dir/target-aarch64_generic_musl/umbim-2019-09-11-184b707d/.built] Error 1
make[3]: Leaving directory '/home/guanzhang/lede/package/network/utils/umbim'
time: package/network/utils/umbim/compile#0.27#0.04#0.36
    ERROR: package/network/utils/umbim failed to build.
make[2]: *** [package/Makefile:116: package/network/utils/umbim/compile] Error 1
make[2]: Leaving directory '/home/guanzhang/lede'
make[1]: *** [package/Makefile:110: /home/guanzhang/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/guanzhang/lede'
make: *** [/home/guanzhang/lede/include/toplevel.mk:230: world] Error 2
```

LXGMAX, 2022-09-16:
`cc1: all warnings being treated as errors`
软件包编译选项开启了-Werror，警告会当作错误

zhangguanzhang, 2022-09-16:
> `cc1: all warnings being treated as errors` 软件包编译选项开启了-Werror，警告会当作错误

那估计是源码里前不久升级了gcc后导致的

anjue39, 2022-09-16:
同样的问题，出230错误

anjue39, 2022-09-16:
K3编译出错，bcm53xx系列target-arm_cortex-a9分支 代码有问题 #10132

zhangguanzhang, 2022-09-20:
@LXGMAX 大佬知道怎么解决吗

anjue39, 2022-09-20:
lede他已降级到5.4，现在正常了。重新deconig配置，编译

发自我的iPhone

zhangguanzhang, 2022-09-20:
> lede他已降级到5.4，现在正常了。重新deconig配置，编译 发自我的iPhone
> […](#)

你的那个问题和我的都不是同一个问题，你那个是linux的patch有问题，我这个是某个包编译有问题

closed 2022-09-21 by aiamadeus