coolsnowwolf/lede #10128: rockchip 平台设备编译失败

haiibo, 2022-09-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

rockchip 平台设备编译失败

### (2) 路由器型号和固件版本

rockchip 平台设备

### (3) 详细日志

Collected errors:
 * resolve_conffiles: Existing conffile /home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/config/socat-opkg.
Setting flags for package libc to hold.
Setting flags for package libgcc1 to hold.
Setting flags for package libatomic1 to hold.
Setting flags for package libstdcpp6 to hold.
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
cp -fpR /home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip /home/summer/lede/build_dir/target-aarch64_generic_musl/root.orig-rockchip
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/appfilter: line 2: /usr/share/libubox/jshn.sh: No such file or directory
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/appfilter: line 3: /lib/functions.sh: No such file or directory
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/mosdns: line 11: uci: command not found
chmod: cannot access '/usr/share/onliner/*': No such file or directory
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/unblockmusic: line 3: /lib/functions/network.sh: No such file or directory
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/unblockmusic: line 82: network_flush_cache: command not found
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/unblockmusic: line 83: network_get_ipaddr: command not found
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
/home/summer/lede/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/nginx: line 12: /usr/bin/nginx-util: No such file or directory
Enabling AdGuardHome
Enabling adbyby
Enabling aliyundrive-fuse
Enabling aliyundrive-webdav
Enabling amule
./etc/init.d/appfilter: line 2: /usr/share/libubox/jshn.sh: No such file or directory
./etc/init.d/appfilter: line 3: /lib/functions.sh: No such file or directory
Enabling appfilter
Enabling aria2
Enabling arpbind
Enabling autoreboot
Enabling avahi-daemon
Enabling boot
Enabling bypass
Enabling cgroupfs-mount
Enabling cifs
Enabling cpufreq
Enabling cron
Enabling dbus
Enabling ddns
Enabling ddnsto
Enabling dnsforwarder
Enabling dnsmasq
Enabling dnsproxy
Enabling dockerd
Enabling dockerman
Enabling done
Enabling dropbear
Enabling eqos
Enabling etherwake
[... 239 lines cut ...]
Number of inodes 13842
Number of files 12250
Number of fragments 168
Number of symbolic links  787
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 804
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
make[5]: Leaving directory '/home/summer/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:24: install] Error 2
make[4]: Leaving directory '/home/summer/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/summer/lede/target/linux'
time: target/linux/install#536.22#27.57#330.40
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/summer/lede'
make[1]: *** [target/Makefile:24: /home/summer/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/summer/lede'
make: *** [/home/summer/lede/include/toplevel.mk:230：world] Error 2

zhangguanzhang, 2022-09-16:
```
error: ext4_allocate_best_fit_partial: failed to allocate 51 blocks, out of space?
make[5]: *** [/home/summer/lede/include/image.mk:336: /home/summer/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
```
ROOTFS size 调大一些

haiibo, 2022-09-16:
好的，感谢解答，好像就是我把 rootfs 调小了 100M 后才出现编译失败的。本 issue 将于一小时后关闭

closed 2022-09-16 by haiibo