coolsnowwolf/lede #10129: K3编译时出错

AndyLin2020, 2022-09-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Patch failed!  Please fix ./patches/100-Implements-AES-and-GCM-with-ARMv8-Crypto-Extensions.patch!

### (2) 路由器型号和固件版本

A：K3 A1

### (3) 详细日志

`
patching file ChangeLog.d/armv8_crypto_extensions.txt

patching file include/mbedtls/armv8ce_aes.h
patching file include/mbedtls/check_config.h
Hunk #1 succeeded at 95 (offset 23 lines).
Hunk #2 succeeded at 776 (offset -125 lines).
patching file include/mbedtls/config.h
Hunk #1 succeeded at 73 (offset 27 lines).
Hunk #2 succeeded at 1906 (offset -408 lines).
patching file library/aes.c
Hunk #1 succeeded at 69 (offset 30 lines).
Hunk #2 succeeded at 1054 (offset 53 lines).
patching file library/armv8ce_aes.c
patching file library/CMakeLists.txt
Hunk #1 succeeded at 7 (offset -8 lines).
patching file library/gcm.c
Hunk #1 succeeded at 71 (offset 30 lines).
Hunk #2 succeeded at 144 with fuzz 2 (offset 53 lines).
Hunk #3 succeeded at 258 (offset 52 lines).
patching file library/Makefile
Hunk #1 FAILED at 74.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file library/version_features.c
Hunk #1 succeeded at 586 (offset -38 lines).
Patch failed!  Please fix ./patches/100-Implements-AES-and-GCM-with-ARMv8-Crypto-Extensions.patch!
Makefile:173: recipe for target '/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/mbedtls-2.16.11/.prepared_0eac5b7a11be4054234dfb9344162b5f_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/mbedtls-2.16.11/.prepared_0eac5b7a11be4054234dfb9344162b5f_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/andy/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.13#0.21#0.30
    ERROR: package/libs/mbedtls failed to build.
package/Makefile:114: recipe for target 'package/libs/mbedtls/compile' failed
make[2]: *** [package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/home/andy/lede'
package/Makefile:110: recipe for target '/home/andy/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/andy/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/andy/lede'
/home/andy/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
andy@ubuntu:~/lede$ 
`

Crazy365, 2022-09-16:
我也出错了，不过跟你不一样，我的日志是这样的：

Applying /workdir/openwrt/target/linux/bcm53xx/patches-5.10/080-v5.13-0002-nvmem-brcm_nvram-new-driver-exposing-Broadcom-s-NVRA.patch using plaintext: 
patching file drivers/nvmem/Kconfig
Hunk #1 succeeded at 270 with fuzz 2 (offset -13 lines).
patching file drivers/nvmem/Makefile
Hunk #1 FAILED at 57.
1 out of 1 hunk FAILED -- saving rejects to file drivers/nvmem/Makefile.rej
patching file drivers/nvmem/brcm_nvram.c
Patch failed!  Please fix /workdir/openwrt/target/linux/bcm53xx/patches-5.10/080-v5.13-0002-nvmem-brcm_nvram-new-driver-exposing-Broadcom-s-NVRA.patch!
make[4]: *** [Makefile:40: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.142/.prepared_c467ab480fc3b6acf10cf8985409c99e] Error 1
make[3]: *** [Makefile:11: compile] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/bcm53xx'
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#11.99#6.14#14.98
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

anjue39, 2022-09-16:
一样的问题，反馈好几次了，至今没有找到解决方法，大佬说让我重新config，重新弄了也不行
[image]

AndyLin2020, 2022-09-16:
> 我也出错了，不过跟你不一样，我的日志是这样的：
>
> Applying /workdir/openwrt/target/linux/bcm53xx/patches-5.10/080-v5.13-0002-nvmem-brcm_nvram-new-driver-exposing-Broadcom-s-NVRA.patch using plaintext: patching file [...]

重新配置后也是一样错误提示，应该是内核补丁问题吧，等老大修复

mitsukileung, 2022-09-16:
我也是一样的提示

Applying /workdir/openwrt/target/linux/bcm53xx/patches-5.10/080-v5.13-0001-dt-bindings-nvmem-add-Broadcom-s-NVRAM.patch using plaintext: 
[2071](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2072)
patching file Documentation/devicetree/bindings/nvmem/brcm,nvram.yaml
[2072](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2073)

[2073](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2074)
Applying /workdir/openwrt/target/linux/bcm53xx/patches-5.10/080-v5.13-0002-nvmem-brcm_nvram-new-driver-exposing-Broadcom-s-NVRA.patch using plaintext: 
[2074](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2075)
patching file drivers/nvmem/Kconfig
[2075](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2076)
Hunk #1 succeeded at 270 with fuzz 2 (offset -13 lines).
[2076](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2077)
patching file drivers/nvmem/Makefile
[2077](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2078)
Hunk #1 FAILED at 57.
[2078](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2079)
1 out of 1 hunk FAILED -- saving rejects to file drivers/nvmem/Makefile.rej
[2079](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2080)
patching file drivers/nvmem/brcm_nvram.c
[2080](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2081)
Patch failed!  Please fix /workdir/openwrt/target/linux/bcm53xx/patches-5.10/080-v5.13-0002-nvmem-brcm_nvram-new-driver-exposing-Broadcom-s-NVRA.patch!
[2081](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2082)
make[4]: *** [Makefile:40: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.142/.prepared_31cf165f8ec5ae59ac02ca41fb286e12] Error 1
[2082](https://github.com/takayukileung/Actions-OpenWrt/actions/runs/3067614756/jobs/4954084284#step:13:2083)
make[3]: *** [Makefile:11: compile] Error 2

closed 2022-09-17 by coolsnowwolf