coolsnowwolf/lede #10130: r5s 最新固件中间的网卡无法使用

solitudechn, 2022-09-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r5s 最新固件中间的网卡无法使用

### (2) 路由器型号和固件版本

A：友善r5s 83903ede7b63691a43173f94caa37b92d9b46975 提交

### (3) 详细日志

A：编译的内核版本为  5.15.67 ，表现为中间网卡插上后不亮灯，无法使用。
使用 ethtool -i eth2 
返回 Cannot get driver information: No such device

内核日志中可以看到 网卡驱动已经成功加载，并且eth1已经成功启动，但未见到eth2启动，下附内核日志

```
[    7.543226] r8125 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
[    7.543868] r8125 0001:01:00.0: enabling device (0000 -> 0003)
[    7.562015] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    7.563130] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Random ether addr 2a:ff:f9:4e:d9:29
[    7.564454] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    7.567478] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
[    7.567478]  This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>. 
[    7.567478]  This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>. 
[    7.571575] usbcore: registered new interface driver r8152
[    7.572691] usbcore: registered new interface driver rtl8150
[    7.575176] usbcore: registered new interface driver ums-alauda
[    7.576048] usbcore: registered new interface driver ums-cypress
[    7.576930] usbcore: registered new interface driver ums-datafab
[    7.577775] usbcore: registered new interface driver ums-freecom
[    7.578829] usbcore: registered new interface driver ums-isd200
[    7.579728] usbcore: registered new interface driver ums-jumpshot
[    7.580603] usbcore: registered new interface driver ums-karma
[    7.581536] usbcore: registered new interface driver ums-sddr09
[    7.582488] usbcore: registered new interface driver ums-sddr55
[    7.583407] usbcore: registered new interface driver ums-usbat
[    7.584434] usbcore: registered new interface driver usblp
[    7.587895] usbcore: registered new interface driver usbserial_generic
[    7.588505] usbserial: USB Serial support registered for generic
[    7.599938] xt_time: kernel timezone is -0000
[    7.601173] usbcore: registered new interface driver asix
[    7.602186] usbcore: registered new interface driver cdc_eem
[    7.603442] usbcore: registered new interface driver cdc_ether
[    7.605066] usbcore: registered new interface driver cdc_ncm
[    7.606030] usbcore: registered new interface driver cdc_subset
[    7.607057] usbcore: registered new interface driver huawei_cdc_ncm
[    7.608506] usbcore: registered new interface driver kalmia
[    7.610261] usbcore: registered new interface driver plusb
[    7.612081] PPP generic driver version 2.4.2
[    7.613071] PPP MPPE Compression module registered
[    7.614120] NET: Registered PF_PPPOX protocol family
[    7.615621] usbcore: registered new interface driver qmi_wwan
[    7.617003] usbcore: registered new interface driver rndis_host
[    7.618234] usbcore: registered new interface driver sierra_net
[    7.620328] usbcore: registered new interface driver cdc_mbim
[    7.621499] usbcore: registered new interface driver option
[    7.622025] usbserial: USB Serial support registered for GSM modem (1-port)
[    7.624252] kmodloader: done loading kernel modules from /etc/modules.d/*
[   28.267510] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
[   28.341545] eth1: 0xffff800009d20000, 42:8a:77:1f:21:1f, IRQ 60
[   28.391522] br-lan: port 1(eth1) entered blocking state
[   28.392009] br-lan: port 1(eth1) entered disabled state
[   28.392677] device eth1 entered promiscuous mode
[   28.494305] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=62)
[   28.500008] rk_gmac-dwmac fe2a0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   28.510212] dwmac4: Master AXI performs any burst length
[   28.510724] rk_gmac-dwmac fe2a0000.ethernet eth0: No Safety Features support found
[   28.511416] rk_gmac-dwmac fe2a0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   28.512558] rk_gmac-dwmac fe2a0000.ethernet eth0: configuring for phy/rgmii-id link mode
[   32.206385] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   32.207252] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   32.346439] fast-classifier: starting up
[   32.346881] fast-classifier: registered
[   76.804211] r8125: eth1: link up
[   76.804622] br-lan: port 1(eth1) entered blocking state
[   76.805112] br-lan: port 1(eth1) entered forwarding state
[   76.806707] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

DHDAXCW, 2022-09-17:
看起来是你硬件问题吧。。
我今天重新拉源码编译了一次，没问题，正常识别
```log]
[    6.522798] r8125 0000:01:00.0: of_irq_parse_pci: failed with rc=-22
[    6.523379] r8125 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
[    6.524024] r8125 0000:01:00.0: enabling device (0000 -> 0003)
[    6.542293] r8125 0000:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    6.543150] r8125 0000:01:00.0 (unnamed net_device) (uninitialized): Random ether addr e6:ab:c1:74:8b:01
[    6.544451] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    6.547475] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com>
[    6.547475]  This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>.
[    6.547475]  This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>.
[    6.550295] r8125 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
[    6.550930] r8125 0001:01:00.0: enabling device (0000 -> 0003)
[    6.551893] r8125 0001:01:00.0: no MSI/MSI-X. Back to INTx.
[    6.567616] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    6.568458] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Random ether addr 42:6f:4b:09:f6:d0
[    6.569660] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    6.572681] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com>
[    6.572681]  This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>.
[    6.572681]  This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>.

[   26.455707] eth1: 0xffff800009470000, d2:94:7c:e4:27:3f, IRQ 0
[   26.456026] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
[   26.505661] br-lan: port 1(eth1) entered blocking state
[   26.506147] br-lan: port 1(eth1) entered disabled state
[   26.506825] device eth1 entered promiscuous mode
[   26.512906] eth2: 0xffff8000094b0000, d2:94:7c:e4:27:3f, IRQ 60
[   26.562523] br-lan: port 2(eth2) entered blocking state
[   26.563015] br-lan: port 2(eth2) entered disabled state
[   26.563650] device eth2 entered promiscuous mode

--------------------------------
我插上网线打印log

[  201.334640] r8125: eth1: link up
[  201.335035] br-lan: port 1(eth1) entered blocking state
[  201.335521] br-lan: port 1(eth1) entered forwarding state

[  213.738525] r8125: eth2: link up
[  213.738923] br-lan: port 2(eth2) entered blocking state
[  213.739413] br-lan: port 2(eth2) entered forwarding state

root@OpenWrt:/# ethtool -i eth2
driver: r8125
version: 9.009.02-NAPI-RSS
firmware-version:
expansion-rom-version:
bus-info: 0001:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
root@OpenWrt:/# ethtool -i eth1
driver: r8125
version: 9.009.02-NAPI-RSS
firmware-version:
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
root@OpenWrt:/#

然后没发现什么异常，再检查pci

root@OpenWrt:/# lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
```
以上检查最新源码没什么问题，大概是你硬件出问题了吧~

solitudechn, 2022-09-17:
> 看起来是你硬件问题吧。。 我今天重新拉源码编译了一次，没问题，正常识别
> 
> ```
> [    6.522798] r8125 0000:01:00.0: of_irq_parse_pci: failed with rc=-22
> [    6.523379] r8125 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
> [    6.524024] r8125 0000:01:00.0: enabling device (0000 -> 0003)
> [    6.542293] r8125 0000:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
> [    6.543150] r8125 0000:01:00.0 (unnamed net_device) (uninitialized): Random ether addr e6:ab:c1:74:8b:01
> [    6.544451] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    6.547475] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com>
> [    6.547475]  This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>.
> [    6.547475]  This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>.
> [    6.550295] r8125 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
> [    6.550930] r8125 0001:01:00.0: enabling device (0000 -> 0003)
> [    6.551893] r8125 0001:01:00.0: no MSI/MSI-X. Back to INTx.
> [    6.567616] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
> [    6.568458] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Random ether addr 42:6f:4b:09:f6:d0
> [    6.569660] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    6.572681] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com>
> [    6.572681]  This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>.
> [    6.572681]  This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>.
> 
> 
> [   26.455707] eth1: 0xffff800009470000, d2:94:7c:e4:27:3f, IRQ 0
> [   26.456026] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
> [   26.505661] br-lan: port 1(eth1) entered blocking state
> [   26.506147] br-lan: port 1(eth1) entered disabled state
> [   26.506825] device eth1 entered promiscuous mode
> [   26.512906] eth2: 0xffff8000094b0000, d2:94:7c:e4:27:3f, IRQ 60
> [   26.562523] br-lan: port 2(eth2) entered blocking state
> [   26.563015] br-lan: port 2(eth2) entered disabled state
> [   26.563650] device eth2 entered promiscuous mode
> 
> --------------------------------
> 我插上网线打印log
> 
> [  201.334640] r8125: eth1: link up
> [  201.335035] br-lan: port 1(eth1) entered blocking state
> [  201.335521] br-lan: port 1(eth1) entered forwarding state
> 
> [  213.738525] r8125: eth2: link up
> [  213.738923] br-lan: port 2(eth2) entered blocking state
> [  213.739413] br-lan: port 2(eth2) entered forwarding state
> 
> root@OpenWrt:/# ethtool -i eth2
> driver: r8125
> version: 9.009.02-NAPI-RSS
> firmware-version:
> expansion-rom-version:
> bus-info: 0001:01:00.0
> supports-statistics: yes
> supports-test: no
> supports-eeprom-access: no
> supports-register-dump: yes
> supports-priv-flags: no
> root@OpenWrt:/# ethtool -i eth1
> driver: r8125
> version: 9.009.02-NAPI-RSS
> firmware-version:
> expansion-rom-version:
> bus-info: 0000:01:00.0
> supports-statistics: yes
> supports-test: no
> supports-eeprom-access: no
> supports-register-dump: yes
> supports-priv-flags: no
> root@OpenWrt:/#
> 
> 然后没发现什么异常，再检查pci
> 
> root@OpenWrt:/# lspci
> 0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
> 0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
> 0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
> 0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
> 0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
> ```
> 
> 以上检查最新源码没什么问题，大概是你硬件出问题了吧~

我昨天试了一下lean大现在的release版本以及其他的openwrt编译版，基于lean最新编译的系统都无法使用，但是老版本（现在的release版本），其他人编译的纯openwrt正式版，均可以正常使用的

DHDAXCW, 2022-09-17:
这么神奇吗？

solitudechn, 2022-09-17:
> 看起来是你硬件问题吧。。 我今天重新拉源码编译了一次，没问题，正常识别
>
> ```
> [    6.522798] r8125 0000:01:00.0: of_irq_parse_pci: failed with rc=-22
> [    6.523379] r8125 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
> [ [...]

不知道是不是我编译的lspci用的是busybox的版本，lspci显示的信息没有详细的设备名称，只有设备id🤦

DHDAXCW, 2022-09-17:
```bash
opkg update
opkg install pciutils
```

solitudechn, 2022-09-17:
> ```shell
> opkg update
> opkg install pciutils
> ```

回去我试一下，谢谢
另外这个是我使用的编译脚本，不知道是不是我编译的时候config配置文件写错了🤦

https://github.com/solitudechn/lede/blob/master/.github/workflows/openwrt-ci.yml

DHDAXCW, 2022-09-17:
我也看不出来什么，你把开机kernel log加载完成复制过来

solitudechn, 2022-09-17:
> 我也看不出来什么，你把开机kernel log加载完成复制过来

好的，我回家的时候导出来，谢谢

solitudechn, 2022-09-17:
> 我也看不出来什么，你把开机kernel log加载完成复制过来

你好，下面是我的内核启动日志

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.67 (solitudechn@DESKTOP-0M7KR5V) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5076-83903ede7) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Sat Sep 17 09:45:26 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x00000000ef000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33304 r8192 d32232 u73728
[    0.000000] pcpu-alloc: s33304 r8192 d32232 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3830048K/3930112K available (8576K kernel code, 2372K rwdata, 2444K rodata, 1664K init, 722K bss, 91872K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001346] Console: colour dummy device 80x25
[    0.001838] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002893] pid_max: default: 32768 minimum: 301
[    0.003587] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.004347] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.007357] rcu: Hierarchical SRCU implementation.
[    0.008117] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009669] smp: Bringing up secondary CPUs ...
[    0.010833] Detected VIPT I-cache on CPU1
[... 389 lines cut ...]
[    7.703975] usbcore: registered new interface driver qmi_wwan
[    7.705334] usbcore: registered new interface driver rndis_host
[    7.706769] usbcore: registered new interface driver sierra_net
[    7.708872] usbcore: registered new interface driver cdc_mbim
[    7.710040] usbcore: registered new interface driver option
[    7.710623] usbserial: USB Serial support registered for GSM modem (1-port)
[    7.712746] kmodloader: done loading kernel modules from /etc/modules.d/*
[   28.995027] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
[   29.042390] eth1: 0xffff800009d20000, 42:8a:77:1f:21:1f, IRQ 60
[   29.092278] br-lan: port 1(eth1) entered blocking state
[   29.092772] br-lan: port 1(eth1) entered disabled state
[   29.093613] device eth1 entered promiscuous mode
[   29.194334] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=62)
[... cut ...]
[   29.207457] rk_gmac-dwmac fe2a0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   29.208606] rk_gmac-dwmac fe2a0000.ethernet eth0: configuring for phy/rgmii-id link mode
[   31.190447] r8125: eth1: link up
[   31.190865] br-lan: port 1(eth1) entered blocking state
[   31.191354] br-lan: port 1(eth1) entered forwarding state
[   31.193160] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   32.298705] fast-classifier: starting up
[   32.299143] fast-classifier: registered
```

solitudechn, 2022-09-17:
> ```shell
> opkg update
> opkg install pciutils
> ```

这个是我刚才使用线刷刷入的 lean 大的  release 版本的包，使用ethtool获得的信息
```
root@OpenWrt:~# ethtool -i eth0
driver: st_gmac
version: Jan_2016
firmware-version: 
expansion-rom-version: 
bus-info: 
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
root@OpenWrt:~# ethtool -i eth1
driver: r8125
version: 9.009.02-NAPI-RSS
firmware-version: 
expansion-rom-version: 
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
root@OpenWrt:~# ethtool -i eth2
driver: r8125
version: 9.009.02-NAPI-RSS
firmware-version: 
expansion-rom-version: 
bus-info: 0001:11:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
```

下面是相应的内核启动日志

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 4.19.237-android (lean@lean-esxi) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4779-6fd6edc2a), GNU ld (GNU Binutils) 2.34) #0 SMP Tue Jul 19 16:23:37 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] bootconsole [uart8250] enabled
[    0.000000] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000010000000, size 8 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1043968
[    0.000000]   DMA32 zone: 15288 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 978432 pages, LIFO batch:63
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 22 pages/cpu s51176 r8192 d30744 u90112
[    0.000000] pcpu-alloc: s51176 r8192 d30744 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Speculative Store Bypassing Safe (SSBS)
[... 778 lines cut ...]
[   32.024255] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: 
[   32.213322] eth1: 0xffffff800dbc0000, ca:4c:0b:90:30:f7, IRQ 103
[   32.263177] br-lan: port 1(eth1) entered blocking state
[   32.263705] br-lan: port 1(eth1) entered disabled state
[   32.264846] device eth1 entered promiscuous mode
[   32.266585] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   32.278053] eth2: 0xffffff800dc00000, 7c:b5:9b:6d:96:02, IRQ 114
[   32.328003] br-lan: port 2(eth2) entered blocking state
[   32.328532] br-lan: port 2(eth2) entered disabled state
[   32.329664] device eth2 entered promiscuous mode
[   32.335489] RTL8211F Gigabit Ethernet stmmac-0:01: attached PHY driver [RTL8211F Gigabit Ethernet] (mii_bus:phy_addr=stmmac-0:01, irq=73)
[   32.347268] dwmac4: Master AXI performs any burst length
[   32.347797] rk_gmac-dwmac fe2a0000.ethernet eth0: No Safety Features support found
[   32.348964] rk_gmac-dwmac fe2a0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   32.349902] rk_gmac-dwmac fe2a0000.ethernet eth0: registered PTP clock
[   32.351653] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   33.259005] S60samba (4214): drop_caches: 3
[   34.522719] r8125: eth2: link up
[   34.523139] br-lan: port 2(eth2) entered blocking state
[   34.523839] br-lan: port 2(eth2) entered forwarding state
[   34.524764] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   34.911355] fast-classifier: starting up
[   34.911808] fast-classifier: registered
[   49.558592] samba (7570): drop_caches: 3
```

solitudechn, 2022-09-17:
> 这个是我刚才使用线刷刷入的 lean 大的 release 版本的包，使用ethtool获得的信息
>
> ```
> root@OpenWrt:~# ethtool -i eth0
> driver: st_gmac
> version: Jan_2016
> firmware-version:
> expansion-rom-version:
> bus-info:
> supports-statistics: [...]

这个是我在上述release下直接使用 
```
dd if=openwrt.bin of=/dev/mmcblk0 bs=1M
sync
```
覆盖的我自己编译的系统

```
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
```

内核日志如下

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.67 (solitudechn@DESKTOP-0M7KR5V) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5076-83903ede7) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Sat Sep 17 09:45:26 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x00000000ef000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33304 r8192 d32232 u73728
[    0.000000] pcpu-alloc: s33304 r8192 d32232 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3830048K/3930112K available (8576K kernel code, 2372K rwdata, 2444K rodata, 1664K init, 722K bss, 91872K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[... 557 lines cut ...]
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcap-bin=y
CONFIG_PACKAGE_libcap-bin-capsh-shell="/bin/sh"
CONFIG_PACKAGE_libeudev=y
CONFIG_PACKAGE_libfdisk=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libkmod=y
CONFIG_PACKAGE_libmount=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libpci=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0
CONFIG_PACKAGE_pciids=y
CONFIG_PACKAGE_pciutils=y
CONFIG_PACKAGE_r8152-firmware=y
CONFIG_PACKAGE_whereis=y
CONFIG_TARGET_KERNEL_PARTSIZE=128
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
```

solitudechn, 2022-09-17:
@DHDAXCW 我刚才试着刷入了一下 https://github.com/DHDAXCW/NanoPi-R5S/releases/tag/2022.09.14-Lean3 这个里面的 [openwrt-rockchip-armv8-friendlyarm_nanopi-r5s-squashfs-sysupgrade.img.gz](https://github.com/DHDAXCW/NanoPi-R5S/releases/download/2022.09.14-Lean3/openwrt-rockchip-armv8-friendlyarm_nanopi-r5s-squashfs-sysupgrade.img.gz) 这个系统包，现象跟我自己编译的是一样的

以下是lspci的结果
```
# root @ OpenWrt in /tmp [17:10:12] 
$ lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor
```

以下为系统内核日志

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.67 (d@d) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5063-812c56e0b) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Tue Sep 13 17:19:25 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x00000000ef000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33368 r8192 d32168 u73728
[    0.000000] pcpu-alloc: s33368 r8192 d32168 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3829984K/3930112K available (8640K kernel code, 2374K rwdata, 2460K rodata, 1664K init, 724K bss, 91936K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[... 451 lines cut ...]
[   11.034126] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[   11.035574] usbcore: registered new interface driver option
[   11.036125] usbserial: USB Serial support registered for GSM modem (1-port)
[   11.039038] kmodloader: done loading kernel modules from /etc/modules.d/*
[   42.218741] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
[   44.886524] eth1: 0xffff800009f10000, 42:8a:77:1f:21:1f, IRQ 60
[   44.936610] br-lan: port 1(eth1) entered blocking state
[   44.937096] br-lan: port 1(eth1) entered disabled state
[   44.937768] device eth1 entered promiscuous mode
[   45.037970] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=62)
[   45.040665] rk_gmac-dwmac fe2a0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   45.049922] dwmac4: Master AXI performs any burst length
[   45.050421] rk_gmac-dwmac fe2a0000.ethernet eth0: No Safety Features support found
[... cut ...]
[   47.106454] br-lan: port 1(eth1) entered blocking state
[   47.106920] br-lan: port 1(eth1) entered forwarding state
[   47.107831] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   50.392179] fast-classifier: starting up
[   50.392646] fast-classifier: registered
[   69.774380] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   69.775111] NFSD: Using legacy client tracking operations.
[   69.775596] NFSD: starting 10-second grace period (net f0000000)
```

iKira, 2022-09-17:
用5.15.67编译我这用没问题，三个口都正常。用iperf3测试内网能跑到2.2Gbps，但是实际从NAS拷文件最多跑到1.2Gbps，还是需要lean的鸡血网卡驱动加持才行。

DHDAXCW, 2022-09-17:
建议把5.15里面的dts删掉，从5.19上拿过来，替换掉 就可以了https://github.com/coolsnowwolf/lede/blob/master/target/linux/rockchip/files-5.19/arch/arm64/boot/dts/rockchip/rk3568-nanopi-r5s.dts

DHDAXCW, 2022-09-17:
> 用5.15.67编译我这用没问题，三个口都正常。用iperf3测试内网能跑到2.2Gbps，但是实际从NAS拷文件最多跑到1.2Gbps，还是需要lean的鸡血网卡驱动加持才行。

你插NVME的时候就有问题，缺少gpio4 c1

solitudechn, 2022-09-17:
> 用5.15.67编译我这用没问题，三个口都正常。用iperf3测试内网能跑到2.2Gbps，但是实际从NAS拷文件最多跑到1.2Gbps，还是需要lean的鸡血网卡驱动加持才行。

方便把固件给我试试么?

hai2012, 2022-09-19:
我的也是的，你是不是加装m.2固态硬盘的啊。加固态eth1就没法使用，去掉固态就好了，

hai2012, 2022-09-19:
加固态eth1就不行了

solitudechn, 2022-09-19:
> 我的也是的，你是不是加装m.2固态硬盘的啊。加固态eth1就没法使用，去掉固态就好了，

我确实是没加装🤦

hai2012, 2022-09-19:
我的是加固态就不行
去掉就好了

hai2012, 2022-09-19:
现在显示双全工，但是没网

solitudechn, 2022-09-19:
> 我的是加固态就不行
> 去掉就好了

我上边重刷完系统后，什么都没插，就在eth2上插了个网线。显示的现象是原本应该是现实lan1的led灯会因为lan2亮起来。led2没任何反应，系统内部也看不到原本应该存在的eth1网卡

solitudechn, 2022-09-20:
@DHDAXCW 
你好，我在网上搜索相关信息的时候看到了使用 lspci -vvx 这个命令显示所有关于pci相关信息的命令。
我对边检查了一下正常识别网卡的lspci日志，和不正常识别的lspci日志。发现了正常识别的固件中，关于0000:01总线
存在
```
Kernel driver in use: pcieport
```
这个日志
而无法使用的固件，关于0000:01总线则没有这个日志
如下图所示
[image]

于此同时，我关注到所有其他的总线都包含这个日志，无论是可识别的固件，或是不可识别的固件
以下是相关信息对比
```
solitudechn@DESKTOP-0M7KR5V:/mnt/c/Users/solitudechn/Desktop$ cat 4.19.md | grep "PCI bridge\|pcieport"
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
0001:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
0001:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
        Kernel driver in use: pcieport
        Kernel driver in use: pcieport
solitudechn@DESKTOP-0M7KR5V:/mnt/c/Users/solitudechn/Desktop$ cat 5.15.md | grep "PCI bridge\|pcieport"
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01) (prog-if 00 [Normal decode])
        Kernel driver in use: pcieport
        Kernel driver in use: pcieport
        Kernel driver in use: pcieport
```
不知道上述信息能否为你提供帮助

iKenzo, 2022-10-27:
碰到了相同的问题，发现使用[DSA](https://openwrt.org/docs/guide-user/network/dsa/converting-to-dsa)的固件即使加装了NVME eth1 工作也是正常的

ufk119, 2022-11-29:
> 我的是加固态就不行 去掉就好了

兄台有解决吗？
我目前也是，只要加装了nvme，eth1就无法使用，显示全双工，但是灯不亮，也获取不到ip。
使用openwrt编译的固件以及iStoreOS（也是基于openwrt）貌似都是可以的。

shen-lan, 2022-11-29:
> ```shell
> opkg install pciutils
> ```
```
[root@OpenWrt:12:34 AM ~] # lspci|grep -i 'eth'
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:11:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```
@DHDAXCW  R5S  eth2无法获取正常ip,  加了nvme固态，插上获取的是169开头的ip,   lspci打印如上

wongwenj, 2022-12-12:
楼主这个问题解决了吗？我是R68s，碰到了和你一模一样的状况，用大雕源码5.15和6.0内核就会丢失eth3网口，内核日志也一样

duojie-sys, 2022-12-19:
现在内核已经更新到6.1了，可是固态nvme和网卡LAN1口的冲突问题依旧没有解决。大佬们都不看故障反馈吗？

donbowman, 2022-12-21:
https://github.com/coolsnowwolf/lede/commit/c4be33b792e926c7c83ae41aa704d0010de0efb5

```
rockchip: force enable rk3568 msi interrupt
Since the interrupt of RK3568 has hardware bug,
we need a workaround to enable msi interrupt.
Without this, the r8125 will not be stable.
```

solves i think.

ufk119, 2023-01-01:
> [c4be33b](https://github.com/coolsnowwolf/lede/commit/c4be33b792e926c7c83ae41aa704d0010de0efb5)
> 
> ```
> rockchip: force enable rk3568 msi interrupt
> Since the interrupt of RK3568 has hardware bug,
> we need a workaround to enable msi interrupt.
> Without this, the r8125 will not be stable.
> ```
> 
> solves i think.

yes it is，thank you!

ufk119, 2023-01-01:
> [c4be33b](https://github.com/coolsnowwolf/lede/commit/c4be33b792e926c7c83ae41aa704d0010de0efb5)
>
> ```
> rockchip: force enable rk3568 msi interrupt
> Since the interrupt of RK3568 has hardware bug,
> we [...]

yes it is，thank you!

ufk119, 2023-01-01:
> [c4be33b](https://github.com/coolsnowwolf/lede/commit/c4be33b792e926c7c83ae41aa704d0010de0efb5)
>
> ```
> rockchip: force enable rk3568 msi interrupt
> Since the interrupt of RK3568 has hardware bug,
> we [...]

yes it is，thank you!
我今天（2023-1-1）编译的固件中间的网卡是可以正常工作的了，即使插着nvm固态的情况下。

closed 2023-01-02 by aiamadeus

lysgwl, 2024-09-29:
各位设备r5s网卡eth2不识别问题，大家都解决了吗？我新买的设备同样也出这样的问题，刷友善的官方固件还有istoreos固件问题就不存在。大家都最后怎么解决的呀？