coolsnowwolf/lede #10133: UPNP没办法使用

jsdfhasuh, 2022-09-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
UPNP没办法打开端口，也无法发送数据包
### (1) 具体问题

A：UPNP无法使用

### (2) 路由器型号和固件版本

A：
hkbox
OpenWrt R22.8.18 / LuCI Master (git-22.236.14286-082687d)

### (3) 详细日志

A：
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=478, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=460, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=446, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=466, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=470, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=398, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:10 2022 daemon.err miniupnpd[2589]: try_sendto failed to send 11 packets
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=478, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=460, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=446, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=466, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=470, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=398, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:46:40 2022 daemon.err miniupnpd[2589]: try_sendto failed to send 11 packets
Sat Sep 17 10:47:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:47:09 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
[... 111 lines cut ...]
Sat Sep 17 10:49:10 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=398, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:10 2022 daemon.err miniupnpd[2589]: try_sendto failed to send 11 packets
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:39 2022 daemon.err miniupnpd[2589]: sendto(udp_notify=6, 192.168.1.1): Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=478, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=460, dest=239.255.255.250:1900): sendto: Required key not available
[... cut ...]
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=470, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto(sock=6, len=398, dest=239.255.255.250:1900): sendto: Required key not available
Sat Sep 17 10:49:40 2022 daemon.err miniupnpd[2589]: try_sendto failed to send 11 packets

xiaojing110, 2022-09-17:
5.19.9  同样的问题

coolsnowwolf, 2022-09-17:
ipv6 的问题

closed 2022-09-17 by coolsnowwolf