coolsnowwolf/lede #10139: 这是什么情况。我自己添加了AX6 编译后涮进去就这样。

gtcar2020, 2022-09-17:
Thu Sep 15 15:36:54 2022 user.info kernel: [ 9.294393] procd: - early -
Thu Sep 15 15:36:54 2022 user.info kernel: [ 9.294514] procd: - watchdog -
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.561776] mtdblock: MTD device '0:sbl1' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.579551] mtdblock: MTD device '0:mibib' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.597029] mtdblock: MTD device 'crash' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.615668] mtdblock: MTD device 'crash_syslog' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.634297] mtdblock: MTD device 'rootfs' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.652922] mtdblock: MTD device 'rsvd0' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.671545] mtdblock: MTD device '0:qsee' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.689030] mtdblock: MTD device '0:devcfg' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.706452] mtdblock: MTD device '0:rpm' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.723898] mtdblock: MTD device '0:cdt' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.741332] mtdblock: MTD device '0:appsblenv' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.758775] mtdblock: MTD device '0:appsbl' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.776211] mtdblock: MTD device '0:art' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.793646] mtdblock: MTD device 'bdata' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.845746] mtdblock: MTD device '0:sbl1' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.863522] mtdblock: MTD device '0:mibib' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.880945] mtdblock: MTD device 'crash' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.899580] mtdblock: MTD device 'crash_syslog' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.918191] mtdblock: MTD device 'rootfs' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.936774] mtdblock: MTD device 'rsvd0' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.955405] mtdblock: MTD device '0:qsee' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.972816] mtdblock: MTD device '0:devcfg' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 9.990210] mtdblock: MTD device '0:rpm' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 10.007621] mtdblock: MTD device '0:cdt' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 10.025043] mtdblock: MTD device '0:appsblenv' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 kern.warn kernel: [ 10.042435] mtdblock: MTD device '0:appsbl' is NAND, please consider using UBI block devices instead.
Thu Sep 15 15:36:54 2022 user.info kernel: [ 10.049032] procd: - watchdog -

lovehackintosh, 2022-09-17:
你用这个仓库吧https://github.com/lovehackintosh/lede/tree/mi-ipq807x

gtcar2020, 2022-09-17:
好的，谢谢。

gtcar2020, 2022-09-17:
重新编译好浪费时间，能告诉我应该改哪里就可以支持ubi了吗

lovehackintosh, 2022-09-17:
你是怎么添加的支持，已经编译成二进制文件没办法改了

gtcar2020, 2022-09-17:
我就是看原先drop的时候修改的文件，我再修改回来，再把删掉的文件再添加回去。https://github.com/coolsnowwolf/lede/commit/edbd8d2e9839357f3a4f0a06174d243f362b1544

gtcar2020, 2022-09-17:
可以编译成功，就是刷进去好像已经无法支持UBI文件系统识别。

Cathgao, 2022-09-17:
https://github.com/Cathgao/AX6-Lean/actions
下最新编译的试试

gtcar2020, 2022-09-17:
好啊，谢谢了。

gtcar2020, 2022-09-17:
> 你用这个仓库吧https://github.com/lovehackintosh/lede/tree/mi-ipq807x
你的里面有两个补丁有问题：900-arm64-dts-add-OpenWrt-DTS-files.patch ，这个文件补丁块长度不对，你检查一下，长度应该是11.

closed 2022-09-17 by aiamadeus

lovehackintosh, 2022-09-17:
> > 你用这个仓库吧https://github.com/lovehackintosh/lede/tree/mi-ipq807x
> > 你的里面有两个补丁有问题：900-arm64-dts-add-OpenWrt-DTS-files.patch ，这个文件补丁块长度不对，你检查一下，长度应该是11.

非常感谢，已改正

gtcar2020, 2022-09-18:
> 你用这个仓库吧https://github.com/lovehackintosh/lede/tree/mi-ipq807x

lede/target/linux/ipq807x/files-5.15/arch/arm64/boot/dts/qcom/ipq8071-ax3600.dtsi引用了ipq8074-memory-512m.dtsi，但是文件夹下没有这个文件，只有ipq8074-512m.dtsi，应该是引用这个文件吗？另外所引用的ipq8074.dtsi也不存在。