coolsnowwolf/lede #10143: backport-5.10 and error with 411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch

TurboBlaze, 2022-09-17:
Hi to everyone!
I have tried to build LEDE for ramips_mt7621_DEVICE_beeline_smartbox-turbo-plus with kernel 5.10 from source, but get this error:

Applying /tmp/lede/target/linux/generic/backport-5.10/**411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch** using plaintext:
patching file drivers/mtd/parsers/Kconfig
patching file drivers/mtd/parsers/Makefile
**The next patch would create the file drivers/mtd/parsers/scpart.c,
which already exists!  Applying it anyway.
patching file drivers/mtd/parsers/scpart.c
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- saving rejects to file drivers/mtd/parsers/scpart.c.rej
Patch failed!  Please fix /tmp/lede/target/linux/generic/backport-5.10/411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch!**
make[4]: *** [Makefile:30: /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.143/.prepared_1b369de61d64e25a9131cddbf20ff4d9] Error 1
make[4]: Leaving directory '/tmp/lede/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/tmp/lede/target/linux'
time: target/linux/compile#5.63#3.90#9.19
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/tmp/lede'
make[1]: *** [target/Makefile:23: /tmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/tmp/lede'
make: *** [/tmp/lede/include/toplevel.mk:230: world] Ошибка 2

Screenshot: https://i.imgur.com/TaX6XIV.png
Log; https://pastebin.com/dkxKTwKF

Can anybody to fix it?

Thanks.

coolsnowwolf, 2022-09-17:
remove 411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch due to upstream patch which already exists

TurboBlaze, 2022-09-18:
I removed 411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch and get another errors:

Applying /tmp/lede/target/linux/generic/pending-5.10/435-mtd-add-routerbootpart-parser-config.patch using plaintext:
patching file drivers/mtd/parsers/Kconfig
Hunk #1 FAILED at 204.
1 out of 1 hunk FAILED -- saving rejects to file drivers/mtd/parsers/Kconfig.rej
patching file drivers/mtd/parsers/Makefile
Hunk #1 succeeded at 13 with fuzz 1 (offset -1 lines).
Patch failed! Please fix /tmp/lede/target/linux/generic/pending-5.10/435-mtd-add-routerbootpart-parser-config.patch!
make[3]: *** [Makefile:122: /tmp/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/linux-5.10.143/.prepared] Error 1
make[3]: Leaving directory '/tmp/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#6.15#3.65#88.88
ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/tmp/lede'
make[1]: *** [toolchain/Makefile:93: /tmp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/tmp/lede'
make: *** [/tmp/lede/include/toplevel.mk:230: world] Ошибка 2

Screenshot https://i.imgur.com/XZ5Kq1u.png

Then I tried **don't remove** 411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch
I removed only lede/target/linux/ramips/files/drivers/mtd/parsers/[scpart.c](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/files/drivers/mtd/parsers/scpart.c) <<<
and apply fix in lede/tree/master/target/linux/ramips/dts/[mt7621_beeline_smartbox-turbo-plus.dts](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/dts/mt7621_beeline_smartbox-turbo-plus.dts)

My fix is here >>> https://pastebin.com/tmgyQWMz <<<

I applied this hack from openwrt-22.03.0-beeline_smartbox-turbo-plus.patch - https://gist.github.com/inoremap/9f846a625d1573b33024b40898ed4c6c

Now I am stuck on this error with drivers/mtd/composite/virt_concat
Full log of error https://pastebin.com/x08qCibW
Screenshot https://i.imgur.com/6V0uKhf.png

Any solution are welcome!

muyeyifeng, 2022-09-19:
I also encountered the same problem when compiling MT7628.
I don't know the principle and process of compilation, but based on the above discussion, I made the following attempts and successfully completed the compilation. (Note: I have not verified that the compiled system works!)

Option I:
'--- /dev/null' on line 51 is changed to '--- a/drivers/mtd/parsers/scpart.c'.

Option II:
Delete "411-v6.0-mtd-parsers-add-support-for-Sercomm-partitions.patch" line 51 and everything after.

The compilation can be completed after modification by these two methods, but unfortunately I have no time to verify whether the modification method is correct on the machine.

aiamadeus, 2022-09-19:
However, our mips here only support the 5.4 kernel.

commit 3929a988 (2022-10-07): kernel/5.4: backport mtd parser for Sercomm parts

Fixes: #10143
closed 2022-10-07 by aiamadeus