coolsnowwolf/lede #10145: samba4或400M Root filesystem partition size导致编译时间超长、存储空间不足？

fxxccp, 2022-09-18:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

最近几天云编译老是遇到编译时间超6小时或存储空间不足的问题，make distclean后依旧。以前同样的插件、同样的配置一直没问题。记得是大雕将samba替换为samba4和把Root filesystem partition size调整为400M后才出现的，尝试修改为samba和128M Root filesystem partition size仍无解。

### (2) 路由器型号和固件版本

A：X86

ecrasy, 2022-09-18:
400M都不够
你还改为128M
你这是倒着来啊
L大的源码
所有包都编译打包
我的x86云编译里需要把Size设置到960

Cathgao, 2022-09-18:
root size又不占用编译时间。
前段时间开了llvm，那玩意比node+系统本体编译时间都要长，后面取消掉了。现在你还是超时那就是插件太多，action的双核不足以6小时编译完毕。
试试pr里的缓存工具链

ecrasy, 2022-09-19:
我挺久之前也碰到这个问题
当时我是编译L大源码里所有的package
然后就编译超时（6小时限制）
我的做法是：
1. 清理不需要的软件和垃圾临时文件
2. 将workflow分解为多个jobs：
- [编译工具链toolchain，首次编译大概耗时1小时+，toolchain保存至release重复使用] 
- [编译源码，编译大概耗时4-5小时] 
- [打包固件上传至release，过程耗时大概20分钟] 

三个jobs如果集成一个job，大概率超时（6小时限制）  
拆分之后，目前还没遇到job超时的。
希望对你有帮助。

closed 2022-09-19 by aiamadeus

TaoRone, 2022-09-23:
> 我挺久之前也碰到这个问题 当时我是编译L大源码里所有的package 然后就编译超时（6小时限制） 我的做法是：
> 
> 1. 清理不需要的软件和垃圾临时文件
> 2. 将workflow分解为多个jobs：
> 
> * [编译工具链toolchain，首次编译大概耗时1小时+，toolchain保存至release重复使用]
> * [编译源码，编译大概耗时4-5小时]
> * [打包固件上传至release，过程耗时大概20分钟]
> 
> 三个jobs如果集成一个job，大概率超时（6小时限制） 拆分之后，目前还没遇到job超时的。 希望对你有帮助。

请问大佬您分解jobs有没有可参考的教程，我现在就遇到了超过6小时问题……很难受……刚接触actions不太懂