coolsnowwolf/lede #10147: 最近upnp服务好像有些异常

AutoCONFIG, 2022-09-18:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：upnp功能在比特彗星软件上无法添加（具体为：UPNP device not found!），无论是IGDv1还是IGDv2。
miniupnpd组件用的是2017年的版本，到现在都没有更新过，不知道是不是出于兼容还是其他原因，能否可以更新一下miniupnpd呢？

### (2) 路由器型号和固件版本

A：x86-64(OpenWrt R22.9.1 / LuCI Master (git-22.255.14339-1ee6726))

### (3) 详细日志

A：Sun Sep 18 16:43:37 2022 kern.info kernel: [26494.741538] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:43:37 2022 kern.info kernel: [26494.747638] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:14 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:14 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:14 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:17 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:17 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:17 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:17 2022 daemon.notice miniupnpd[2685]: upnp_event_send: send(): Connection reset by peer
Sun Sep 18 16:45:23 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:23 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:23 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 16:45:23 2022 kern.warn kernel: [26601.009316] net_ratelimit: 1050 callbacks suppressed
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.009329] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.020587] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.026765] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.032868] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.039041] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.045220] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.051311] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.057399] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.063489] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:23 2022 kern.info kernel: [26601.069582] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.warn kernel: [26606.709378] net_ratelimit: 945 callbacks suppressed
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.709390] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.720513] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.726728] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.732964] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.739180] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.745381] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.751584] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.757850] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.764013] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:45:29 2022 kern.info kernel: [26606.770307] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 16:48:17 2022 daemon.err uhttpd[1825]: cat: can't open '/etc/bench.log': No such file or directory
Sun Sep 18 17:15:23 2022 kern.warn kernel: [28401.289480] net_ratelimit: 211 callbacks suppressed
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.289492] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.300858] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.307098] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.313286] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.319493] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.326851] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.333150] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.339398] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.345874] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:23 2022 kern.info kernel: [28401.352125] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:15:43 2022 daemon.err uhttpd[1825]: cat: can't open '/etc/bench.log': No such file or directory
Sun Sep 18 17:15:43 2022 daemon.err uhttpd[1825]: cat: can't open '/etc/bench.log': No such file or directory
Sun Sep 18 17:15:50 2022 daemon.err uhttpd[1825]: sh: out of range
Sun Sep 18 17:15:50 2022 daemon.err uhttpd[1825]: sh: out of range
Sun Sep 18 17:17:01 2022 daemon.err uhttpd[1825]: sh: out of range
Sun Sep 18 17:17:01 2022 daemon.err uhttpd[1825]: sh: out of range
Sun Sep 18 17:17:13 2022 kern.warn kernel: [28510.854984] net_ratelimit: 1393 callbacks suppressed
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.854996] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.866214] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.872307] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.878424] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.884593] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.890746] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.896901] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.903032] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.909170] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:17:13 2022 kern.info kernel: [28510.915355] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.warn kernel: [28629.823870] net_ratelimit: 1619 callbacks suppressed
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.823882] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.835237] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.841451] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.847632] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.853822] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.861147] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.869640] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.875808] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.881921] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:19:12 2022 kern.info kernel: [28629.890328] neighbour: ndisc_cache: neighbor table overflow!
Sun Sep 18 17:20:46 2022 daemon.err miniupnpd[2685]: Failed to get ip address for interface
Sun Sep 18 17:21:16 2022 daemon.err miniupnpd[2685]: recv (state0): Connection reset by peer

firkerword, 2022-09-18:
昨天不是刚回滚吗

AutoCONFIG, 2022-09-18:
> 昨天不是刚回滚吗

我这边用的是回滚后编译的固件，用了官方的miniupnpc测试了也是提示“No IGD UPnP Device found on the network !”
怀疑是固件这边的问题。
lean大佬的miniupnpd版本也确实太老旧了，而且以前upnp也会随缘出现问题。
upnp算是路由器基本功能之一，也希望lean大这边的能适当保持更新和稳定。

star-o0O, 2022-09-18:
> > 昨天不是刚回滚吗
> 
> 我这边用的是回滚后编译的固件，用了官方的miniupnpc v2.1测试了也是提示“No IGD UPnP Device found on the network !” 怀疑是固件这边的问题。 lean大佬的miniupnpd版本也确实太老旧了，而且以前upnp也会随缘出现问题。 upnp算是路由器基本功能之一，也希望lean大这边的能适当保持更新和稳定。

如果我没记错，国内许多下载类以及使用到UPNP的软件不兼容新版

minlang112, 2022-09-18:
和upnp版本无关，别的问题

Cathgao, 2022-09-18:
qbitorrent，utorrent，弹弹play，网心云一个都没问题。你是不是应该怀疑下是比特彗星的问题？

coolsnowwolf, 2022-09-18:
> > 昨天不是刚回滚吗
> 
> 我这边用的是回滚后编译的固件，用了官方的miniupnpc v2.1测试了也是提示“No IGD UPnP Device found on the network !” 怀疑是固件这边的问题。 lean大佬的miniupnpd版本也确实太老旧了，而且以前upnp也会随缘出现问题。 upnp算是路由器基本功能之一，也希望lean大这边的能适当保持更新和稳定。

你错了，这个是唯一兼容国内大多数应用的 upnp 版本，甚至我们还打补丁支持了 5.x 系列。 
bump 个新版本比什么都容易，找到个适当大多数人场景的并维护却是很难的

AutoCONFIG, 2022-09-18:
> qbitorrent，utorrent，弹弹play，网心云一个都没问题。你是不是应该怀疑下是比特彗星的问题？

其实我也有考虑过是比特彗星本身的问题，因为我自己用迅雷测试过是可以正常添加映射的。之后在比特彗星论坛上询问了下开发组（可能是），他们说比特彗星用的是miniupnpc源码实现的，支持igdv1和igdv2，于是自己用miniupnpc做了下测试，就是提示“No IGD UPnP Device found on the network !”。所以怀疑是不是lean固件的miniupnpd版本比较旧，对新版miniupnpc支持不太好。

AutoCONFIG, 2022-09-18:
> > > 昨天不是刚回滚吗
> > 
> > 
> > 我这边用的是回滚后编译的固件，用了官方的miniupnpc v2.1测试了也是提示“No IGD UPnP Device found on the network !” 怀疑是固件这边的问题。 lean大佬的miniupnpd版本也确实太老旧了，而且以前upnp也会随缘出现问题。 upnp算是路由器基本功能之一，也希望lean大这边的能适当保持更新和稳定。
> 
> 你错了，这个是唯一兼容国内大多数应用的 upnp 版本，甚至我们还打补丁支持了 5.x 系列。 bump 个新版本比什么都容易，找到个适当大多数人场景的并维护却是很难的

感谢大佬的解答，我也十分赞同合适且稳定的版本比一味追求更新要好。

我猜测是比特彗星的miniupnpc版本比较新，与lede的miniupnpd出现一些兼容问题，毕竟也不可能做到100%兼容。

其实我这个100开头的内网IP用户，开不开upnp都没差别。只是目前发现了这个bug，两边都给反馈一下，尽可能对bug修复有些帮助。

我这个版本里总是会出现“miniupnpd[2685]: Failed to get ip address for interface”的问题，也没有开多拨，这个是什么问题呢？

closed 2022-09-19 by coolsnowwolf