coolsnowwolf/lede #10148: 编译时卡在miniupnpd，不知道是不是缺少依赖

AndyLin2020, 2022-09-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：recipe for target 'netfilter/iptcrdr.o' failed

### (2) 路由器型号和固件版本

A：K3

### (3) 详细日志

A：
`
 make[4]: Entering directory '/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/miniupnpd-2.0.20170421'
please install uuid-dev package / libuuid
arm-openwrt-linux-muslgnueabi-gcc -DIPTABLES_143 -lip4tc -luuid -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -fmacro-prefix-map=/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/miniupnpd-2.0.20170421=miniupnpd-2.0.20170421 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/andy/lede/staging_dir/toolchain-arm_cortex-a9_gcc-11.3.0_musl_eabi/usr/include -I/home/andy/lede/staging_dir/toolchain-arm_cortex-a9_gcc-11.3.0_musl_eabi/include/fortify -I/home/andy/lede/staging_dir/toolchain-arm_cortex-a9_gcc-11.3.0_musl_eabi/include  -fno-strict-aliasing -fno-common -Wall -Wextra -Wstrict-prototypes -Wdeclaration-after-statement -I/home/andy/lede/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include  -D_GNU_SOURCE  -c -o netfilter/iptcrdr.o netfilter/iptcrdr.c
In file included from netfilter/iptcrdr.c:11:
/home/andy/lede/staging_dir/toolchain-arm_cortex-a9_gcc-11.3.0_musl_eabi/include/sys/errno.h:1:2: warning: #warning redirecting incorrect #include <sys/errno.h> to <errno.h> [-Wcpp]
    1 | #warning redirecting incorrect #include <sys/errno.h> to <errno.h>
      |  ^~~~~~~
netfilter/iptcrdr.c: In function 'addnatrule':
netfilter/iptcrdr.c:1119:22: error: 'NFC_IP_DST_PT' undeclared (first use in this function)
 1119 |         e->nfcache = NFC_IP_DST_PT;
      |                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1119:22: note: each undeclared identifier is reported only once for each function it appears in
netfilter/iptcrdr.c:1121:23: error: 'NFC_UNKNOWN' undeclared (first use in this function)
 1121 |         e->nfcache |= NFC_UNKNOWN;
      |                       ^~~~~~~~~~~
netfilter/iptcrdr.c: In function 'addmasqueraderule':
netfilter/iptcrdr.c:1189:22: error: 'NFC_IP_DST_PT' undeclared (first use in this function)
 1189 |         e->nfcache = NFC_IP_DST_PT;
      |                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1191:23: error: 'NFC_UNKNOWN' undeclared (first use in this function)
 1191 |         e->nfcache |= NFC_UNKNOWN;
      |                       ^~~~~~~~~~~
netfilter/iptcrdr.c: In function 'addpeernatrule':
netfilter/iptcrdr.c:1269:22: error: 'NFC_IP_DST_PT' undeclared (first use in this function)
 1269 |         e->nfcache = NFC_IP_DST_PT | NFC_IP_SRC_PT;
      |                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1269:38: error: 'NFC_IP_SRC_PT' undeclared (first use in this function)
 1269 |         e->nfcache = NFC_IP_DST_PT | NFC_IP_SRC_PT;
      |                                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1271:23: error: 'NFC_UNKNOWN' undeclared (first use in this function)
 1271 |         e->nfcache |= NFC_UNKNOWN;
      |                       ^~~~~~~~~~~
netfilter/iptcrdr.c: In function 'addpeerdscprule':
netfilter/iptcrdr.c:1340:22: error: 'NFC_IP_DST_PT' undeclared (first use in this function)
 1340 |         e->nfcache = NFC_IP_DST_PT | NFC_IP_SRC_PT;
      |                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1340:38: error: 'NFC_IP_SRC_PT' undeclared (first use in this function)
 1340 |         e->nfcache = NFC_IP_DST_PT | NFC_IP_SRC_PT;
      |                                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1342:23: error: 'NFC_UNKNOWN' undeclared (first use in this function)
 1342 |         e->nfcache |= NFC_UNKNOWN;
      |                       ^~~~~~~~~~~
netfilter/iptcrdr.c: In function 'add_filter_rule':
netfilter/iptcrdr.c:1423:22: error: 'NFC_IP_DST_PT' undeclared (first use in this function)
 1423 |         e->nfcache = NFC_IP_DST_PT;
      |                      ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1427:23: error: 'NFC_UNKNOWN' undeclared (first use in this function)
 1427 |         e->nfcache |= NFC_UNKNOWN;
      |                       ^~~~~~~~~~~
<builtin>: recipe for target 'netfilter/iptcrdr.o' failed
make[4]: *** [netfilter/iptcrdr.o] Error 1
make[4]: Leaving directory '/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/miniupnpd-2.0.20170421'
Makefile:83: recipe for target '/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/miniupnpd-2.0.20170421/.built' failed
make[3]: *** [/home/andy/lede/build_dir/target-arm_cortex-a9_musl_eabi/miniupnpd-2.0.20170421/.built] Error 2
make[3]: Leaving directory '/home/andy/lede/feeds/packages/net/miniupnpd'
time: package/feeds/packages/miniupnpd/compile#0.24#0.06#0.28
    ERROR: package/feeds/packages/miniupnpd failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/miniupnpd/compile' failed
make[2]: *** [package/feeds/packages/miniupnpd/compile] Error 1
make[2]: Leaving directory '/home/andy/lede'
package/Makefile:110: recipe for target '/home/andy/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/andy/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/andy/lede'
/home/andy/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

`

ecrasy, 2022-09-18:
redirecting incorrect #include <sys/errno.h> to <errno.h> [-Wcpp]
这导致iptcrdr.c编译失败了
netfilter/iptcrdr.c:1423:22: error: 'NFC_IP_DST_PT' undeclared (first use in this function)
1423 | e->nfcache = NFC_IP_DST_PT;
| ^~~~~~~~~~~~~
netfilter/iptcrdr.c:1427:23: error: 'NFC_UNKNOWN' undeclared (first use in this function)
1427 | e->nfcache |= NFC_UNKNOWN;
| ^~~~~~~~~~~
: recipe for target 'netfilter/iptcrdr.o' failed

closed 2022-09-18 by coolsnowwolf

AndyLin2020, 2022-09-19:
可能是版本更新，删除后重新git pull就正常了