coolsnowwolf/lede #10152: WSR-2533DHPL固件编译失败

Lorilandly, 2022-09-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：这个路由器型号什么都不勾选也编译报错。如果编译x86是正常的，所以环境没问题。在openwrt官方repo编译没问题，所以应该也不是上游的锅？

### (2) 路由器型号和固件版本

A：WSR-2533DHPL

### (3) 详细日志

A：
```
mipsel-openwrt-linux-musl-gcc -D__KERNEL__ -Wall -Wstrict-prototypes -Wno-trigraphs -Os -fno-strict-aliasing -fno-common -fomit-frame-pointer -G 0 -mno-abicalls -fno-pic -ffunction-sections -pipe -mlong-calls -fno-common -ffreestanding -fhonour-copts -nostartfiles -mabi=32 -march=mips32r2 -Wa,-32 -Wa,-march=mips32r2 -Wa,-mips32r2 -Wa,--trap -D_LZMA_PROB32 -flto -DCONFIG_ICACHE_SIZE="(32 * 1024)" -DCONFIG_DCACHE_SIZE="(16 * 1024)" -DCONFIG_CACHELINE_SIZE=32 -DLZMA_WRAPPER=1 -DLOADADDR=0x80001000 -c -o board-mt7621.o board-mt7621.c
mipsel-openwrt-linux-musl-gcc -D__KERNEL__ -Wall -Wstrict-prototypes -Wno-trigraphs -Os -fno-strict-aliasing -fno-common -fomit-frame-pointer -G 0 -mno-abicalls -fno-pic -ffunction-sections -pipe -mlong-calls -fno-common -ffreestanding -fhonour-copts -nostartfiles -mabi=32 -march=mips32r2 -Wa,-32 -Wa,-march=mips32r2 -Wa,-mips32r2 -Wa,--trap -D_LZMA_PROB32 -flto -DCONFIG_ICACHE_SIZE="(32 * 1024)" -DCONFIG_DCACHE_SIZE="(16 * 1024)" -DCONFIG_CACHELINE_SIZE=32 -DLZMA_WRAPPER=1 -DLOADADDR=0x80001000 -c -o printf.o printf.c
mipsel-openwrt-linux-musl-gcc -D__KERNEL__ -Wall -Wstrict-prototypes -Wno-trigraphs -Os -fno-strict-aliasing -fno-common -fomit-frame-pointer -G 0 -mno-abicalls -fno-pic -ffunction-sections -pipe -mlong-calls -fno-common -ffreestanding -fhonour-copts -nostartfiles -mabi=32 -march=mips32r2 -Wa,-32 -Wa,-march=mips32r2 -Wa,-mips32r2 -Wa,--trap -D_LZMA_PROB32 -flto -DCONFIG_ICACHE_SIZE="(32 * 1024)" -DCONFIG_DCACHE_SIZE="(16 * 1024)" -DCONFIG_CACHELINE_SIZE=32 -DLZMA_WRAPPER=1 -DLOADADDR=0x80001000 -c -o LzmaDecode.o LzmaDecode.c
mipsel-openwrt-linux-musl-ld -r -b binary --oformat elf32-tradlittlemips -T lzma-data.lds -o data.o /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin
mipsel-openwrt-linux-musl-gcc -D__KERNEL__ -Wall -Wstrict-prototypes -Wno-trigraphs -Os -fno-strict-aliasing -fno-common -fomit-frame-pointer -G 0 -mno-abicalls -fno-pic -ffunction-sections -pipe -mlong-calls -fno-common -ffreestanding -fhonour-copts -nostartfiles -mabi=32 -march=mips32r2 -Wa,-32 -Wa,-march=mips32r2 -Wa,-mips32r2 -Wa,--trap -D_LZMA_PROB32 -flto -DCONFIG_ICACHE_SIZE="(32 * 1024)" -DCONFIG_DCACHE_SIZE="(16 * 1024)" -DCONFIG_CACHELINE_SIZE=32 -DLZMA_WRAPPER=1 -DLOADADDR=0x80001000 -static -Wl,--gc-sections -Wl,-no-warn-mismatch -Wl,-e,startup -T loader.lds -Wl,-Ttext,0x81800000 -flto -fwhole-program -Wl,-z,max-page-size=4096 -o loader head.o loader.o cache.o board-mt7621.o printf.o LzmaDecode.o data.o
mipsel-openwrt-linux-musl-objcopy -O binary -R .reginfo -R .note -R .comment -R .mdebug -R .MIPS.abiflags -S loader loader.bin
mipsel-openwrt-linux-musl-ld -r -b binary --oformat elf32-tradlittlemips -o loader2.o loader.bin
mipsel-openwrt-linux-musl-ld -e startup -T loader2.lds -Ttext 0x80001000 -z max-page-size=4096 -o loader.elf loader2.o
make[7]: Leaving directory '/home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.src'
make[6]: Leaving directory '/home/lori/Documents/lede/target/linux/ramips/image/lzma-loader'
mv "/home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.bin" "/home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin"
rm -rf /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.src
mkimage -A mips -O linux -T kernel -C none -a 0x80001000 -e 0x80001000 -n 'MIPS OpenWrt Linux-5.4.211'   -d /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.new
Image Name:   MIPS OpenWrt Linux-5.4.211
Created:      Sun Sep 18 13:42:25 2022
Image Type:   MIPS Linux Kernel Image (uncompressed)
Data Size:    8415329 Bytes = 8218.09 KiB = 8.03 MiB
Load Address: 80001000
Entry Point:  80001000
mv /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.new /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin
make[5]: Leaving directory '/home/lori/Documents/lede/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/home/lori/Documents/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/lori/Documents/lede/target/linux'
time: target/linux/install#41.47#8.60#24.54
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/lori/Documents/lede'
make[1]: *** [target/Makefile:24: /home/lori/Documents/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/lori/Documents/lede'
make: *** [/home/lori/Documents/lede/include/toplevel.mk:230: world] Error 2
```

coolsnowwolf, 2022-09-19:
-j1 V=s

coolsnowwolf, 2022-09-19:
看起来是内核太大超过这个分区的大小了

coolsnowwolf, 2022-09-19:
[image]

固件分区最大 7.9MB，你这 8MB 了

aiamadeus, 2022-09-19:
你是选了啥导致内核这么大

Lorilandly, 2022-09-20:
我应该是什么都没选，全默认。貌似默认就勾选了挺多东西的？我对lede选项还不是很熟悉。我再去仔细检查一下勾选的选项
感谢提供建议

Lorilandly, 2022-09-20:
我的最终目的是希望能在这个路由器上编译使用mt7615闭源驱动的固件。参考这篇issue #5897。想请教一下我的思路是否正确：

在默认选项的基础上
- 取消勾选
```
kmod-mt7615-firmware
kmod-mt7615e
```
- 勾选
```
kmod-mt7615d
mt-wifi
luci-app-mtwifi
```

Lorilandly, 2022-09-26:
Retried after deselecting loads of default options. The kernel should be under 7936k now. The error somehow still persists.
```
mipsel-openwrt-linux-musl-ld -e startup -T loader2.lds -Ttext 0x80001000 -z max-page-size=4096 -o loader.elf loader2.o
make[7]: Leaving directory '/home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.src'
make[6]: Leaving directory '/home/lori/Documents/lede/target/linux/ramips/image/lzma-loader'
mv "/home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.bin" "/home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin"
rm -rf /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.src
mkimage -A mips -O linux -T kernel -C none -a 0x80001000 -e 0x80001000 -n 'MIPS OpenWrt Linux-5.4.203'   -d /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.new
Image Name:   MIPS OpenWrt Linux-5.4.203
Created:      Sat Jul 16 09:28:21 2022
Image Type:   MIPS Linux Kernel Image (uncompressed)
Data Size:    8001729 Bytes = 7814.19 KiB = 7.63 MiB
Load Address: 80001000
Entry Point:  80001000
mv /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin.new /home/lori/Documents/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-buffalo_wsr-2533dhpl-initramfs-kernel.bin
make[5]: Leaving directory '/home/lori/Documents/lede/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/home/lori/Documents/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/lori/Documents/lede/target/linux'
time: target/linux/install#39.04#7.61#22.23
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/lori/Documents/lede'
make[1]: *** [target/Makefile:19: /home/lori/Documents/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/lori/Documents/lede'
make: *** [/home/lori/Documents/lede/include/toplevel.mk:230: world] Error 2
```

closed 2026-09-05 by coolsnowwolf