coolsnowwolf/lede #10159: ipq806x编译make download qca-nss-ecm 下载不了

WowRouter, 2022-09-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ipq806x编译默认沛喆make download qca-nss-ecm 下载不了

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A： make[2] -C package/qca/nss/qca-nss-clients download
 make[2] -C package/qca/nss/qca-nss-drv download
 make[2] -C package/qca/nss/qca-nss-ecm download
    ERROR: package/lean/shortcut-fe/simulated-driver failed to build.
    ERROR: package/qca/nss/qca-nss-clients failed to build.
    ERROR: package/qca/nss/qca-nss-ecm failed to build (build variant: standard).
    ERROR: package/qca/nss/qca-nss-drv failed to build.

moonerick, 2022-09-20:
建议后面加个V=s -j1 看看究竟是什么下载失败了，然后手动去下载来放到dl文件夹在编译，最近梯子非常不稳定，稍微大一点的包会出问题……

WowRouter, 2022-09-20:
> 建议后面加个V=s -j1 看看究竟是什么下载失败了，然后手动去下载来放到dl文件夹在编译，最近梯子非常不稳定，稍微大一点的包会出问题……
https://sources.openwrt.org/qca-nss-crypto-e7651c29.tar.xz
手动下载也不行

closed 2022-09-22 by coolsnowwolf