coolsnowwolf/lede #10160: 请问默认设置编译的插件在编译的时候，在make menuconfig是不是取消勾选也会被编译？

yanxin152133, 2022-09-20:
比如这个`luci-app-adbyby-plus`，在`make menuconfig`的时候取消勾选，
[image]

但是默认设置还是有的
[image]

hzjnet, 2022-09-20:
> 比如这个`luci-app-adbyby-plus`，在`make menuconfig`的时候取消勾选， [image]
> 
> 但是默认设置还是有的 [image]

不会的

yanxin152133, 2022-09-20:
> > 比如这个`luci-app-adbyby-plus`，在`make menuconfig`的时候取消勾选， [image]
> > 但是默认设置还是有的 [image]
> 
> 不会的
我再试试，好几次编译这个插件都还在

hzjnet, 2022-09-22:
> > > 比如这个`luci-app-adbyby-plus`，在`make menuconfig`的时候取消勾选， [image]
> > > 但是默认设置还是有的 [image]
> > 
> > 
> > 不会的
> > 我再试试，好几次编译这个插件都还在

那是其它地方还有该配置，比如：target/linux/ipq896x/Makefile

GreatMichaelLee, 2022-09-23:
我也发现你这个问题，我一般都是采用暴力方式，直接编辑.config然后直接搜索相关的包，全部注释掉。这样才算干净，不然总是有狗皮膏药粘着。

closed 2023-02-28 by aiamadeus