coolsnowwolf/lede #10161: ERROR: package/network/services/e2guardian failed to build.

zhangguanzhang, 2022-09-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.v3.2.0.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "v3.2.0.tar.gz" "cae011aacf6b250fc021c861fe46e15310349c474bca4553c0f829620e53e3bf" "" "https://github.com/e2guardian/e2guardian/archive/"    '
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/.built_check
CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0=e2guardian-3.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0=e2guardian-3.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro " make  -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0=e2guardian-3.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[3]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0'
make  all-recursive
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0'
Making all in .
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0'
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0'
Making all in doc
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/doc'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/doc'
Making all in data
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data'
Making all in languages
make[6]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data/languages'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data/languages'
Making all in .
make[6]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data'
make[6]: Nothing to be done for 'all-am'.
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data'
Making all in scripts
make[6]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data/scripts'
Making all in .
make[7]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data/scripts'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data/scripts'
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data/scripts'
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/data'
Making all in configs
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs'
Making all in lists
make[6]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists'
Making all in phraselists
make[7]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists/phraselists'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists/phraselists'
Making all in .
make[7]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists'
Making all in authplugins
make[7]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists/authplugins'
Making all in .
make[8]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/configs/lists/authplugins'
make[8]: Nothing to be done for 'all-am'.
[... 46 lines cut ...]
BaseSocket.hpp:54:72: error: ISO C++17 does not allow dynamic exception specifications
   54 |     void readyForOutput(int timeout, bool honour_reloadconfig = false) throw(std::exception);
      |                                                                        ^~~~~
BaseSocket.hpp:57:132: error: ISO C++17 does not allow dynamic exception specifications
   57 |     int getLine(char *buff, int size, int timeout, bool honour_reloadconfig = false, bool *chopped = NULL, bool *truncated = NULL) throw(std::exception);
      |                                                                                                                                    ^~~~~
BaseSocket.hpp:60:40: error: ISO C++17 does not allow dynamic exception specifications
   60 |     void writeString(const char *line) throw(std::exception);
      |                                        ^~~~~
BaseSocket.hpp:68:119: error: ISO C++17 does not allow dynamic exception specifications
   68 |     void writeToSockete(const char *buff, int len, unsigned int flags, int timeout, bool honour_reloadconfig = false) throw(std::exception);
      |                                                                                                                       ^~~~~
make[5]: *** [Makefile:692: e2guardian-FDTunnel.o] Error 1
make[4]: *** [Makefile:400: all-recursive] Error 1
make[3]: *** [Makefile:[341](https://github.com/zhangguanzhang/Actions-OpenWrt/actions/runs/3087478169/jobs/4992885758#step:18:342): all] Error 2
make[2]: *** [Makefile:86: /workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/.built] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0/src'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0'
make[3]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/e2guardian-3.2.0'
make[2]: Leaving directory '/workdir/openwrt/package/network/services/e2guardian'
time: package/network/services/e2guardian/compile#0.38#0.14#0.60
    ERROR: package/network/services/e2guardian failed to build.
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:116: package/network/services/e2guardian/compile] Error 1
```

ecrasy, 2022-09-20:
# 修正e2guardian编译错误
sed -i 's/-fno-rtti/-fno-rtti -std=c++14/g' package/network/services/e2guardian/Makefile
echo "Fixing e2guardian compile error!!!"

我的解决方法

zhangguanzhang, 2022-09-21:
> # 修正e2guardian编译错误
> sed -i 's/-fno-rtti/-fno-rtti -std=c++14/g' package/network/services/e2guardian/Makefile echo "Fixing e2guardian compile error!!!"
> 
> 我的解决方法

感谢，我试下

zhangguanzhang, 2022-09-22:
Fixes by: https://github.com/coolsnowwolf/lede/commit/9ee64a9f385065b32ea23353f8ea298f3f3f9a66

closed 2022-09-22 by zhangguanzhang