coolsnowwolf/lede #10171: ubuntu20.04编译K3报错

younger911, 2022-09-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译到kmod-phy-bcm-ns-usb3报错

### (2) 路由器型号和固件版本

A：K3 

### (3) 详细日志

A：
错误信息如下:
SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usb-net-hso/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usb-net-hso_installed
echo "kmod-usb-net-hso" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usb-hid/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usb-hid_installed
echo "kmod-usb-hid" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usb-yealink/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usb-yealink_installed
echo "kmod-usb-yealink" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usbip/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usbip_installed
echo "kmod-usbip" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usbip-client/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usbip-client_installed
echo "kmod-usbip-client" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usbip-server/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usbip-server_installed
echo "kmod-usbip-server" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usbmon/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usbmon_installed
echo "kmod-usbmon" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usb3/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usb3_installed
echo "kmod-usb3" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-usb-xhci-hcd/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-usb-xhci-hcd_installed
echo "kmod-usb-xhci-hcd" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-phy-bcm-ns-usb2/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-phy-bcm-ns-usb2_installed
echo "kmod-phy-bcm-ns-usb2" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/younger/lede/tmp/.root-copy.flock -c 'cp -fpR /home/younger/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kmod-phy-bcm-ns-usb3/. /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-phy-bcm-ns-usb3_installed
echo "kmod-phy-bcm-ns-usb3" >> /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
make[3]: Leaving directory '/home/younger/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.62#0.84#4.59
make[2]: Leaving directory '/home/younger/lede'
make[1]: *** [package/Makefile:110: /home/younger/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/younger/lede'
make: *** [/home/younger/lede/include/toplevel.mk:230: world] Error 2

------------------------------------------------------------------------------------------------------------------------------------
.config配置如下:

[https://github.com/younger911/openwrtx86/blob/main/config](url)

mitsukileung, 2022-09-25:
K3支持USB3吗？

mitsukileung, 2022-09-25:
5.4内核没有问题

[image]