coolsnowwolf/lede #10178: ubuntu 20.04编译wndr4300报错

zengerlin, 2022-09-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：试着不添加插件编译、更换系统配置，都是报这个错，迷茫了

### (2) 路由器型号和固件版本

A：netgear WNDR4300编译最新版

### (3) 详细日志

A：drivers/mtd/composite/virt_concat.c: In function 'virt_concat_probe':
drivers/mtd/composite/virt_concat.c:80:9: error: implicit declaration of function 'of_get_mtd_device_by_node'; did you mean 'of_find_device_by_node'? [-Werror=implicit-function-declaration]
   mtd = of_get_mtd_device_by_node(it.node);
         ^~~~~~~~~~~~~~~~~~~~~~~~~
         of_find_device_by_node
drivers/mtd/composite/virt_concat.c:80:7: warning: assignment to 'struct mtd_info *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   mtd = of_get_mtd_device_by_node(it.node);
       ^
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:286: drivers/mtd/composite/virt_concat.o] Error 1
make[7]: *** [scripts/Makefile.build:503: drivers/mtd/composite] Error 2
make[6]: *** [scripts/Makefile.build:503: drivers/mtd] Error 2
make[5]: *** [Makefile:1832: drivers] Error 2
make[5]: Leaving directory '/home/zengerlin/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.145'
make[4]: *** [Makefile:21: /home/zengerlin/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.10.145/.modules] Error 2
make[4]: Leaving directory '/home/zengerlin/lede/target/linux/ath79'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/zengerlin/lede/target/linux'
time: target/linux/compile#424.56#63.57#471.22
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/zengerlin/lede'
make[1]: *** [target/Makefile:23: /home/zengerlin/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/zengerlin/lede'
make: *** [/home/zengerlin/lede/include/toplevel.mk:230: world] Error 2

uioasder, 2022-09-25:
重新安装下依赖就好了

closed 2022-09-26 by aiamadeus

zengerlin, 2022-09-26:
ninja-build is already the newest version (1.10.1-1).
p7zip is already the newest version (16.02+dfsg-8).
p7zip-full is already the newest version (16.02+dfsg-8).
pkgconf is already the newest version (1.8.0-1).
python3-pip is already the newest version (22.0.2+dfsg-1).
scons is already the newest version (4.0.1+dfsg-2).
swig is already the newest version (4.0.2-1ubuntu1).
texinfo is already the newest version (6.8-4build1).
uglifyjs is already the newest version (3.15.1-1).
upx-ucl is already the newest version (3.96-3).
xmlto is already the newest version (0.0.28-2.1).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
curl is already the newest version (7.81.0-1ubuntu1.4).
git is already the newest version (1:2.34.1-1ubuntu1.4).
libc6-dev-i386 is already the newest version (2.35-0ubuntu3.1).
libpython3-dev is already the newest version (3.10.6-1~22.04).
libssl-dev is already the newest version (3.0.2-0ubuntu1.6).
python3 is already the newest version (3.10.6-1~22.04).
qemu-utils is already the newest version (1:6.2+dfsg-2ubuntu6.3).
vim is already the newest version (2:8.2.3995-1ubuntu2.1).
xxd is already the newest version (2:8.2.3995-1ubuntu2.1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu9.1).
python2.7 is already the newest version (2.7.18-13ubuntu1.1).
subversion is already the newest version (1.14.1-3ubuntu0.22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

依赖已经全部都安装上了还是一样的错，哭～～～