coolsnowwolf/lede #10179: target/linux failed to build.

by-JohnChen, 2022-09-25:
内核 编译失败？
WARNING: Makefile 'package/feeds/luci/luci/Makefile' has a dependency on 'luci-theme-edge', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-mod-freifunk/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-mod-freifunk-community/Makefile' has a dependency on 'freifunk-mapupdate', which does not exist
tmp/.config-package.in:85868:error: recursive dependency detected!
tmp/.config-package.in:85868:	symbol PACKAGE_iptables is selected by PACKAGE_ip6tables
tmp/.config-package.in:85802:	symbol PACKAGE_ip6tables is selected by PACKAGE_olsrd
tmp/.config-package.in:92579:	symbol PACKAGE_olsrd is selected by PACKAGE_luci-mod-freifunk-community
tmp/.config-package.in:63240:	symbol PACKAGE_luci-mod-freifunk-community depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

#
# configuration written to .config
#
WARNING: Makefile 'package/feeds/luci/luci/Makefile' has a dependency on 'luci-theme-edge', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-mod-freifunk/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-mod-freifunk-community/Makefile' has a dependency on 'freifunk-mapupdate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci/Makefile' has a dependency on 'luci-theme-edge', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-mod-freifunk/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-mod-freifunk-community/Makefile' has a dependency on 'freifunk-mapupdate', which does not exist
mt7621.mk:462: *** empty variable name.  Stop.
make[5]: *** [Makefile:23: image-prereq] Error 2
make[4]: *** [Makefile:11: prereq] Error 2
make[3]: *** [target/Makefile:30: target/linux/prereq] Error 1
make[2]: *** [target/Makefile:22: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_prereq] Error 2
make[1]: *** [/workdir/openwrt/include/toplevel.mk:208: prereq] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:2[28](https://github.com/by-JohnChen/OpenWrt-Newifi_D2/actions/runs/3122619060/jobs/5064676104#step:10:29): tools/flock/compile] Error 2
time: target/linux/prereq#0.12#0.04#0.15
    ERROR: target/linux failed to build.

aiamadeus, 2022-09-25:
这个确定不是你自己改错了啥？

by-JohnChen, 2022-09-25:
> 这个确定不是你自己改错了啥？

貌似是改错了-.-

closed 2022-09-28 by aiamadeus