coolsnowwolf/lede #10189: 更新到最新源码后 package/feeds/packages/jq failed to build

mitsukileung, 2022-09-28:
K3

[image]

AXBNG, 2022-09-28:
https://github.com/coolsnowwolf/packages/pull/543

等不及就自己改一下

zwillhill, 2022-09-28:
同是k3，云编译，rrdtool也是报hash错误
`Download Results:
gid   |stat|avg speed  |path/URI
======+====+===========+=======================================================
1c9e44|OK  |   5.3MiB/s|/workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_BSuy7Ha6bZ

Status Legend:
(OK):download completed.
Hash of the downloaded file does not match (file: 88dcddf2a92b9dc9243e929340fdc6f3431b779e22438093d7a31a400f1c3909, requested: 42aa7c213dedbd95d33ca84d92f4187880f7e96062c6a3fb05bfb16f77ba2a91) - deleting download.
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/rrdtool1'
No more mirrors to try - giving up.
time: package/feeds/packages/rrdtool1/compile#0.18#0.03#0.95
make[3]: *** [Makefile:134: /workdir/openwrt/dl/rrdtool-1.0.50.tar.gz] Error 2
    ERROR: package/feeds/packages/rrdtool1 failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/feeds/packages/rrdtool1/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.`

mitsukileung, 2022-09-28:
刚刚试了把HASH换成提示的哪个可以编译成功，楼上可以试试

closed 2022-09-28 by mitsukileung

bradfordzhang, 2022-10-04:
#10195