coolsnowwolf/lede #10195: JQ hash mismatch?

GreatMichaelLee, 2022-09-29:
rm -rf ./tmp 后也还是报这个错

mkdir -p /home/builder/lede_x86/dl
SHELL= flock /home/builder/lede_x86/tmp/.iputils-20211215.tar.gz.flock -c '     /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "iputils-20211215.tar.gz" "b6f67fc705490673ff4471d006221b4a2f1b1180b929d9fefd771352621ccedf" "" "https://codeload.github.com/iputils/iputils/tar.gz/20211215?"    '
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/iputils-20211215/.pkgdir/iputils-arping/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.iputils-arping_installed
echo "iputils-arping" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/iputils.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/iputils'
time: package/feeds/packages/iputils/compile#0.18#0.12#0.26
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/utils/jq'
mkdir -p /home/builder/lede_x86/dl
SHELL= flock /home/builder/lede_x86/tmp/.jq-1.6.tar.gz.flock -c '       /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "jq-1.6.tar.gz" "5de8c8e29aaa3fb9cc6b47bb27299f271354ebb72514e3accadc7d38b5bbaa72" "" "https://github.com/stedolan/jq/releases/download/jq-1.6/"    '
Hash of the local file jq-1.6.tar.gz does not match (file: 9625784cf2e4fd9842f1d407681ce4878b5b0dcddbcd31c6135114a30c71e6a8, requested: 5de8c8e29aaa3fb9cc6b47bb27299f271354ebb72514e3accadc7d38b5bbaa72) - deleting download.
make[3]: *** [Makefile:52: /home/builder/lede_x86/dl/jq-1.6.tar.gz] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/utils/jq'
time: package/feeds/packages/jq/compile#0.16#0.08#0.20
    ERROR: package/feeds/packages/jq failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/jq/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~$
builder@Build-Server:~$

zhangguanzhang, 2022-09-29:
```
    sed -ri 's#9625784cf2e4fd9842f1d407681ce4878b5b0dcddbcd31c6135114a30c71e6a8#5de8c8e29aaa3fb9cc6b47bb27299f271354ebb72514e3accadc7d38b5bbaa72#' ./feeds/packages/utils/jq/Makefile
    
```

1054009064, 2022-09-29:
## 借贴反馈个问题：

近期合并了使用 aria2 进行 `make download` 的代码，部分用户安装了 aria2 进行 `make download`，
就会返回 openwrt cdn 这类比较快的下载，导致 hash 校验不正确 (如 #10185)
而 openwrt cdn 的 `jq-1.6.tar.gz` 和 `https://github.com/stedolan/jq` 的大小就存在差异（如图二）

@smallprogram 能否找到一个更完美的方法呢？

jq-1.6.tar(1).gz 来自 `https://github.com/stedolan/jq/releases/download/jq-1.6/jq-1.6.tar.gz`
jq-1.6.tar(2).gz 来自 `https://sources.cdn.openwrt.org/jq-1.6.tar.gz`
jq-1.6.tar(3).gz 来自` https://sources.openwrt.org/jq-1.6.tar.gz`

### 图一：
[image]
### 图二：
[image]

smallprogram, 2022-09-29:
> ## 借贴反馈个问题：
> 近期合并了使用 aria2 进行 `make download` 的代码，部分用户安装了 aria2 进行 `make download`， 就会返回 openwrt cdn 这类比较快的下载，导致 hash 校验不正确 (如 #10185) 而 openwrt cdn 的 `jq-1.6.tar.gz` 和 `https://github.com/stedolan/jq` 的大小就存在差异（如图二）
> 
> @smallprogram 能否找到一个更完美的方法呢？ 另外猜测 #10193 也可能遇到了这个问题
> 
> jq-1.6.tar(1).gz 来自 `https://github.com/stedolan/jq/releases/download/jq-1.6/jq-1.6.tar.gz` jq-1.6.tar(2).gz 来自 `https://sources.cdn.openwrt.org/jq-1.6.tar.gz` jq-1.6.tar(3).gz 来自 `https://sources.openwrt.org/jq-1.6.tar.gz`
> 
> ### 图一：
> [image]
> 
> ### 图二：
> [image]

目前我只知道有这么几种方案
1. 简单粗暴，取消hash校验，`PKG_HASH:=skip`，但是可能会出现安全性问题
2. 干掉非op cdn的源，统一用hash一致的源，工作量很大
3. revert aria2脚本

1054009064, 2022-09-29:
> 
> 目前我只知道有这么几种方案
> 
> 1. 简单粗暴，取消hash校验，`PKG_HASH:=skip`，但是可能会出现安全性问题
> 2. 干掉非op cdn的源，统一用hash一致的源，工作量很大
> 3. revert aria2脚本

哈哈，实在不清楚他们为什么源码用 `stedolan/jq` 的包，但 cdn 又魔改了另外一个包...

1582130940, 2022-09-30:
个人建议revert掉以下四条（按顺序revert）
https://github.com/coolsnowwolf/lede/commit/03a0395eb25bdb0c9f45ebbf981cfc90a4ae794d
https://github.com/coolsnowwolf/lede/commit/5ae6c9be691778a7e4ccc6f429b92a98ea0e79a4
https://github.com/coolsnowwolf/lede/commit/b0ea2f3b0ad059e62002ca4b8d6e2cffd566741e
https://github.com/coolsnowwolf/lede/commit/320adae6f63a7fdec770440e47f33821cf34128a

Cathgao, 2022-09-30:
问题就出在魔改上面，后备源不是应该为同一份文件的异地备份么，文件被改了这本身就是不合理的事情吧？

bradfordzhang, 2022-09-30:
> 
> 
> > ## 借贴反馈个问题：
> > 近期合并了使用 aria2 进行 `make download` 的代码，部分用户安装了 aria2 进行 `make download`， 就会返回 openwrt cdn 这类比较快的下载，导致 hash 校验不正确 (如 #10185) 而 openwrt cdn 的 `jq-1.6.tar.gz` 和 `https://github.com/stedolan/jq` 的大小就存在差异（如图二）
> > 
> > @smallprogram 能否找到一个更完美的方法呢？ 另外猜测 #10193 也可能遇到了这个问题
> > 
> > jq-1.6.tar(1).gz 来自 `https://github.com/stedolan/jq/releases/download/jq-1.6/jq-1.6.tar.gz` jq-1.6.tar(2).gz 来自 `https://sources.cdn.openwrt.org/jq-1.6.tar.gz` jq-1.6.tar(3).gz 来自 `https://sources.openwrt.org/jq-1.6.tar.gz`
> > 
> > ### 图一：
> > [image]
> > 
> > ### 图二：
> > [image]
> 
> 目前我只知道有这么几种方案
> 1. 简单粗暴，取消hash校验，`PKG_HASH:=skip`，但是可能会出现安全性问题
> 2. 干掉非op cdn的源，统一用hash一致的源，工作量很大
> 3. revert aria2脚本
> 
> @aiamadeus @Ansuel @bradfordzhang 各位大佬有啥好方案吗？

在op上游交了[pr](https://github.com/openwrt/packages/pull/19496)

1054009064, 2022-09-30:
> 
> 在op上游交了[pr](https://github.com/openwrt/packages/pull/19496)

@bradfordzhang 感谢您，这个 jq 只是个例，应用了 aria2 后，日后可能会有越来越多的 openwrt cdn 和源码包 hash 不一致。但 aria2 确实提高了下载效率，糟糕的 cdn 缓存让这看起来变得糟糕了。😂

bradfordzhang, 2022-09-30:
> > 
> > 在op上游交了[pr](https://github.com/openwrt/packages/pull/19496)
> 
> @bradfordzhang 感谢您，这个 jq 只是个例，应用了 aria2 后，日后可能会有越来越多的 openwrt cdn 和源码包 hash 不一致。但 aria2 确实提高了下载效率，糟糕的 cdn 缓存让这看起来变得糟糕了。😂
> 
> 

举手之劳，我在想能不能采用 多个hash满足任意一个即代表校验通过 这种方式进行校验

bradfordzhang, 2022-09-30:
> > 
> > 在op上游交了[pr](https://github.com/openwrt/packages/pull/19496)
> 
> @bradfordzhang 感谢您。这个 jq 只是个例，应用了 aria2 后，日后可能会有越来越多的 openwrt cdn 和源码包 hash 不一致。但 aria2 确实提高了下载效率，糟糕的 cdn 缓存让这看起来变得糟糕了。😂
> 
> 

其实是jq force push tag

1054009064, 2022-10-01:
@bradfordzhang 目前 aria2 的下载只会在判断到最快的链接下载，而一旦 hash 不匹配，就直接判断为失败，而不是去选下一个网址继续进行尝试，这是否是个 bug 呢？

bradfordzhang, 2022-10-01:
> @bradfordzhang 目前 aria2 的下载只会在判断到最快的链接下载，而一旦 hash 不匹配，就直接判断为失败，而不是去选下一个网址继续进行尝试，这是否是个 bug 呢？

是的 我正准备解决这个问题

bradfordzhang, 2022-10-04:
暂时可以通过
```
wget https://github.com/openwrt/packages/pull/19496.patch
git apply 19496.patch
```
解决这个问题
上游也不清楚发生了什么……😓

hyyz17200, 2022-10-28:
我今天编译还是遇到这个问题。
能不能手工自己下载JQ放进目录去临时解决呢？

zhangguanzhang, 2022-10-28:
> 我今天编译还是遇到这个问题。 能不能手工自己下载JQ放进目录去临时解决呢？

看二楼

hyyz17200, 2022-10-28:
> > 我今天编译还是遇到这个问题。 能不能手工自己下载JQ放进目录去临时解决呢？
> 
> 看二楼

谢谢您。 
我看您二楼代码是跳过hash校验吧。 我想既然是cdn的问题，直接下github的覆盖是不是更好？hash应该是正确的。
刚试了一下，删除了lede/dl里的jq文件，wget进去再编译就通过了。临时解决。

zhangguanzhang, 2022-10-28:
> 

只有这个release版本被re-push了，我选择这个版本跳过

closed 2022-10-30 by aiamadeus