coolsnowwolf/lede #10199: Unable to compile uboot-rockchip

celgus, 2022-09-29:
## 1. About the question you want to submit

Q: Is the issue searched (use "x" to select)
* [ ] No similar issue
* [x] #9757

## 2. Detailed description

### (1) Specific questions

Unable to compile uboot-rockchip for Rockchip RK3328 (OrangePi R1 Plus LTS)

### (2) Router model and firmware version

OrangePi R1 Plus LTS (latest version, tried to compile 29/09/2022 after commit 2b67b80)

### (3) Detailed log

>   CC      tpl/drivers/ram/rockchip/sdram_rk3328.o
drivers/ram/rockchip/sdram_rk3328.c: In function 'rk3328_dmc_init':
drivers/ram/rockchip/sdram_rk3328.c:525:55: error: 'struct dtd_rockchip_rk3328_dmc' has no member named 'rockchip_sdram_params'
  525 |                                         (void *)dtplat->rockchip_sdram_params;
      |                                                       ^~
make[8]: *** [scripts/Makefile.build:258: tpl/drivers/ram/rockchip/sdram_rk3328.o] Error 1
make[7]: *** [scripts/Makefile.build:398: tpl/drivers/ram/rockchip] Error 2
make[6]: *** [scripts/Makefile.build:398: tpl/drivers/ram] Error 2
make[5]: *** [scripts/Makefile.spl:531: tpl/drivers] Error 2
make[4]: *** [Makefile:2109: tpl/u-boot-tpl] Error 2
make[4]: Leaving directory '/home/ilya/lede/build_dir/target-aarch64_generic_musl/u-boot-orangepi-r1-plus-lts-rk3328/u-boot-2022.07'
make[3]: *** [Makefile:247: /home/ilya/lede/build_dir/target-aarch64_generic_musl/u-boot-orangepi-r1-plus-lts-rk3328/u-boot-2022.07/.built] Error 2
make[3]: Leaving directory '/home/ilya/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/orangepi-r1-plus-lts-rk3328/compile#65.03#5.97#68.44
    ERROR: package/boot/uboot-rockchip failed to build (build variant: orangepi-r1-plus-lts-rk3328).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/ilya/lede'
make[1]: *** [package/Makefile:110: /home/ilya/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ilya/lede'
make: *** [/home/ilya/lede/include/toplevel.mk:230: world] Error 2

QYKing233, 2022-10-11:
我今天编译r1 plus lts也是这个问题

QYKing233, 2022-10-13:
可以用20220505的稳定版本先编译用用

closed 2022-12-13 by aiamadeus