coolsnowwolf/lede #10200: qtbase 每次下载的 hash 值都不一样

haiibo, 2022-09-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
2022-09-29T22:01:39.6522587Z echo "iputils-arping" >> /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/iputils.default.install
2022-09-29T22:01:39.6540052Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/packages/net/iputils'
2022-09-29T22:01:39.6545983Z time: package/feeds/packages/iputils/compile#0.16#0.05#0.20
2022-09-29T22:01:39.6666602Z make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/packages/utils/jq'
2022-09-29T22:01:39.7935420Z mkdir -p /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/dl
2022-09-29T22:01:39.7965233Z SHELL= flock /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/.jq-1.6.tar.gz.flock -c '  	/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/scripts/download.pl "/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/dl" "jq-1.6.tar.gz" "5de8c8e29aaa3fb9cc6b47bb27299f271354ebb72514e3accadc7d38b5bbaa72" "" "https://github.com/stedolan/jq/releases/download/jq-1.6/"    '
2022-09-29T22:01:39.8810310Z echo "jq" >> /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/jq.default.install
2022-09-29T22:01:39.8830095Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/packages/utils/jq'
2022-09-29T22:01:39.8836014Z time: package/feeds/packages/jq/compile#0.20#0.02#0.22
2022-09-29T22:01:39.8955953Z make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/luci/applications/luci-app-pushbot'
2022-09-29T22:01:40.0471198Z echo "luci-app-pushbot" >> /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/luci-app-pushbot.default.install
2022-09-29T22:01:40.0492500Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/luci/applications/luci-app-pushbot'
2022-09-29T22:01:40.0498270Z time: package/feeds/luci/luci-app-pushbot/compile#0.13#0.03#0.16
2022-09-29T22:01:40.0623859Z make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/packages/libs/qtbase'
2022-09-29T22:01:40.2246642Z mkdir -p /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/dl
2022-09-29T22:01:40.2281423Z SHELL= flock /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/.qtbase-everywhere-opensource-src-5.15.4.tar.xz.flock -c '  	/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/scripts/download.pl "/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/dl" "qtbase-everywhere-opensource-src-5.15.4.tar.xz" "f93207bbf86663bd7abd16fac91deb4c753c1c133d3ac768fe61464b17ae8a18" "" "http://download.qt.io/official_releases/qt/5.15/5.15.4/submodules" "http://master.qt.io/archive/qt/5.15/5.15.4/submodules" "http://mirrors.tuna.tsinghua.edu.cn/qt/archive/qt/5.15/5.15.4/submodules" "http://qt.mirror.constant.com/archive/qt/5.15/5.15.4/submodules"    '
2022-09-29T22:01:40.2669447Z + [ -d /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c ] || mkdir /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c; touch /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz_spp; aria2c --stderr -c -x2 -s10 -j10 -k1M http://download.qt.io/official_releases/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz http://master.qt.io/archive/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz http://mirrors.tuna.tsinghua.edu.cn/qt/archive/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz http://qt.mirror.constant.com/archive/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz https://sources.cdn.openwrt.org/qtbase-everywhere-opensource-src-5.15.4.tar.xz https://sources.openwrt.org/qtbase-everywhere-opensource-src-5.15.4.tar.xz https://mirror2.openwrt.org/sources/qtbase-everywhere-opensource-src-5.15.4.tar.xz --server-stat-of=/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz_spp --server-stat-if=/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz_spp -d /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c -o qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz; cat /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz; rm /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz_spp
2022-09-29T22:01:40.2734451Z 
2022-09-29T22:01:40.2735272Z 09/29 22:01:40 [[1;32mNOTICE[0m] Downloading 1 item(s)
2022-09-29T22:01:40.2814852Z 
2022-09-29T22:01:40.2815637Z 09/29 22:01:40 [[1;32mNOTICE[0m] ServerStat file /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz_spp loaded successfully.
2022-09-29T22:01:40.7671083Z 
2022-09-29T22:01:40.7673768Z 09/29 22:01:40 [[1;32mNOTICE[0m] CUID#7 - Redirecting to https://download.qt.io/official_releases/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz
2022-09-29T22:01:41.3394463Z [#10f956 7.8KiB/0B CN:1 DL:69KiB]
2022-09-29T22:01:41.3419935Z 
2022-09-29T22:01:41.3420831Z 09/29 22:01:41 [[1;32mNOTICE[0m] Download complete: /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz
2022-09-29T22:01:41.3425897Z 
2022-09-29T22:01:41.3426486Z 09/29 22:01:41 [[1;32mNOTICE[0m] GID#26facc221d6b9a9e - Download has already completed: /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz
2022-09-29T22:01:41.3427401Z 
2022-09-29T22:01:41.3427907Z 09/29 22:01:41 [[1;32mNOTICE[0m] Download complete: /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz
2022-09-29T22:01:41.3443073Z 
2022-09-29T22:01:41.3443685Z 09/29 22:01:41 [[1;32mNOTICE[0m] ServerStat file /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz_spp saved successfully.
2022-09-29T22:01:41.3444184Z 
2022-09-29T22:01:41.3444353Z Download Results:
2022-09-29T22:01:41.3444578Z gid   |stat|avg speed  |path/URI
2022-09-29T22:01:41.3444909Z ======+====+===========+=======================================================
2022-09-29T22:01:41.3445520Z 10f956|OK  |    14KiB/s|/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_MMbi3qXMNz
2022-09-29T22:01:41.3447277Z 26facc|OK  |        n/a|/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz
2022-09-29T22:01:41.3447559Z 
2022-09-29T22:01:41.3447807Z Status Legend:
2022-09-29T22:01:41.3448112Z (OK):download completed.
2022-09-29T22:01:41.3499284Z Hash of the downloaded file does not match (file: 608f64b6b24a426ed73973d3237c0e5afad98182676bdd26aed3735eb346352d, requested: f93207bbf86663bd7abd16fac91deb4c753c1c133d3ac768fe61464b17ae8a18) - deleting download.
2022-09-29T22:01:41.3499980Z No more mirrors to try - giving up.
2022-09-29T22:01:41.3511393Z make[2]: *** [Makefile:201: /home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/dl/qtbase-everywhere-opensource-src-5.15.4.tar.xz] Error 2
2022-09-29T22:01:41.3511995Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/feeds/packages/libs/qtbase'
2022-09-29T22:01:41.3512503Z time: package/feeds/packages/qtbase/compile#0.19#0.05#1.29
2022-09-29T22:01:41.3518723Z     ERROR: package/feeds/packages/qtbase failed to build.
2022-09-29T22:01:41.3527266Z make[1]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt'
2022-09-29T22:01:41.3531112Z make[1]: *** [package/Makefile:116: package/feeds/packages/qtbase/compile] Error 1
2022-09-29T22:01:41.3535290Z make: *** [/home/runner/work/OpenWrt/OpenWrt/workspace/openwrt/include/toplevel.mk:230: package/compile] Error 2
2022-09-29T22:01:41.3601839Z ##[error]Process completed with exit code 2.

haiibo, 2022-09-30:
目前我关闭了 qtbase 的 hash 验证，不知道能不能编译成功

aoeII, 2022-09-30:
我这里是rrdtool hash错，估计源有点问题。

```
2022-09-29T16:56:24.1506168Z mkdir -p /workdir/openwrt/dl
2022-09-29T16:56:24.1536795Z SHELL= flock /workdir/openwrt/tmp/.rrdtool-1.0.50.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "rrdtool-1.0.50.tar.gz" "42aa7c213dedbd95d33ca84d92f4187880f7e96062c6a3fb05bfb16f77ba2a91" "" "http://oss.oetiker.ch/rrdtool/pub/rrdtool-1.0.x" "http://www.fastmirrors.org/rrdtool/rrdtool-1.0.x" "ftp://ftp.sunet.se/pub/network/monitoring/rrdtool/rrdtool-1.0.x"    '
2022-09-29T16:56:24.2011950Z + [ -d /workdir/openwrt/tmp/aria2c ] || mkdir /workdir/openwrt/tmp/aria2c; touch /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU_spp; aria2c --stderr -c -x2 -s10 -j10 -k1M http://oss.oetiker.ch/rrdtool/pub/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz http://www.fastmirrors.org/rrdtool/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz ftp://ftp.sunet.se/pub/network/monitoring/rrdtool/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz https://sources.cdn.openwrt.org/rrdtool-1.0.50.tar.gz https://sources.openwrt.org/rrdtool-1.0.50.tar.gz https://mirror2.openwrt.org/sources/rrdtool-1.0.50.tar.gz --server-stat-of=/workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU_spp --server-stat-if=/workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU_spp -d /workdir/openwrt/tmp/aria2c -o rrdtool-1.0.50.tar.gz_cHMChp5tEU; cat /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU; rm /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU_spp
2022-09-29T16:56:24.2081799Z 
2022-09-29T16:56:24.2082514Z 09/29 16:56:24 [[1;32mNOTICE[0m] Downloading 1 item(s)
2022-09-29T16:56:24.2179300Z 
2022-09-29T16:56:24.2180274Z 09/29 16:56:24 [[1;32mNOTICE[0m] ServerStat file /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU_spp loaded successfully.
2022-09-29T16:56:24.4085328Z 
2022-09-29T16:56:24.4086302Z 09/29 16:56:24 [[1;32mNOTICE[0m] CUID#7 - Redirecting to https://oss.oetiker.ch/rrdtool/pub/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz
2022-09-29T16:56:24.7895771Z 
2022-09-29T16:56:24.7897093Z 09/29 16:56:24 [[1;32mNOTICE[0m] CUID#7 - Redirecting to http://oss.oetiker.ch/rrdtool/404.var
2022-09-29T16:56:24.8833874Z 
2022-09-29T16:56:24.8834703Z 09/29 16:56:24 [[1;32mNOTICE[0m] CUID#7 - Redirecting to https://oss.oetiker.ch/rrdtool/404.var
2022-09-29T16:56:24.9804123Z 
2022-09-29T16:56:24.9805191Z 09/29 16:56:24 [[1;32mNOTICE[0m] Download complete: /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU
2022-09-29T16:56:24.9813120Z 
2022-09-29T16:56:24.9813756Z 09/29 16:56:24 [[1;32mNOTICE[0m] ServerStat file /workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU_spp saved successfully.
2022-09-29T16:56:24.9814087Z 
2022-09-29T16:56:24.9814602Z Download Results:
2022-09-29T16:56:24.9814876Z gid   |stat|avg speed  |path/URI
2022-09-29T16:56:24.9815180Z ======+====+===========+=======================================================
2022-09-29T16:56:24.9817129Z c5a2be|OK  |   5.3MiB/s|/workdir/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_cHMChp5tEU
2022-09-29T16:56:24.9817391Z 
2022-09-29T16:56:24.9817485Z Status Legend:
2022-09-29T16:56:24.9817733Z (OK):download completed.
2022-09-29T16:56:24.9887766Z Hash of the downloaded file does not match (file: 88dcddf2a92b9dc9243e929340fdc6f3431b779e22438093d7a31a400f1c3909, requested: 42aa7c213dedbd95d33ca84d92f4187880f7e96062c6a3fb05bfb16f77ba2a91) - deleting download.
2022-09-29T16:56:24.9888393Z No more mirrors to try - giving up.
2022-09-29T16:56:24.9891101Z make[3]: *** [Makefile:134: /workdir/openwrt/dl/rrdtool-1.0.50.tar.gz] Error 2
2022-09-29T16:56:24.9891623Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/rrdtool1'
2022-09-29T16:56:24.9898761Z time: package/feeds/packages/rrdtool1/compile#0.19#0.02#0.97
2022-09-29T16:56:24.9902288Z     ERROR: package/feeds/packages/rrdtool1 failed to build.
2022-09-29T16:56:24.9918104Z make[2]: *** [package/Makefile:116: package/feeds/packages/rrdtool1/compile] Error 1
2022-09-29T16:56:24.9918662Z make[2]: Leaving directory '/workdir/openwrt'
2022-09-29T16:56:24.9929695Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2022-09-29T16:56:24.9930155Z make[1]: Leaving directory '/workdir/openwrt'
2022-09-29T16:56:24.9932898Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2022-09-29T16:56:25.0152077Z ##[error]Process completed with exit code 2.

```

haiibo, 2022-09-30:
> 我这里是rrdtool hash错，估计源有点问题。
>
> ```
> 2022-09-29T16:56:24.1506168Z mkdir -p /workdir/openwrt/dl
> 2022-09-29T16:56:24.1536795Z SHELL= flock /workdir/openwrt/tmp/.rrdtool-1.0.50.tar.gz.flock -c ' [...]

你这个应该是这条https://github.com/coolsnowwolf/packages/pull/544/commits/9e084e7ed41551f0de67a4fb2072efdd2b2f54ca改掉了导致的，你下载的hash值和修改之前是一样的