coolsnowwolf/lede #10201: 云编译出rrdtool1错误，到底怎么才能不出错啊

qqsir-dev, 2022-09-30:
反馈bug/问题模板，提建议请删除
前面也有人提，但是怎么才能顺利编译？有点晕
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
 make[2] -C customfeeds/packages/utils/rrdtool1 compile
    ERROR: package/feeds/packages/rrdtool1 failed to build.
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:230: package/compile] Error 1
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

aoeII, 2022-09-30:
https://github.com/coolsnowwolf/packages/commit/9e084e7ed41551f0de67a4fb2072efdd2b2f54ca
应该是这个commit还没有合并到main branch里，下载的文件已经是新的了，但Makefile里的代码还没更新
本地编译的话可以手动替换文件，云编译的话得写脚本替换

1054009064, 2022-10-01:
```
2022-10-01T00:09:58.4809042Z make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/customfeeds/packages/utils/rrdtool1'
2022-10-01T00:09:58.6368787Z mkdir -p /home/runner/work/AutoLede/AutoLede/openwrt/dl
2022-10-01T00:09:58.6445307Z SHELL= flock /home/runner/work/AutoLede/AutoLede/openwrt/tmp/.rrdtool-1.0.50.tar.gz.flock -c '  	/home/runner/work/AutoLede/AutoLede/openwrt/scripts/download.pl "/home/runner/work/AutoLede/AutoLede/openwrt/dl" "rrdtool-1.0.50.tar.gz" "42aa7c213dedbd95d33ca84d92f4187880f7e96062c6a3fb05bfb16f77ba2a91" "" "http://oss.oetiker.ch/rrdtool/pub/rrdtool-1.0.x" "http://www.fastmirrors.org/rrdtool/rrdtool-1.0.x" "ftp://ftp.sunet.se/pub/network/monitoring/rrdtool/rrdtool-1.0.x"    '
2022-10-01T00:09:58.6901778Z + [ -d /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c ] || mkdir /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c; touch /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT_spp; aria2c --stderr -c -x2 -s10 -j10 -k1M http://oss.oetiker.ch/rrdtool/pub/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz http://www.fastmirrors.org/rrdtool/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz ftp://ftp.sunet.se/pub/network/monitoring/rrdtool/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz https://sources.cdn.openwrt.org/rrdtool-1.0.50.tar.gz https://sources.openwrt.org/rrdtool-1.0.50.tar.gz https://mirror2.openwrt.org/sources/rrdtool-1.0.50.tar.gz --server-stat-of=/home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT_spp --server-stat-if=/home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT_spp -d /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c -o rrdtool-1.0.50.tar.gz_p3qVayOIvT; cat /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT; rm /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT_spp
2022-10-01T00:09:58.6986760Z 
2022-10-01T00:09:58.6987555Z 10/01 00:09:58 [[1;32mNOTICE[0m] Downloading 1 item(s)
2022-10-01T00:09:58.7093379Z 
2022-10-01T00:09:58.7094092Z 10/01 00:09:58 [[1;32mNOTICE[0m] ServerStat file /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT_spp loaded successfully.
2022-10-01T00:09:59.2320943Z 
2022-10-01T00:09:59.2322039Z 10/01 00:09:59 [[1;32mNOTICE[0m] CUID#7 - Redirecting to https://oss.oetiker.ch/rrdtool/pub/rrdtool-1.0.x/rrdtool-1.0.50.tar.gz
2022-10-01T00:09:59.7181321Z [#044a29 0B/0B CN:1 DL:0B]
2022-10-01T00:09:59.8864110Z 
2022-10-01T00:09:59.8865290Z 10/01 00:09:59 [[1;32mNOTICE[0m] CUID#7 - Redirecting to http://oss.oetiker.ch/rrdtool/404.var
2022-10-01T00:10:00.0485410Z 
2022-10-01T00:10:00.0487019Z 10/01 00:10:00 [[1;32mNOTICE[0m] CUID#7 - Redirecting to https://oss.oetiker.ch/rrdtool/404.var
2022-10-01T00:10:00.2150098Z 
2022-10-01T00:10:00.2151972Z 10/01 00:10:00 [[1;32mNOTICE[0m] Download complete: /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT
2022-10-01T00:10:00.2155845Z 
2022-10-01T00:10:00.2156575Z 10/01 00:10:00 [[1;32mNOTICE[0m] ServerStat file /home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT_spp saved successfully.
2022-10-01T00:10:00.2157310Z 
2022-10-01T00:10:00.2157569Z Download Results:
2022-10-01T00:10:00.2158165Z gid   |stat|avg speed  |path/URI
2022-10-01T00:10:00.2158586Z ======+====+===========+=======================================================
2022-10-01T00:10:00.2159442Z 044a29|OK  |   2.6MiB/s|/home/runner/work/AutoLede/AutoLede/openwrt/tmp/aria2c/rrdtool-1.0.50.tar.gz_p3qVayOIvT
2022-10-01T00:10:00.2159856Z 
2022-10-01T00:10:00.2160568Z Status Legend:
2022-10-01T00:10:00.2160959Z (OK):download completed.
2022-10-01T00:10:00.2234232Z Hash of the downloaded file does not match (file: 88dcddf2a92b9dc9243e929340fdc6f3431b779e22438093d7a31a400f1c3909, requested: 42aa7c213dedbd95d33ca84d92f4187880f7e96062c6a3fb05bfb16f77ba2a91) - deleting download.
2022-10-01T00:10:00.2235279Z No more mirrors to try - giving up.
2022-10-01T00:10:00.2249674Z make[2]: *** [Makefile:134: /home/runner/work/AutoLede/AutoLede/openwrt/dl/rrdtool-1.0.50.tar.gz] Error 2
2022-10-01T00:10:00.2250879Z make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/customfeeds/packages/utils/rrdtool1'
2022-10-01T00:10:00.2252049Z time: package/feeds/packages/rrdtool1/compile#0.19#0.08#1.75
2022-10-01T00:10:00.2258248Z     ERROR: package/feeds/packages/rrdtool1 failed to build.
2022-10-01T00:10:00.2272202Z make[1]: *** [package/Makefile:116: package/feeds/packages/rrdtool1/compile] Error 1
2022-10-01T00:10:00.2278121Z make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
2022-10-01T00:10:00.2285054Z make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:230: package/compile] Error 2
2022-10-01T00:10:00.2552368Z ##[error]Process completed with exit code 2.
```
[image]

closed 2022-10-01 by coolsnowwolf