coolsnowwolf/lede #10208: qtbase compile Error

xiangfeidexiaohuo, 2022-09-30:
2022-09-30T21:08:15.1333014Z make[3]: Entering directory '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/feeds/packages/libs/qtbase'
2022-09-30T21:08:15.2912512Z depbase=`echo r/hfs/journal.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
2022-09-30T21:08:15.2914656Z /bin/bash ../../libtool  --tag=CC   --mode=compile x86_64-openwrt-linux-musl-gcc  -I. -I../../lib  -I../../libparted/labels -I../../include -I../../include -I../../lib  -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/parted-3.4=parted-3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT r/hfs/journal.lo -MD -MP -MF $depbase.Tpo -c -o r/hfs/journal.lo r/hfs/journal.c &&\
2022-09-30T21:08:15.2915946Z mv -f $depbase.Tpo $depbase.Plo
2022-09-30T21:08:15.3104611Z mkdir -p /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/dl
2022-09-30T21:08:15.3112641Z SHELL= flock /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/.qtbase-everywhere-opensource-src-5.15.4.tar.xz.flock -c '  	/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/scripts/download.pl "/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/dl" "qtbase-everywhere-opensource-src-5.15.4.tar.xz" "f93207bbf86663bd7abd16fac91deb4c753c1c133d3ac768fe61464b17ae8a18" "" "http://download.qt.io/official_releases/qt/5.15/5.15.4/submodules" "http://master.qt.io/archive/qt/5.15/5.15.4/submodules" "http://mirrors.tuna.tsinghua.edu.cn/qt/archive/qt/5.15/5.15.4/submodules" "http://qt.mirror.constant.com/archive/qt/5.15/5.15.4/submodules"    '
2022-09-30T21:08:15.3491438Z + [ -d /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c ] || mkdir /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c; touch /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi_spp; aria2c --stderr -c -x2 -s10 -j10 -k1M http://download.qt.io/official_releases/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz http://master.qt.io/archive/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz http://mirrors.tuna.tsinghua.edu.cn/qt/archive/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz http://qt.mirror.constant.com/archive/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz https://sources.cdn.openwrt.org/qtbase-everywhere-opensource-src-5.15.4.tar.xz https://sources.openwrt.org/qtbase-everywhere-opensource-src-5.15.4.tar.xz https://mirror2.openwrt.org/sources/qtbase-everywhere-opensource-src-5.15.4.tar.xz --server-stat-of=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi_spp --server-stat-if=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi_spp -d /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c -o qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi; cat /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi; rm /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi_spp
2022-09-30T21:08:15.3495722Z libtool: compile:  x86_64-openwrt-linux-musl-gcc -I. -I../../lib -I../../libparted/labels -I../../include -I../../include -I../../lib -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/parted-3.4=parted-3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT r/hfs/journal.lo -MD -MP -MF r/hfs/.deps/journal.Tpo -c r/hfs/journal.c  -fPIC -DPIC -o r/hfs/.libs/journal.o
2022-09-30T21:08:15.3565992Z 
2022-09-30T21:08:15.3569056Z 09/30 21:08:15 [[1;32mNOTICE[0m] Downloading 1 item(s)
2022-09-30T21:08:15.3637955Z 
2022-09-30T21:08:15.3640757Z 09/30 21:08:15 [[1;32mNOTICE[0m] ServerStat file /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/tmp/aria2c/qtbase-everywhere-opensource-src-5.15.4.tar.xz_ZITmPnuwEi_spp loaded successfully.
2022-09-30T21:08:15.5088336Z libtool: compile:  x86_64-openwrt-linux-musl-gcc -I. -I../../lib -I../../libparted/labels -I../../include -I../../include -I../../lib -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/parted-3.4=parted-3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT r/hfs/journal.lo -MD -MP -MF r/hfs/.deps/journal.Tpo -c r/hfs/journal.c -o r/hfs/journal.o >/dev/null 2>&1
2022-09-30T21:08:15.6381706Z 
2022-09-30T21:08:15.6384383Z 09/30 21:08:15 [[1;32mNOTICE[0m] CUID#7 - Redirecting to https://download.qt.io/official_releases/qt/5.15/5.15.4/submodules/qtbase-everywhere-opensource-src-5.15.4.tar.xz
2022-09-30T21:08:15.6774686Z depbase=`echo r/hfs/probe.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
2022-09-30T21:08:15.6777296Z /bin/bash ../../libtool  --tag=CC   --mode=compile x86_64-openwrt-linux-musl-gcc  -I. -I../../lib  -I../../libparted/labels -I../../include -I../../include -I../../lib  -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/parted-3.4=parted-3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT r/hfs/probe.lo -MD -MP -MF $depbase.Tpo -c -o r/hfs/probe.lo r/hfs/probe.c &&\
2022-09-30T21:08:15.6779432Z mv -f $depbase.Tpo $depbase.Plo
2022-09-30T21:08:15.7320589Z libtool: compile:  x86_64-openwrt-linux-musl-gcc -I. -I../../lib -I../../libparted/labels -I../../include -I../../include -I../../lib -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/parted-3.4=parted-3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT r/hfs/probe.lo -MD -MP -MF r/hfs/.deps/probe.Tpo -c r/hfs/probe.c  -fPIC -DPIC -o r/hfs/.libs/probe.o
[... cut ...]
2022-09-30T21:08:16.3637545Z make[3]: *** [Makefile:201: /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/dl/qtbase-everywhere-opensource-src-5.15.4.tar.xz] Error 2
2022-09-30T21:08:16.3638261Z make[3]: Leaving directory '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/feeds/packages/libs/qtbase'
2022-09-30T21:08:16.3641531Z time: package/feeds/packages/qtbase/compile#0.16#0.07#1.24
2022-09-30T21:08:16.3650368Z     ERROR: package/feeds/packages/qtbase failed to build.
2022-09-30T21:08:16.3657546Z make[2]: *** [package/Makefile:116: package/feeds/packages/qtbase/compile] Error 1

X86平台，突然这qtbase就编译出错

ecrasy, 2022-10-01:
make download 下载的包和makefile的hash值不一致。
等修复吧，估计是升级了下载路径，没更新hash

ecrasy, 2022-10-01:
如果等不及的话
可以用命令强行修改hash，以继续编译
sed -i 's/f93207bbf86663bd7abd16fac91deb4c753c1c133d3ac768fe61464b17ae8a18/dcddcd8614546c56d62ec447e1d01bc71840399167d9e0bacd1f268f501e876a/g' package/feeds/packages/qtbase/Makefile

1054009064, 2022-10-01:
别改 hash，也不要跳过校验 hash，有一个下载链接有问题，下载的包不完整。

1054009064, 2022-10-01:
可以参考这个，通过删除这个链接临时解决下载问题https://github.com/1054009064/packages/commit/44a85da1e9bab2b94b9007a63bddf8b6562ccfa9

1054009064, 2022-10-01:
待合并后帮忙测试能否下载成功，我测试了是可以的 https://github.com/coolsnowwolf/packages/pull/548

closed 2022-10-01 by coolsnowwolf