coolsnowwolf/lede #10209: ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle'

geniusyjiang, 2022-10-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：昨天编译的固件版本跟之前选择的包一样，能编译成功，但是启动后ip6tables运行异常，遇到ip6tables -t mangle命令就报“ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.”错误。上次编译是两个月前，以及之前编译的都没问题。这次编译选择的firewall和netfilter相关包并不比以前少。

### (2) 路由器型号和固件版本

A：ACRH17，20220930自己编译

### (3) 详细日志

A：编译的配置：
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-acrh17=y
CONFIG_TARGET_BOARD="ipq40xx"
CONFIG_BUILD_PATENTED=y
CONFIG_IPTABLES_NFTABLES=y
CONFIG_MINIUPNPD_IGDv2=y
CONFIG_PACKAGE_6in4=y
# CONFIG_PACKAGE_UnblockNeteaseMusic-Go is not set
# CONFIG_PACKAGE_adbyby is not set
CONFIG_PACKAGE_arptables=y
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct is not set
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-full-htt=y
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct is not set
CONFIG_PACKAGE_ath10k-firmware-qca9984-ct-full-htt=y
# CONFIG_PACKAGE_attr is not set
# CONFIG_PACKAGE_autosamba is not set
CONFIG_PACKAGE_avahi-autoipd=y
CONFIG_PACKAGE_avahi-daemon-service-http=y
CONFIG_PACKAGE_avahi-daemon-service-ssh=y
CONFIG_PACKAGE_avahi-utils=y
CONFIG_PACKAGE_bash=y
# CONFIG_PACKAGE_ddns-scripts is not set
# CONFIG_PACKAGE_ddns-scripts_aliyun is not set
# CONFIG_PACKAGE_ddns-scripts_dnspod is not set
CONFIG_PACKAGE_dnsmasq_full_auth=y
CONFIG_PACKAGE_dnsmasq_full_conntrack=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=y
# CONFIG_PACKAGE_dnsmasq_full_tftp is not set
# CONFIG_PACKAGE_etherwake is not set
# CONFIG_PACKAGE_ethtool is not set
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_ip6tables=y
CONFIG_PACKAGE_iptables-mod-conntrack-extra=y
CONFIG_PACKAGE_iptables-mod-filter=y
# CONFIG_PACKAGE_iptables-mod-ipsec is not set
CONFIG_PACKAGE_iptables-mod-nat-extra=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_irqbalance=y
# CONFIG_PACKAGE_iw is not set
CONFIG_PACKAGE_iw-full=y
CONFIG_PACKAGE_kmod-arptables=y
# CONFIG_PACKAGE_kmod-crypto-cbc is not set
# CONFIG_PACKAGE_kmod-crypto-deflate is not set
# CONFIG_PACKAGE_kmod-crypto-des is not set
# CONFIG_PACKAGE_kmod-crypto-echainiv is not set
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ip6tables-extra=y
# CONFIG_PACKAGE_kmod-ipsec is not set
CONFIG_PACKAGE_kmod-ipt-conntrack-extra=y
CONFIG_PACKAGE_kmod-ipt-filter=y
# CONFIG_PACKAGE_kmod-ipt-ipsec is not set
CONFIG_PACKAGE_kmod-ipt-nat-extra=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-ipt-nathelper-rtsp=y
CONFIG_PACKAGE_kmod-ipt-raw6=y
CONFIG_PACKAGE_kmod-iptunnel=y
# CONFIG_PACKAGE_kmod-iptunnel6 is not set
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nf-log6=y
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-nf-reject6=y
CONFIG_PACKAGE_kmod-shortcut-fe=y
CONFIG_PACKAGE_kmod-shortcut-fe-cm=y
CONFIG_PACKAGE_kmod-sit=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_kmod-zram=y
# CONFIG_PACKAGE_libatomic is not set
# CONFIG_PACKAGE_libattr is not set
CONFIG_PACKAGE_libgdbm=y
# CONFIG_PACKAGE_libgnutls is not set
# CONFIG_PACKAGE_libminiupnpc is not set
# CONFIG_PACKAGE_libnatpmp is not set
CONFIG_PACKAGE_libnetfilter-conntrack=y
CONFIG_PACKAGE_libnfnetlink=y
CONFIG_PACKAGE_libnftnl=y
# CONFIG_PACKAGE_libpopt is not set
# CONFIG_PACKAGE_libstdcpp is not set
# CONFIG_PACKAGE_libtasn1 is not set
# CONFIG_PACKAGE_libtirpc is not set
# CONFIG_PACKAGE_liburing is not set
CONFIG_PACKAGE_lrzsz=y
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
# CONFIG_PACKAGE_luci-app-adbyby-plus is not set
# CONFIG_PACKAGE_luci-app-arpbind is not set
# CONFIG_PACKAGE_luci-app-autoreboot is not set
# CONFIG_PACKAGE_luci-app-ddns is not set
# CONFIG_PACKAGE_luci-app-filetransfer is not set
# CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
# CONFIG_PACKAGE_luci-app-nlbwmon is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui is not set
# CONFIG_PACKAGE_luci-app-samba4 is not set
# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_OFFLOADING is not set
# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_PDNSD is not set
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_SHORTCUT_FE_CM=y
# CONFIG_PACKAGE_luci-app-unblockmusic is not set
# CONFIG_PACKAGE_luci-app-unblockmusic_INCLUDE_UnblockNeteaseMusic_Go is not set
# CONFIG_PACKAGE_luci-app-vsftpd is not set
CONFIG_PACKAGE_luci-app-wireguard=y
# CONFIG_PACKAGE_luci-app-wol is not set
# CONFIG_PACKAGE_luci-app-zerotier is not set
CONFIG_PACKAGE_luci-i18n-wireguard-zh-cn=y
# CONFIG_PACKAGE_luci-lib-fs is not set
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-wireguard=y
# CONFIG_PACKAGE_nlbwmon is not set
CONFIG_PACKAGE_ntpdate=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0
# CONFIG_PACKAGE_openssl-util is not set
# CONFIG_PACKAGE_pdnsd-alt is not set
# CONFIG_PACKAGE_samba4-libs is not set
# CONFIG_PACKAGE_samba4-server is not set
# CONFIG_PACKAGE_strongswan is not set
# CONFIG_PACKAGE_vsftpd-alt is not set
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wol is not set
# CONFIG_PACKAGE_wsdd2 is not set
# CONFIG_PACKAGE_zerotier is not set
CONFIG_PACKAGE_zram-swap=y
CONFIG_PROCD_ZRAM_TMPFS=y
CONFIG_PACKAGE_kmod-ipt-offload=y
CONFIG_PACKAGE_kmod-nf-flow=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PCRE_JIT_ENABLED=y

coolsnowwolf, 2022-10-01:
已修复

geniusyjiang, 2022-10-02:
重新编译后确认已解决，谢谢

closed 2022-10-02 by geniusyjiang