coolsnowwolf/lede #10213: 拉取最新源码编译5.15和5.19内核x86固件报错

bi4tms, 2022-10-01:
拉取最新源码编译5.15和5.19内核x86固件报错，具体日志如下：
time: tools/ninja/compile#0.03#0.20#0.25
make[3]: Entering directory '/home/x86/lede/tools/meson'
install -d -m0755 /home/x86/lede/staging_dir/host/bin
/home/x86/lede/build_dir/host/meson-0.63.1/packaging/create_zipapp.py /home/x86/lede/build_dir/host/meson-0.63.1 --outfile /home/x86/lede/staging_dir/host/bin/meson.py
Traceback (most recent call last):
  File "/home/x86/lede/build_dir/host/meson-0.63.1/packaging/create_zipapp.py", line 23, in <module>
    zipapp.create_archive(d, interpreter=options.interpreter, target=options.outfile, compressed=options.compress)
TypeError: create_archive() got an unexpected keyword argument 'compressed'
Makefile:36: recipe for target '/home/x86/lede/staging_dir/host/stamp/.meson_installed' failed
make[3]: *** [/home/x86/lede/staging_dir/host/stamp/.meson_installed] Error 1
make[3]: Leaving directory '/home/x86/lede/tools/meson'
time: tools/meson/compile#0.23#0.40#0.68
    ERROR: tools/meson failed to build.
tools/Makefile:171: recipe for target 'tools/meson/compile' failed
make[2]: *** [tools/meson/compile] Error 1
make[2]: Leaving directory '/home/x86/lede'
tools/Makefile:169: recipe for target '/home/x86/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyn' failed
make[1]: *** [/home/x86/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyn] Error 2
make[1]: Leaving directory '/home/x86/lede'
/home/x86/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

lovehackintosh, 2022-10-01:
请更新python到3.9试试

bi4tms, 2022-10-01:
> 请更新python到3.9试试

请问如何更新

bi4tms, 2022-10-01:
> 请更新python到3.9试试

将Python升级至3.9.14后编译后仍然失败，日志同上

lovehackintosh, 2022-10-01:
```shell 
wget -O tools/meson/Makefile https://github.com/coolsnowwolf/lede/raw/de230d72c8101bfb243012aaaa31d890071c3b89/tools/meson/Makefile
```

lovehackintosh, 2022-10-01:
我在Arch Linux 和Ubuntu 22.04 中，python 3.10.7编译都没有问题，你用上面那条命令还原版本再试试

PR #10214 (2022-10-01): Revert "tools/meson: update to 0.63.1"
This reverts commit 5fcaadd8ce51e46a32577f7bda023930a5a217c8.
fix: #10213 
Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
closed 2022-10-01 by coolsnowwolf