coolsnowwolf/lede #10216: gcc12编译libnl-tiny报错error: 'strncpy' output may be truncated copying 15 bytes from a string of length 15 [-Werror=stringop-truncation]

lovehackintosh, 2022-10-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：gcc12编译libnl-tiny报错error: 'strncpy' output may be truncated copying 15 bytes from a string of length 15 [-Werror=stringop-truncation] 
TARGET_CFLAGES添加-Wno-error无效

### (2) 路由器型号和固件版本

A：ipq807x/ax3600

### (3) 详细日志

A：
```
make[3]: Entering directory '/home/clover/lede/package/libs/libnl-tiny'
mkdir -p /home/clover/lede/dl
SHELL= flock /home/clover/lede/tmp/.libnl-tiny-2022-05-17-b5b2ba09.tar.xz.flock -c '    /home/clover/lede/scripts/download.pl "/home/clover/lede/dl" "libnl-tiny-2022-05-17-b5b2ba09.tar.xz" "b957d56aa8c2e7b55184111be69eb8dea734f1feba19e670a91f302459a48a78" "" || (    echo "Checking out files from the git repository..."; mkdir -p /home/clover/lede/tmp/dl && cd /home/clover/lede/tmp/dl && rm -rf libnl-tiny-2022-05-17-b5b2ba09 && [ \! -d libnl-tiny-2022-05-17-b5b2ba09 ] && git clone  https://git.openwrt.org/project/libnl-tiny.git libnl-tiny-2022-05-17-b5b2ba09 && (cd libnl-tiny-2022-05-17-b5b2ba09 && git checkout b5b2ba09c4f1c8b3c21580aea7223edc2f5e92be && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd libnl-tiny-2022-05-17-b5b2ba09 && git log -1 --format='\''@%ct'\''` && rm -rf libnl-tiny-2022-05-17-b5b2ba09/.git &&      tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libnl-tiny-2022-05-17-b5b2ba09 |      xz -zc -7e > /home/clover/lede/tmp/dl/libnl-tiny-2022-05-17-b5b2ba09.tar.xz && mv /home/clover/lede/tmp/dl/libnl-tiny-2022-05-17-b5b2ba09.tar.xz /home/clover/lede/dl/ && rm -rf libnl-tiny-2022-05-17-b5b2ba09;  )    '
rm -f /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/.built
touch /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/.built_check
MAKEFLAGS="" /home/clover/lede/staging_dir/host/bin/ninja  -j1 -C /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09 
ninja: Entering directory `/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09'
[1/9] Building C object CMakeFiles/nl-tiny.dir/genl_family.c.o
FAILED: CMakeFiles/nl-tiny.dir/genl_family.c.o 
/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc -Dnl_tiny_EXPORTS -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/include -Ofast -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09=libnl-tiny-2022-05-17-b5b2ba09 -fPIC   -Wall -Werror -Wextra -Wno-unused-parameter -MD -MT CMakeFiles/nl-tiny.dir/genl_family.c.o -MF CMakeFiles/nl-tiny.dir/genl_family.c.o.d -o CMakeFiles/nl-tiny.dir/genl_family.c.o -c /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/genl_family.c
In function 'genl_family_add_grp',
    inlined from 'family_clone' at /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/genl_family.c:73:9,
    inlined from 'family_clone' at /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/genl_family.c:58:12:
/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/genl_family.c:148:9: error: 'strncpy' output may be truncated copying 15 bytes from a string of length 15 [-Werror=stringop-truncation]
  148 |         strncpy(grp->name, name, GENL_NAMSIZ - 1);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:54: /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/libnl-tiny-2022-05-17-b5b2ba09/.built] Error 1
make[3]: Leaving directory '/home/clover/lede/package/libs/libnl-tiny'
time: package/libs/libnl-tiny/compile#0.13#0.03#0.16
    ERROR: package/libs/libnl-tiny failed to build.
make[2]: *** [package/Makefile:116: package/libs/libnl-tiny/compile] Error 1
make[2]: Leaving directory '/home/clover/lede'
make[1]: *** [package/Makefile:110: /home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/clover/lede'
make: *** [/home/clover/lede/include/toplevel.mk:230: world] Error 2
make V=s -j1  20.16s user 2.76s system 100% cpu 22.752 total
```

aiamadeus, 2022-10-02:
Try this: https://github.com/aiamadeus/lede/commit/1775dd181c13dd327ebfc8ae45bfe0c0eaacb3e0

lovehackintosh, 2022-10-02:
Thank you very much,it worked for me!

lovehackintosh, 2022-10-02:
但是,package/system/rpcd同样编译报错
```
make[3]: Entering directory '/home/clover/lede/package/system/rpcd'                             mkdir -p /home/clover/lede/dl                   SHELL= flock /home/clover/lede/tmp/.rpcd-2022-09-12-e80d0b2d.tar.xz.flock -c '          /home/clover/lede/scripts/download.pl "/home/clover/lede/dl" "rpcd-2022-09-12-e80d0b2d.tar.xz" "90ce347d391694e6207ae3c57aaf8b80b7cfe043823f3610244c8f47e9017935" "" || (       echo "Checking out files from the git repository..."; mkdir -p /home/clover/lede/tmp/dl && cd /home/clover/lede/tmp/dl && rm -rf rpcd-2022-09-12-e80d0b2d && [ \! -d rpcd-2022-09-12-e80d0b2d ] && git clone  https://git.openwrt.org/project/rpcd.git rpcd-2022-09-12-e80d0b2d && (cd rpcd-2022-09-12-e80d0b2d && git checkout e80d0b2d77f8ed163872971200d24c423db60bb8 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd rpcd-2022-09-12-e80d0b2d && git log -1 --format='\''@%ct'\''` && rm -rf rpcd-2022-09-12-e80d0b2d/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c rpcd-2022-09-12-e80d0b2d |         xz -zc -7e > /home/clover/lede/tmp/dl/rpcd-2022-09-12-e80d0b2d.tar.xz && mv /home/clover/lede/tmp/dl/rpcd-2022-09-12-e80d0b2d.tar.xz /home/clover/lede/dl/ && rm -rf rpcd-2022-09-12-e80d0b2d;  )    '                  rm -f /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/.builttouch /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/.built_check                                          MAKEFLAGS="" /home/clover/lede/staging_dir/host/bin/ninja  -j1 -C /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d                                           ninja: Entering directory `/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d'                                 [1/7] Building C objec...s/rpcd.dir/session.c.o
FAILED: CMakeFiles/rpcd.dir/session.c.o         /home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc -DHAVE_SHADOW -DINSTALL_PREFIX=\"/usr\" -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/include -Ofast -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d=rpcd-2022-09-12-e80d0b2d -DNDEBUG   -Os -Wall -Werror --std=gnu99 -g3 -Wmissing-declarations -MD -MT CMakeFiles/rpcd.dir/session.c.o -MF CMakeFiles/rpcd.dir/session.c.o.d -o CMakeFiles/rpcd.dir/session.c.o -c /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/session.c                                   /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/session.c: In function 'rpc_session_to_blob':               /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/session.c:229:42: error: implicit declaration of function 'uloop_timeout_remaining64'; did you mean 'uloop_timeout_remaining'? [-Werror=implicit-function-declaration]                                        229 |         blobmsg_add_u64(&buf, "expires", uloop_timeout_remaining64(&ses->t) / 1000);          |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~                            |                                          uloop_timeout_remaining                        cc1: all warnings being treated as errors       ninja: build stopped: subcommand failed.        make[3]: *** [Makefile:104: /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/.built] Error 1                 make[3]: Leaving directory '/home/clover/lede/package/system/rpcd'                              time: package/system/rpcd/compile#0.38#0.02#0.39    ERROR: package/system/rpcd failed to build. make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1                        make[2]: Leaving directory '/home/clover/lede'  make[1]: *** [package/Makefile:110: /home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2                make[1]: Leaving directory '/home/clover/lede'  make: *** [/home/clover/lede/include/toplevel.mk:230: world] Error 2
```

lovehackintosh, 2022-10-02:
但是,package/system/rpcd同样编译报错，op官方最新版本同样报错
```
make[3]: Entering directory '/home/clover/lede/package/system/rpcd'
mkdir -p /home/clover/lede/dl
SHELL= flock /home/clover/lede/tmp/.rpcd-2022-09-12-e80d0b2d.tar.xz.flock -c '  	/home/clover/lede/scripts/download.pl "/home/clover/lede/dl" "rpcd-2022-09-12-e80d0b2d.tar.xz" "90ce347d391694e6207ae3c57aaf8b80b7cfe043823f3610244c8f47e9017935" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /home/clover/lede/tmp/dl && cd /home/clover/lede/tmp/dl && rm -rf rpcd-2022-09-12-e80d0b2d && [ \! -d rpcd-2022-09-12-e80d0b2d ] && git clone  https://git.openwrt.org/project/rpcd.git rpcd-2022-09-12-e80d0b2d && (cd rpcd-2022-09-12-e80d0b2d && git checkout e80d0b2d77f8ed163872971200d24c423db60bb8 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd rpcd-2022-09-12-e80d0b2d && git log -1 --format='\''@%ct'\''` && rm -rf rpcd-2022-09-12-e80d0b2d/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c rpcd-2022-09-12-e80d0b2d | 	xz -zc -7e > /home/clover/lede/tmp/dl/rpcd-2022-09-12-e80d0b2d.tar.xz && mv /home/clover/lede/tmp/dl/rpcd-2022-09-12-e80d0b2d.tar.xz /home/clover/lede/dl/ && rm -rf rpcd-2022-09-12-e80d0b2d;  )    '
rm -f /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/.built
touch /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/.built_check
MAKEFLAGS="" /home/clover/lede/staging_dir/host/bin/ninja  -j1 -C /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d 
ninja: Entering directory `/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d'
[1/7] Building C object CMakeFiles/rpcd.dir/session.c.o
FAILED: CMakeFiles/rpcd.dir/session.c.o 
/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc -DHAVE_SHADOW -DINSTALL_PREFIX=\"/usr\" -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/include -Ofast -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d=rpcd-2022-09-12-e80d0b2d -DNDEBUG   -Os -Wall -Werror --std=gnu99 -g3 -Wmissing-declarations -MD -MT CMakeFiles/rpcd.dir/session.c.o -MF CMakeFiles/rpcd.dir/session.c.o.d -o CMakeFiles/rpcd.dir/session.c.o -c /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/session.c
/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/session.c: In function 'rpc_session_to_blob':
/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/session.c:229:42: error: implicit declaration of function 'uloop_timeout_remaining64'; did you mean 'uloop_timeout_remaining'? [-Werror=implicit-function-declaration]
  229 |         blobmsg_add_u64(&buf, "expires", uloop_timeout_remaining64(&ses->t) / 1000);
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                          uloop_timeout_remaining
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:104: /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/rpcd-2022-09-12-e80d0b2d/.built] Error 1
make[3]: Leaving directory '/home/clover/lede/package/system/rpcd'
time: package/system/rpcd/compile#0.38#0.01#0.39
    ERROR: package/system/rpcd failed to build.
make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1
make[2]: Leaving directory '/home/clover/lede'
make[1]: *** [package/Makefile:110: /home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/clover/lede'
make: *** [/home/clover/lede/include/toplevel.mk:230: world] Error 2
```

aiamadeus, 2022-10-02:
Try this: https://github.com/aiamadeus/lede/tree/libs

lovehackintosh, 2022-10-02:
Fixed,thanks.But [with uhttp 2022-08-12](https://git.openwrt.org/?p=project/uhttpd.git;a=commit;h=e3395cd90bed9b7b9fc319e79528fedcc0d947fe) .......
```
make[3]: Entering directory '/home/clover/lede/package/network/services/uhttpd'
mkdir -p /home/clover/lede/dl
SHELL= flock /home/clover/lede/tmp/.uhttpd-2022-08-12-e3395cd9.tar.xz.flock -c '  	/home/clover/lede/scripts/download.pl "/home/clover/lede/dl" "uhttpd-2022-08-12-e3395cd9.tar.xz" "14e9df9f85c406b8abbb14427e5f678383782500c549d842c0481cd954fc4ea3" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /home/clover/lede/tmp/dl && cd /home/clover/lede/tmp/dl && rm -rf uhttpd-2022-08-12-e3395cd9 && [ \! -d uhttpd-2022-08-12-e3395cd9 ] && git clone  https://git.openwrt.org/project/uhttpd.git uhttpd-2022-08-12-e3395cd9 && (cd uhttpd-2022-08-12-e3395cd9 && git checkout e3395cd90bed9b7b9fc319e79528fedcc0d947fe && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd uhttpd-2022-08-12-e3395cd9 && git log -1 --format='\''@%ct'\''` && rm -rf uhttpd-2022-08-12-e3395cd9/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c uhttpd-2022-08-12-e3395cd9 | 	xz -zc -7e > /home/clover/lede/tmp/dl/uhttpd-2022-08-12-e3395cd9.tar.xz && mv /home/clover/lede/tmp/dl/uhttpd-2022-08-12-e3395cd9.tar.xz /home/clover/lede/dl/ && rm -rf uhttpd-2022-08-12-e3395cd9;  )    '
touch /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9/.prepared_f89cddd7a5fa78f73e0f351307ec79f8_6664517399ebbbc92a37c5bb081b5c53_check
. /home/clover/lede/include/shell.sh; xzcat /home/clover/lede/dl/uhttpd-2022-08-12-e3395cd9.tar.xz | tar -C /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9
touch /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9/.prepared_f89cddd7a5fa78f73e0f351307ec79f8_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9/.configured_*
rm -f /home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.uhttpd_installed
mkdir -p /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9
(cd /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9; CFLAGS="-Ofast -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9=uhttpd-2022-08-12-e3395cd9 " CXXFLAGS="-Ofast -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/uhttpd-2022-08-12-e3395cd9=uhttpd-2022-08-12-e3395cd9 " LDFLAGS="-L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -lcrypt " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=aarch64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -lcrypt" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -lcrypt -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -lcrypt -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/usr;/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl -DCMAKE_PREFIX_PATH=/home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EX [... cut ...]
time: package/network/services/uhttpd/compile#1.31#0.14#1.39
    ERROR: package/network/services/uhttpd failed to build.
make[2]: *** [package/Makefile:116: package/network/services/uhttpd/compile] Error 1
make[2]: Leaving directory '/home/clover/lede'
make[1]: *** [package/Makefile:110: /home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/clover/lede'
make: *** [/home/clover/lede/include/toplevel.mk:230: world] Error 2
```

aiamadeus, 2022-10-02:
Try again with my branch;)

lovehackintosh, 2022-10-02:
编译成功，辛苦您了

closed 2022-10-02 by lovehackintosh