coolsnowwolf/lede #10217: 拉取最新源码编译5.19内核的r2s和r4s固件均报错

bi4tms, 2022-10-03:
拉取最新源码编译5.19内核的r2s和r4s固件均报错，日志如下：
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[3]: *** [Makefile:101: /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-05-31-1c482573/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/r2s.5.19/lede/package/system/rpcd'
time: package/system/rpcd/compile#0.18#0.78#1.08
    ERROR: package/system/rpcd failed to build.
make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1
make[2]: Leaving directory '/home/r2s.5.19/lede'
make[1]: *** [package/Makefile:110: /home/r2s.5.19/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r2s.5.19/lede'
make: *** [/home/r2s.5.19/lede/include/toplevel.mk:230: world] Error 2

lovehackintosh, 2022-10-03:
我用gcc12编译没有问题,你的日志没有发全make V=s -j1再试一次

bi4tms, 2022-10-03:
> 我用gcc12编译没有问题,你的日志没有发全make V=s -j1再试一次

已使用make V=s -j1重新编译，错误依旧，编译环境Ubuntu 20.04，具体日志如下：
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
ucode
    linked by target "ucode_plugin" in directory /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-05-31-1c482573

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[3]: *** [Makefile:101: /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-05-31-1c482573/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/r2s.5.19/lede/package/system/rpcd'
time: package/system/rpcd/compile#0.28#0.98#1.46
    ERROR: package/system/rpcd failed to build.
make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1
make[2]: Leaving directory '/home/r2s.5.19/lede'
make[1]: *** [package/Makefile:110: /home/r2s.5.19/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r2s.5.19/lede'
make: *** [/home/r2s.5.19/lede/include/toplevel.mk:230: world] Error 2

bi4tms, 2022-10-03:
> 我用gcc12编译没有问题,你的日志没有发全make V=s -j1再试一次

已使用make V=s -j1重新编译，错误依旧，编译环境Ubuntu 20.04，具体日志同上（已更新）

lovehackintosh, 2022-10-03:
```
wget -O package/system/rpcd/Makefile https://github.com/coolsnowwolf/lede/raw/4f4d41021721f22c745f11f93ca2ac4acb6ed554/package/system/rpcd/Makefile
```

bi4tms, 2022-10-03:
> ```
> wget -O package/system/rpcd/Makefile https://github.com/openwrt/openwrt/raw/94129cbefb6027cdfe2b7801a6e27a36d4ec58b8/package/system/rpcd/Makefile
> ```

更新代码后出现了新的问题，具体日志如下：
-- Build files have been written to: /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d
touch /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/.built
touch /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/.built_check
MAKEFLAGS="" /home/r2s.5.19/lede/staging_dir/host/bin/ninja  -j1 -C /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d
ninja: Entering directory `/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d'
[14/15] Building C object CMakeFiles/ucode_plugin.dir/ucode.c.o
FAILED: CMakeFiles/ucode_plugin.dir/ucode.c.o
/home/r2s.5.19/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc -DHAVE_SHADOW -DINSTALL_PREFIX=\"/usr\" -Ducode_plugin_EXPORTS -I/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/include -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d=rpcd-2022-09-12-e80d0b2d -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fPIC   -Os -Wall -Werror --std=gnu99 -g3 -Wmissing-declarations -MD -MT CMakeFiles/ucode_plugin.dir/ucode.c.o -MF CMakeFiles/ucode_plugin.dir/ucode.c.o.d -o CMakeFiles/ucode_plugin.dir/ucode.c.o -c /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/ucode.c
/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/ucode.c: In function 'rpc_ucode_api_init':
/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/ucode.c:1028:9: error: implicit declaration of function 'uc_search_path_init' [-Werror=implicit-function-declaration]
 1028 |         uc_search_path_init(&config.module_search_path);
      |         ^~~~~~~~~~~~~~~~~~~
/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/ucode.c:1028:36: error: 'uc_parse_config_t' has no member named 'module_search_path'
 1028 |         uc_search_path_init(&config.module_search_path);
      |                                    ^
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:104: /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2022-09-12-e80d0b2d/.built] Error 1
make[3]: Leaving directory '/home/r2s.5.19/lede/package/system/rpcd'
time: package/system/rpcd/compile#6.87#3.54#17.54
    ERROR: package/system/rpcd failed to build.
make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1
make[2]: Leaving directory '/home/r2s.5.19/lede'
make[1]: *** [package/Makefile:110: /home/r2s.5.19/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r2s.5.19/lede'
make: *** [/home/r2s.5.19/lede/include/toplevel.mk:230: world] Error 2

bi4tms, 2022-10-03:
> ```
> wget -O package/system/rpcd/Makefile https://github.com/coolsnowwolf/lede/raw/4f4d41021721f22c745f11f93ca2ac4acb6ed554/package/system/rpcd/Makefile
> ```

实在抱歉，更新以后依然报错，具体日志如下：
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/x86.5.10/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for getspnam
-- Looking for getspnam - found
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
ucode
    linked by target "ucode_plugin" in directory /home/x86.5.10/lede/build_dir/target-x86_64_musl/rpcd-2022-05-31-1c482573

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_LAUNCHER
    CMAKE_CXX_COMPILER
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_EXPORT_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY
    CMAKE_FIND_USE_PACKAGE_REGISTRY
    CMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY
    CMAKE_SHARED_LINKER_FLAGS
    DL_LIBRARY

CMake Generate step failed.  Build files cannot be regenerated correctly.
make[3]: *** [Makefile:101: /home/x86.5.10/lede/build_dir/target-x86_64_musl/rpcd-2022-05-31-1c482573/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/x86.5.10/lede/package/system/rpcd'
time: package/system/rpcd/compile#0.42#1.10#1.45
    ERROR: package/system/rpcd failed to build.
make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1
make[2]: Leaving directory '/home/x86.5.10/lede'
make[1]: *** [package/Makefile:110: /home/x86.5.10/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.5.10/lede'
make: *** [/home/x86.5.10/lede/include/toplevel.mk:230: world] Error 2

bi4tms, 2022-10-03:
> ```
> wget -O package/system/rpcd/Makefile https://github.com/coolsnowwolf/lede/raw/4f4d41021721f22c745f11f93ca2ac4acb6ed554/package/system/rpcd/Makefile
> ```

实在抱歉，更新后依旧报错，具体日志：
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/r2s.5.19/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for getspnam
-- Looking for getspnam - found
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_LAUNCHER
    CMAKE_CXX_COMPILER
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_EXPORT_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY
    CMAKE_FIND_USE_PACKAGE_REGISTRY
    CMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY
    CMAKE_SHARED_LINKER_FLAGS
    DL_LIBRARY

-- Build files have been written to: /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765
touch /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/.built
touch /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/.built_check
MAKEFLAGS="" /home/r2s.5.19/lede/staging_dir/host/bin/ninja  -j1 -C /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765
ninja: Entering directory `/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765'
[3/13] Building C object CMakeFiles/rpcd.dir/session.c.o
FAILED: CMakeFiles/rpcd.dir/session.c.o
/home/r2s.5.19/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc -DHAVE_SHADOW -I/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/include -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765=rpcd-2021-07-13-1fa35765 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG   -Os -Wall -Werror --std=gnu99 -g3 -Wmissing-declarations -MD -MT CMakeFiles/rpcd.dir/session.c.o -MF CMakeFiles/rpcd.dir/session.c.o.d -o CMakeFiles/rpcd.dir/session.c.o -c /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/session.c
/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/session.c: In function 'rpc_session_to_blob':
/home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/session.c:229:9: error: 'uloop_timeout_remaining' is deprecated: use uloop_timeout_remaining64 [-Werror=deprecated-declarations]
  229 |         blobmsg_add_u32(&buf, "expires", uloop_timeout_remaining(&ses->t) / 1000);
      |         ^~~~~~~~~~~~~~~
In file included from /home/r2s.5.19/lede/staging_dir/target-aarch64_generic_musl/usr/include/libubus.h:24,
                 from /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/session.c:25:
/home/r2s.5.19/lede/staging_dir/target-aarch64_generic_musl/usr/include/libubox/uloop.h:95:5: note: declared here
   95 | int uloop_timeout_remaining(struct uloop_timeout *timeout) __attribute__((deprecated("use uloop_timeout_remaining64")));
      |     ^~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:99: /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/.built] Error 1
make[3]: Leaving directory '/home/r2s.5.19/lede/package/system/rpcd'
time: package/system/rpcd/compile#1.53#1.67#3.26
    ERROR: package/system/rpcd failed to build.
make[2]: *** [package/Makefile:116: package/system/rpcd/compile] Error 1
make[2]: Leaving directory '/home/r2s.5.19/lede'
make[1]: *** [package/Makefile:110: /home/r2s.5.19/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r2s.5.19/lede'
make: *** [/home/r2s.5.19/lede/include/toplevel.mk:230: world] Error 2

lovehackintosh, 2022-10-03:
`rm -rf /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/`
版本已经回退了，不应该报错，你执行上面那条命令再编译试试

bi4tms, 2022-10-03:
> `rm -rf /home/r2s.5.19/lede/build_dir/target-aarch64_generic_musl/rpcd-2021-07-13-1fa35765/` 版本已经回退了，不应该报错，你执行上面那条命令再编译试试

首先make clean，然后wget -O package/system/rpcd/Makefile https://github.com/coolsnowwolf/lede/raw/4f4d41021721f22c745f11f93ca2ac4acb6ed554/package/system/rpcd/Makefile，接着make menuconfig，最后make V=s。结果同上

closed 2022-10-04 by aiamadeus