coolsnowwolf/lede #10221: 使用nginx时，Turbo ACC插件获取状态502

WROIATE, 2022-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用luci-nginx-ssl，Turbo ACC插件在开启第一个选项时，状态获取显示502，关闭第一个选项时恢复；同时其余插件获取状态正常

### (2) 路由器型号和固件版本

A：x86_64
OpenWrt R22.9.1 / LuCI Master (git-22.270.62759-d376b9d)

### (3) 详细日志

A：
[image]
[image]

nginx日志
```nginx
192.168.1.x - - [03/Oct/2022:19:21:16 +0800] "GET /cgi-bin/luci/admin/network/turboacc/status?_=0.1644270072060614 HTTP/1.1" 502 559 "http://192.168.1.1:8080/cgi-bin/luci/admin/network/turboacc" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/105.0.0.0 Safari/537.36 Edg/105.0.1343.53"
```
nginx配置
```nginx
server {
    listen 192.168.1.1:8080;
    server_name localhost;
    index nginx.html index.html index.htm index.php default.html default.htm default.php;
    root /www;
    client_max_body_size 300M;
    location /cgi-bin/luci {
        index index.html;
        include uwsgi_params;
        uwsgi_param SERVER_ADDR $server_addr;
        uwsgi_modifier1 9;
        uwsgi_pass unix:////var/run/luci-webui.socket;
    }
    location ~ /cgi-bin/cgi-(backup|download|upload|exec) {
        include uwsgi_params;
        uwsgi_param SERVER_ADDR $server_addr;
        uwsgi_modifier1 9;
        uwsgi_pass unix:////var/run/luci-cgi_io.socket;
    }

    location /luci-static {
        error_log stderr crit;
    }

    location /ubus {
        ubus_interpreter;
        ubus_socket_path /var/run/ubus/ubus.sock;
        ubus_parallel_req 2;
    }

    location ~* \.(?:css(\.map)?|js(\.map)?|jpe?g|png|gif|ico|cur|heic|webp|tiff?|mp3|m4a|aac|ogg|midi?|wav|mp4|mov|webm|mpe?g|avi|ogv|flv|wmv)$ {
        expires 30d;
    }

    location ~* \.(?:svgz?|ttf|ttc|otf|eot|woff2?)$ {
        add_header Access-Control-Allow-Origin "*";
        expires 30d;
    }
}
```

mkmichael, 2022-10-11:
遇到同样的问题，搜寻答案，发现楼主提出来了。敬礼，敬礼

WROIATE, 2022-10-20:
> 

你这个是配置了http2才会出现的，报错不是502，一般是https配的有问题；
我这个是502，是网关报错

tanst, 2022-10-20:
> > 
> 
> 你这个是配置了http2才会出现的，报错不是502，一般是https配的有问题； 我这个是502，是网关报错

我这个问题和它这个一模一样：https://www.jianshu.com/p/8d8de360f8ff
谷歌上的哪些什么改缓存的方法都试过了，其他页面也都正常

WROIATE, 2022-10-20:
@tanst 可以去了解下http2，http2传输是二进制压缩数据([wiki](https://zh.wikipedia.org/wiki/HTTP/2#%E6%96%B0%E7%89%B9%E6%80%A7))，如果数据不对，浏览器获取到数据后会解析失败报错，你的问题最简单的解决办法应该是关掉http2

WROIATE, 2022-10-26:
尝试将lua代码中检查fastpath的代码输出定向到/dev/null后恢复正常
```lua
local function fastpath_status()
	return luci.sys.call("/etc/init.d/turboacc check_status fastpath >/dev/null") == 0
end
```

tanst, 2022-10-28:
> 尝试将lua代码中检查fastpath的代码输出定向到/dev/null后恢复正常
> 
> ```lua
> local function fastpath_status()
> 	return luci.sys.call("/etc/init.d/turboacc check_status fastpath >/dev/null") == 0
> end
> ```

改哪个文件？

WROIATE, 2022-10-29:
> 改哪个文件？
/usr/lib/lua/luci/controller/turboacc.lua

WROIATE, 2022-10-29:
@tanst /usr/lib/lua/luci/controller/turboacc.lua

fairok, 2022-11-13:
我的使用最新的代码也出一样的问题

tanst, 2023-02-02:
> 尝试将lua代码中检查fastpath的代码输出定向到/dev/null后恢复正常
> 
> ```lua
> local function fastpath_status()
> 	return luci.sys.call("/etc/init.d/turboacc check_status fastpath >/dev/null") == 0
> end
> ```

改了需要重启吗？

tanst, 2023-02-22:
> 尝试将lua代码中检查fastpath的代码输出定向到/dev/null后恢复正常
> 
> ```lua
> local function fastpath_status()
> 	return luci.sys.call("/etc/init.d/turboacc check_status fastpath >/dev/null") == 0
> end
> ```

试了试，无效

ppdragon16, 2023-03-12:
> 尝试将lua代码中检查fastpath的代码输出定向到/dev/null后恢复正常
> 
> ```lua
> local function fastpath_status()
> 	return luci.sys.call("/etc/init.d/turboacc check_status fastpath >/dev/null") == 0
> end
> ```

试了，可以。多谢！

改完后：
rm /tmp/luci-indexcache
rm -rf /tmp/luci-modulecache/

FanxJK, 2023-05-19:
将 nginx 错误日志调整为 info 级别发现:

```
upstream sent invalid header: "Flow\\x20..." while reading response header from upstream
```
发现是无法处理`/etc/init.d/turboacc check_status fastpath`命令的输出:  `Flow Offloading` 中的空格
导致 response 中的 header 出现：`Flow OffloadingStatus: 200 OK`，而 HTTP Header 中不允许出现空格，所以 nginx 会出现 502 错误。
同理，`MediaTek HWNAT`和`Shortcut-FE ECM`也会有此问题
不知道哪里把`Flow Offloading`给加到 Header 里去了

wanmyj, 2023-11-01:
修改后测试成功，总结如下

我的错误信息是 `[error] 4188#0: *64 upstream sent invalid header: "Flow\x20..." while reading response header from upstream, client: 192.168.1.228, server: _lan, request: "GET /cgi-bin/luci/admin/network/turboacc/status?1698850818711 HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/network/turboacc"
`

找到 `/usr/lib/lua/luci/controller/turboacc.lua`文件，把对应内容改成
```lua
local function fastpath_status()
	return luci.sys.call("/etc/init.d/turboacc check_status fastpath >/dev/null") == 0
end
```

然后运行
`rm -rf /tmp/luci-modulecache/`
再去配置turboacc，万事大吉

coolsnowwolf, 2026-09-05:
该 issue 基于 2022 年 LuCI/nginx 自定义配置，唯一日志是状态请求返回 502，没有 nginx upstream 错误、rpcd 日志或 TurboACC RPC 响应。当前 luci-app-turboacc 已改为通过 `luci.turboacc` RPC 获取状态，且页面其它插件正常并不能证明该自定义 uWSGI/ubus 配置兼容；应先直接用 uhttpd 验证，再检查 nginx 的 uWSGI 与 `/ubus` 代理。缺少现行版本和可复现配置，无法提交通用补丁，关闭历史 issue。

closed 2026-09-05 by coolsnowwolf